kbialek/deye-inverter-mqtt #132: Integration "time of use"

m1k3f15h, 2023-10-13:
**Is your feature request related to a problem? Please describe.**
Would be cool to edit the "Timetable" over mqtt.

**Describe the solution you'd like**
Edit the "Time of use" over mqtt (like active power regulation).

**Describe alternatives you've considered**
Change it on the Display :(

**Additional context**
Did anyone else had a usecase for this?
[image]

greetings from austria
Mike

finfin82, 2023-10-28:
This would be awesome to use in combination with an dynamic energy price like tibber or awattar...
I hope this feature will be implemented

kbialek, 2023-10-28:
The tricky part is to identify the respective Modbus registers.
In the Deye docs I see that these registers could fit, but someone would have to confirm that, by at least reading them using the command line tool, as described here https://github.com/kbialek/deye-inverter-mqtt#reading-and-writing-raw-register-values
[image]
[image]

finfin82, 2023-10-29:
So you don't own an deye 12k or so yourself?
A complete tibber api Implementation would be awesome!

Maybe I can find the time tonight, if nobody is faster than me

finfin82, 2023-10-29:
So before trying to write values, i tried to read the values:
[image]
Thats the result.. any ideas?
Greetings

kbialek, 2023-10-30:
@finfin82 Assuming, that you have the time table configured, these register addresses do not work for your inverter.

finfin82, 2023-10-30:
> @finfin82 Assuming, that you have the time table configured, these register addresses do not work for your inverter.
maybe i did something wrong?
i have a timetable configured - may be the registers are wrong?
Reading these values via the deye or solarman pro app works successful.

kbialek, 2023-10-30:
@finfin82 I found another Modbus registry spec [here](https://github.com/StephanJoubert/home_assistant_solarman/files/9368219/MODBUS.RTUDEYEubersetzt.pdf)

Try reading these registers
[image]

finfin82, 2023-10-31:
Thats it! i will try to write sell power of 154 as well...
[image]

finfin82, 2023-11-03:
@kbialek well any progress here? does it make sense to support you in any way?
guess you will be faster implementing the timetable-settings now where we know the correct registers...
since python is far away from my mother tongue :-)

greetings and thank you for your time and for this great project!

kbialek, 2023-11-04:
@finfin82 No progress code-wise. I have other, more important topics to work on at the moment.

> does it make sense to support you in any way?

Yes, you can think what could be the most desired time-table mqtt "API". I can propose two extreme options at the moment:

1. Fine-grained - Map each property (time, soc, etc...) to an individual mqtt command topic. This means that you have to publish multiple mqtt messages to update the whole table. On the other hand you can modify individual properties easily.
2. Coarse-grained - Map the entire table to a single mqtt command topic. Send a time-table JSON object in the mqtt message. Update entire table at once.

finfin82, 2023-11-04:
Well my tibber contract starts in June, so for my part there is no hurry.

I can imagine that there is a use case for both.
I'm using node red to fetch the tibber prices and can imagine to push the whole time table at once, cause I have the whole day available in one task.
There is a use case for manipulating single values as well.

So may be just implement the easiest way at first and add the more complex part when there is more time for that.

When you implement the settings, can you please make sure that there is a way of retry in case one single command fails... I have some crc errors from time to time. The next try works perfect then.

Thank you very much!

kbialek, 2023-12-07:
I've started working on it. There is a lot to be done, so be patient. I will implement the reading side first. So far I've imported updated HA definitions for `deye_hybrid` where ToU registers are already defined. The next step is to add registers used by your inverters. In case you are interested in the code changes, the PR is already opened https://github.com/kbialek/deye-inverter-mqtt/pull/138

kbialek, 2023-12-09:
@finfin82 @m1k3f15h What inverter models do you have exactly? What metric groups do you have set in the config?

finfin82, 2023-12-09:
good evening @kbialek my model is the 12K-SG04LP3 according to the manual.
my metrics: deye_sg04lp3,deye_sg04lp3_ups
thanks for your great work and infos about he PR... i will have a look next days
greetings

kbialek, 2023-12-10:
@finfin82 What metric group did you add to your config?

finfin82, 2023-12-10:
i added the deye_hybrid_timeofuse metric.
Just realized this was for another model defined? 

here are my current settings:
[image]

kbialek, 2023-12-10:
Use `deye_sg04lp3_timeofuse` metric group

finfin82, 2023-12-10:
ahh just got it... mixed up something... sorry.
works perfect here is the output:
[image]

kbialek, 2023-12-10:
@finfin82 Can you really control at which day energy selling is enabled? This register's description indicates that such a feature might exist, however I couldn't find it in the inverter's manual https://www.deyeinverter.com/deyeinverter/2021/04/14/502012252%E3%80%90b%E3%80%91%E5%82%A8%E8%83%BD%E4%B8%89%E7%9B%B8%E8%AF%B4%E6%98%8E%E4%B9%A6sun-8-12k-sg04lp3-ver2.1-deye.pdf
[image]

finfin82, 2023-12-10:
@kbialek yes this feature came by one of the lated firmware-updates of the inverter and userinterface.
guess its only visible on the interface directly at the inverter... not even in those online-tools for editing settings

Not really necessary for our usecase but would make it complete.

m1k3f15h, 2023-12-11:
> @finfin82 @m1k3f15h What inverter models do you have exactly? What metric groups do you have set in the config?

Hy,
Inverter: SUN-12K-SG04LP3
Metric Groups: deye_sg04lp3, deye_sg04lp3_battery, deye_sg04lp3_ups, settings

Thanks for working on this feature request :)

mike

finfin82, 2023-12-18:
@kbialek just give me a ping when you need me to test something... Looking nice so far

finfin82, 2023-12-22:
awesome work!
i checked it out and was trying to send commands to those topics ..
here are my results as seen in mqtt-explorer... guess it didnt work ... but im sure im doing something wrong.
im using ubuntu do you have a suggestion for mqtt-client to use there?
[image]

kbialek, 2023-12-22:
@finfin82 please collect and share debug logs

finfin82, 2023-12-22:
[debug.log](https://github.com/kbialek/deye-inverter-mqtt/files/13755316/debug.log)
looks like there is some kind of race condition?!
says mqtt-client not connected but can successful publish timeofuse data.
my way to send the data is right? <-- just for the record and be sure?

kbialek, 2023-12-22:
@finfin82 

> looks like there is some kind of race condition?!

Yes. There is a race condition between establishing mqtt server connection and subscribing to the command topics. This is not a blocker and I will address this later.

I couldn't find write command in the debug log. Did you send it?

finfin82, 2023-12-22:
Yes I did, but couldn't find it... I made several attempts... Guess the subscription was not successful either. Even when there was no error... 
Maybe the local class client get lost? 
Just an idea... I had not enough time to have a look at the code 

Greetings

kbialek, 2023-12-22:
@finfin82 I fixed the subscription race condition. Please try to send "write" again.

finfin82, 2023-12-22:
awesome, works perfect!
first i had some similar issues - then i realised, that there is my old deye-inverter-mqtt docker running that seemed to blocking the mqtt functionality.

i tested some of each kind... only manipulating the time didnt work - maybe there is a wrong register in the background?
should have been 148 for my inverter?
do you need a logging for that?
thanks again for your awesome work!

kbialek, 2023-12-22:
> only manipulating the time didnt work

Collect and share debug logs

finfin82, 2023-12-23:
a bit longer log in line 609 you see the /timeofuse/time/1/command setting to 31
followed by the write command
the next read you can see its still 30
[debug.log](https://github.com/kbialek/deye-inverter-mqtt/files/13758786/debug.log)

kbialek, 2023-12-23:
@finfin82 Yes, you're right. For some reason value 31 is not processed by the inverter.
I've extracted modbus request and response frames from your log. They both look good. In the screenshot below, you can see that value 31 has been sent in the request.

Perhaps the inverter does some sort of rounding? What is the time granularity in the App or inverter UI?
Can you try to write register 148 (`timeofuse/time/1`) individually? Does any of those command have effect?
```
# Set 01:00
docker run --rm --env-file ./config.env deye-inverter-mqtt w 148 100

# Set 01:01
docker run --rm --env-file ./config.env deye-inverter-mqtt w 148 101

# Set 01:05
docker run --rm --env-file ./config.env deye-inverter-mqtt w 148 105

# Set 01:30
docker run --rm --env-file ./config.env deye-inverter-mqtt w 148 130
```

Request:
[image]

Response:
[image]

finfin82, 2023-12-23:
yep thats it, seems to be some kind of 5 minute grid.
first setting via commandline confirmed this - setting via mqtt works perfectly with this knowledge...
sorry for my bad debugging skills at this time - more to blame the lack of time these days...
guess we have a working release now?!

anybody here able to build a node-red automation including tibber api? :-)

will give it a try after chrismas i think.

@kbialek thank you very much - awesome work!

kbialek, 2023-12-23:
@finfin82 Nice! Now you owe me a tasty drink :laughing:

kbialek, 2023-12-26:
@finfin82 Beta image is ready for testing. Please check it out. `ghcr.io/kbialek/deye-inverter-mqtt:2024.01.1-beta-1`

finfin82, 2023-12-26:
@kbialek i tried every setting property - soc is not possible for me to test, since im running on BatV configuration.
seems to work perfect for me!
maybe @m1k3f15h could give it a try too?

thanks

finfin82, 2024-01-03:
@m1k3f15h hi mike, any time to test this release?
my instance is working fine

m1k3f15h, 2024-01-04:
Hey guys
Sorry for my late response.
I was on holidays..

I hope i find some time in the next few days to test this great feature :)

Greetings 
Mike

joe-678, 2024-01-07:
Hi all, 
I have tested TimeofUse with SUN-12K-SG04LP3. It looks very good.
I will implement some variable loading based on tibber prices this week and will give feedback until end of the week.
Until now awesome work. Thanks a lot @kbialek !!!

finfin82, 2024-01-07:
@joe-678 nice! can you please share your solution how you integrated tibber in this context?

joe-678, 2024-01-08:
No problem. I have integrated my deye via mqtt running on a docker same as an node red implementation. For node red you can retrieve price info for tomorrow and with a solar production forecast for my PV(could also be integrated into node red) i would like to derieve estimated battery yield and then calculate how much I need to charge the battery and when it is the best time to do so. Not ready yet, but Friday I am moving to tibber ;-)

finfin82, 2024-01-08:
im thinking about a similar solution
maybe you can share your configuration... i still didnt manged to compose an according message from the detected shedule
[image]

kbialek, 2024-01-13:
I assume the feature is working well for you. I will release it tomorrow.

finfin82, 2024-01-13:
@kbialek yep working well for me, please let me know how to buy you a beer ;-)

kbialek, 2024-01-14:
> @kbialek yep working well for me, please let me know how to buy you a beer ;-)

Okay, it's out. @finfin82 Just read the first line of the [README](https://github.com/kbialek/deye-inverter-mqtt/blob/main/README.md) :smile:

closed 2024-01-14 by kbialek

kbialek, 2024-01-16:
> only one little "mistake" in the Readme:
> {MQTT_TOPIC_PREFIX}/timeofuse/power/(1-6)/command --> say only 0-8000 works.. i tested 12000 (default of my inverter) --> also works great :)

Thanks. README updated. I took the original value from the modbus protocol documentation, but obviously it depends on the inverter max power.