kbialek/deye-inverter-mqtt #133: Problem reading UPS power when value is negative

fakehec, 2023-11-14:
Hi,

I have a Deye SUN SG01HP3-EU
I am using the metrics 
DEYE_METRIC_GROUPS=deye_sg01hp3,deye_sg01hp3_battery,deye_sg01hp3_ups,settings

This model has three load connections, one named UPS (which is normally used for backup loads), one for smart loads and/or a generator, and the main connection which not necessarily has to be connected directly to the gird, as there can be loads connected in between the inverter and the grid.  If there are other solar inverters in the system, here is where they should be connected. If so, there should be a set of CT's or a modbus power metter conected to the entry grid for the Deye to know about the feed-in/buying power to/from the grid.  When in this last setup, you should select the Work Mode "Zero Export to CT" .

Problem arises when selecting "Zero Export to CT", and there are other inverters on the system. It looks like whe in this mode, the server consider UPS load not the load on the UPS line, but the sum of the loads on the UPS line and the main line, as it considers grid connection where the CT are, not at the main line connection.  Being other inverters in the system, the UPS power reported can be negative. And this is correctly reported by the Deye screen. 

But when negative UPS power happens, deye-inverter-mqtt does not understand it, reporting something around 65KW in each on the negative lines (something around 65535, or 2^16)
  
The expected behabiour should be reporting the negative power correctly.

Regards.

kbialek, 2023-11-14:
@fakehec It means that this register is currently mapped as unsigned int, while it should be mapped as signed int. Which metric is it exactly?

fakehec, 2023-11-14:
ac/ups/l1/power
ac/ups/l2/power
ac/ups/l3/power

Guess they are at deye_sg01hp3_ups

kbialek, 2023-11-14:
@fakehec Fixed in the new release

fakehec, 2023-11-14:
Thanks a lot!