kbialek/deye-inverter-mqtt #135: How to run plugin from docker image

fakehec, 2023-11-15:
Hi,

I have a working installation of deye-inverter-mqtt.

I have followed the guidelines to star the docker image with the plugins directory.

I haver downloaded stdout-publisher to the plugins directory.

How to run the docker file to use a plugin?

Regards.

kbialek, 2023-11-15:
@fakehec It's documented here https://github.com/kbialek/deye-inverter-mqtt#how-to-start-the-docker-container-with-custom-plugins

`./plugins` is the path to the directory where you put the plugin file

fakehec, 2023-11-15:
Yes, i read it. My question is how to make the docker program to run the plugin, as for example stdout-publishing.

kbialek, 2023-11-15:
You do not have to anything more. If the plugins directory is mounted correctly, you should see log messages similar to these
```
2023-11-15 21:04:48,754 - DeyePluginLoader - INFO - Loading plugin: 'deye_plugin_sample'
2023-11-15 21:04:48,754 - DeyePluginLoader - INFO - Loading custom event processor: 'sample_publisher'
```

fakehec, 2023-11-15:
Thanks a lot

closed 2023-11-15 by fakehec