kbialek/deye-inverter-mqtt #137: docker compose file / portainer

GerhardLang, 2023-11-21:
**Hardware (please complete the following information):**
 - Deye 800 and 1600:

**Software (please complete the following information):**
 - logger version: actual
 - deye-inverter-mqtt version: actual
 - compose file:
 version: "3"

services:
  deye-mqtt:
    container_name: deye
    image: ghcr.io/kbialek/deye-inverter-mqtt:latest
    volumes:
      - /volume1/docker/deyemqtt/config.env:config.env
    env_file:
      - config.env
    network_mode: host
    restart: unless-stopped
    environment:
      - TZ=Europe/Amsterdam

Hello, I would like to create the container with portainer but the config.env file is not existing, it is named config.env.example.
The container will not be created, so I can not rename this file.
Can I do a correct file mapping to my NAS-drive with the volumes: option? What is the correct source path for the expected  config.env.

Thank you for a not in to deep user

Gerhard

kbialek, 2023-11-21:
The Docker container completely disregards the config.env file. This file is intended to define environment variables, which the Docker engine sets for the container upon startup.
Maybe this article will help you https://www.portainer.io/blog/using-env-files-in-stacks-with-portainer

GerhardLang, 2023-11-21:
Fantastic, it works Great!!!

closed 2023-11-22 by kbialek