kbialek/deye-inverter-mqtt #139: Deye 50K Battery 2 data?

SaraDark, 2023-12-13:
**Hardware (please complete the following information):**
Deye SUN-50K-SG01HP3-EU

**Software (please complete the following information):**
 - Logger firmware version: LSW3_15_FFFF_1.0.96

Question:

Hello and welcome,

I have Deye 50K Hybrid with 2 batteries connected directly to the battery outputs and active parallel option bat1&bat2.
Is the battery data averaged for 2 batteries or for 1, if for 1, how to get data also for 2?

Thank you and best regards.

kbialek, 2023-12-14:
hi @SaraDark 

> I have Deye 50K Hybrid with 2 batteries connected directly to the battery outputs and active parallel option bat1&bat2.
> Is the battery data averaged for 2 batteries or for 1, if for 1, how to get data also for 2?

I don't know that. Only you can figure out how it works by comparing the values reported over MQTT with the values you see in the mobile app.

SaraDark, 2023-12-14:
> hi @SaraDark
>
>
> I don't know that. Only you can figure out how it works by comparing the values reported over MQTT with the values you see in the mobile app.

I'm trying to figure it out over time :)

kbialek, 2024-01-02:
I will add mqtt topics for the 2nd battery only if you identify the corresponding Modbus registers. The 1st battery data is reported at these registers https://github.com/kbialek/deye-inverter-mqtt/blob/release/2024.01.1/docs/metric_group_deye_sg01hp3_battery.md

You can try reading other registers using this command:
```
docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r <reg_address>
```

Substitute `<reg_address>` with the address you want to check. Good luck!

kbialek, 2024-01-21:
Thank you for your contribution @voipmonitor https://github.com/kbialek/deye-inverter-mqtt/issues/41#issuecomment-1902238368

Looks good to me. I will simplify the grouping. Imho a single `battery` and `bms` metric group will be enough.
I will also adapt the topic names, so they will look like:
* `battery` - for battery totals
* `battery/1`, `battery/2` - for specific battery packs
* `bms/1` and `bms/2` - for bms units

I will add it to the code and release a new version soon.

voipmonitor, 2024-01-21:
hello, 

this is what I see in the solarmanpv portal - there is no battery, battery1 and battery2 concept, it seems that the battery2 registers was just added later. The question is what would you expect in the "battery" group only - would people know that you have to add battery + battery1 in all cases and battery + battery1 + battery2 in case you have two battery inputs? 

here is what you can see in the solarmanpv portal: 

Battery:
Battery Status：Discharging
Battery Voltage：631.8V
Battery Current：0.38A
Battery Power1：240W
Battery Power：500W
SoC：34%
Total Charging Energy：2.6928MWh
Total Discharging Energy：2.9848MWh
Daily Charging Energy：13.8kWh
Daily Discharging Energy：4.9kWh
Battery Type：lithium
Battery Factory：Deye_HV
Battery 1 Status：Discharging
Battery Number：2-way battery input

Battery2:
Battery Voltage2：635.5V
Battery Current 2：0.41ABattery 
Power 2：260W
SoC 2：35%
Battery2 Factory：Deye_HV
Battery 2 Status：Discharging
Temperature- Battery 2：10℃

kbialek, 2024-01-21:
> this is what I see in the solarmanpv portal - there is no battery, battery1 and battery2 concept, it seems that the battery2 registers was just added later. The question is what would you expect in [...]

This is exactly my point. I want to have just two metric groups available: `deye_sg01hp3_battery` and `deye_sg01hp3_bms`

However the MQTT topics will be grouped as follows:

* `battery/...` - for battery totals
* `battery/1/...`, `battery/2/...` - for specific battery packs
* `bms/1/...` and `bms/2/...` - for bms units

voipmonitor, 2024-01-21:
Oh I know what you mean now - I have missed " single battery and bms metric group " - looks good to me. I'm not using MQQT at all, I'm feeding influxdb with telegraf by the output plugin (which I had to slightely modify it)

closed 2024-01-28 by kbialek

championc, 2024-02-04:
I wonder if someone can assist me.  I cannot see any battery stats at all

This is what I see
deye
status = online
logger_status = online
day_energy = 6551.7
total_energy = 1532.6
>ac
>>l1
>>l2
>>l3
freq = 50.0
apparent_power = 11.1
active_power = 40901017.6
reactive_power = 0.0
uptime = 0.0
>dc
>>pv1
>>pv2
>>pv3
>>pv4
total_power = 0.0
operating_power = 0.0
radiator_temp = 38.8
igbt_temp = 35.0

Yes, I have a battery connected, although it is currently setup using voltage settings.  I hope my JK BMS will arrive this week