kbialek/deye-inverter-mqtt #140: "Logger is offline" Message

Chris-DE76, 2024-01-12:
**Hardware (please complete the following information):**
 - Inverter model: Deye SUN-100G3-EU-Q0

**Software (please complete the following information):**
 - Logger firmware version: MW3_16U_5406_2.32-D1
 - deye-inverter-mqtt version: 2023.11.1

----------------------------------------------------------------------------------

Hi,
I got my inverter about 2 weeks ago and have since then maybe poured 30hrs of effort into getting it connected to the Docker container I set up on my QNAP NAS to get the logger values into my InfluxDB/Grafana. Unfortunately, so far I failed. My problem is that the container's log tells me the data logger is offline.

I have also installed the "Solarman" app and I see the devices (logger, inverter, etc.) there as online and I am seeing data in the app too. So the logger itself (hardware, software) should be fine as far as I can tell.

When I start the container on my QNAP:

2024-01-12 07:06:19,621 - DeyeDaemon - INFO - Reading start
2024-01-12 07:06:19,621 - DeyeMqttClient - DEBUG - Publishing message. topic: 'Deye/logger_status', value: 'offline'
2024-01-12 07:06:19,622 - paho.mqtt.client - DEBUG - Sending CONNECT (u0, p0, wr1, wq1, wf1, c1, k60) client_id=b'deye-inverter-3805680698'
2024-01-12 07:06:19,623 - paho.mqtt.client - DEBUG - Received CONNACK (0, 0)
2024-01-12 07:06:19,623 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m1), 'b'Deye/logger_status'', ... (7 bytes)
2024-01-12 07:06:19,624 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m2), 'b'Deye/logger_status'', ... (7 bytes)
2024-01-12 07:06:19,625 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m3), 'b'Deye/logger_status'', ... (7 bytes)
2024-01-12 07:06:19,625 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 1)
2024-01-12 07:06:19,626 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 2)
2024-01-12 07:06:19,626 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 3)
2024-01-12 07:06:20,624 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r1, m4), 'b'Deye/status'', ... (6 bytes)
2024-01-12 07:06:20,624 - DeyeMqttClient - INFO - Successfully connected to MQTT Broker located at 192.168.178.xyz:1883
2024-01-12 07:06:20,625 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m5), 'b'Deye/logger_status'', ... (7 bytes)
2024-01-12 07:06:20,626 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 4)
2024-01-12 07:06:20,626 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 5)
2024-01-12 07:06:20,627 - DeyeMqttClient - INFO - Logger is offline
2024-01-12 07:06:20,627 - DeyeDaemon - INFO - Reading completed

I would like to understand better, what "Logger is offline" actually means to get some bearings for further investigations:
- does it mean the container established contact via the network and the inverter responded with a logger offline? (no idea what to do in this case as the app is showing data)
- does it mean the container was not able to contact the inverter at all via the network and the result is shown as logger offline? (maybe I need to look at my network)
- any other reason that could lead to "logger offline"?

My network setup is that the inverter is in a Wifi network (IP: 192.168.100.xyz) whereas the QNAP NAS is in the LAN (IP: 192.168.178.xyz). So far I have:
- disabled all firewall rules between the 2 networks
- tried ModBus/TCP and /AT on their respective ports

Any hint is greatly appreciated. Thank you everyone in advance!

Best regards,
Chris

kbialek, 2024-01-12:
hey @Chris-DE76 

> I would like to understand better, what "Logger is offline" actually means to get some bearings for further investigations:
>
>     * does it mean the container established contact via the network and [...]

It may mean any of those. The logger is considered to be online when the service is able to communicate with it.

> * Logger firmware version: MW3_16U_5406_2.32-D1

It's version 2.xx, so I guess only AT protocol will work.

> * trying port 48899 showed a "connection refused" ... so Modbus/tcp seems to the way to go.

Please note that the AT protocol uses 48899/**udp**. Also when switching to AT you should remove `DEYE_LOGGER_PORT` from the config (or set it to 48899)

> "DEYE_LOGGER_SERIAL_NUMBER=3805680xyz"

And the last but not least. Is this the S/N of the **logger**? Sometimes ppl set it to the inverter's S/N, but this will not work.

Did you remove log message from the debug log? I'd expect to see at least request frames in it.

Chris-DE76, 2024-01-13:
hey @kbialek ,

thank you so much for your reply. 

I double checked the SN and I changed the protocol to AT and the port to 48899. Initially it was still "Logger is offline". Then however - honestly no idea why exactly - I was wondering about your "I'd expect to see at least request frames in it." statement. 

Reading through your documentation again I noted that I did not have the "DEYE_METRIC_GROUPS" in my ENV items. My initial approach was to connect to the logger first (so establish a network connection) and then worry about picking the right configuration to retrieve data from the logger. This however was the acutal issue. With the "DEYE_METRIC_GROUPS" in ENV added and set to "micro", the data logger became online and data was visible immediately.

Well ... seems like my English or my general understanding of the topic were the issue here :) . 

Thanks again for your reply and - maybe not intentionally - pushing me in the reight direction.

Cheers,
Chris

closed 2024-01-13 by kbialek