kbialek/deye-inverter-mqtt #141: deye-sun-5k-sg03lp1 causes timeout when use port 48899 and AT commands

pfostenberg, 2024-01-23:
**Describe the bug**
A new deye-sun-5k-sg03lp1 ( port 8899) is closed an need to use 48899.

DEYE_LOGGER_IP_ADDRESS=192.168.x.x
DEYE_LOGGER_PORT=48899 
DEYE_LOGGER_SERIAL_NUMBER=271xxxxxx
DEYE_LOGGER_PROTOCOL=at

The config and deye-inverter-mqtt work with a deye-sun-8k-sg01lp1 like a charm.

DEYE_LOGGER_IP_ADDRESS=192.168.x.x 
DEYE_LOGGER_PORT=8899
DEYE_LOGGER_SERIAL_NUMBER=271051xxxx
DEYE_LOGGER_PROTOCOL=tcp
DEYE_METRIC_GROUPS=deye_hybrid,deye_hybrid_battery

but when used with the sg03lp1 it stops when reading more than 0x40 bytes.
 
**To Reproduce**
Steps to reproduce the behavior:

use the actual ( 1/2024) deye inverter.
start the docker image with config below.

sudo docker run --name deye-mqtt --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt
2024-01-22 23:02:58,884 - DeyeDaemon - INFO - Please help me build the list of compatible inverters. https://github.com/kbialek/deye-inverter-mqtt/issues/41
2024-01-22 23:02:58,884 - DeyeConnectorFactory - INFO - Creating Modbus/AT Logger connector
2024-01-22 23:02:58,921 - paho.mqtt.client - DEBUG - Sending CONNECT (u1, p1, wr1, wq1, wf1, c1, k60) client_id=b'deye-inverter-2799381840'
2024-01-22 23:02:58,923 - paho.mqtt.client - DEBUG - Received CONNACK (0, 0)
2024-01-22 23:02:59,924 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r1, m1), 'b'deye1/status'', ... (6 bytes)
2024-01-22 23:02:59,925 - DeyeMqttClient - INFO - Successfully connected to MQTT Broker located at 192.168.178.128:1883
2024-01-22 23:02:59,926 - DeyeDaemon - INFO - Feature "Report metrics over MQTT": enabled
2024-01-22 23:02:59,927 - DeyeDaemon - INFO - Feature "Set inverter time once online": disabled
2024-01-22 23:02:59,927 - DeyeDaemon - INFO - Feature "Time-of-use configuration over MQTT": disabled
2024-01-22 23:02:59,928 - DeyeDaemon - DEBUG - Start to execute the daemon at intervals of 60 seconds
2024-01-22 23:02:59,930 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 1)
2024-01-22 23:02:59,932 - DeyeDaemon - INFO - Reading start
2024-01-22 23:02:59,933 - DeyeDaemon - INFO - Reading registers [metrics group: deye_hybrid_battery, range: 0003-0070]
2024-01-22 23:02:59,934 - DeyeAtConnector - DEBUG - Sending AT command: b'WIFIKIT-214028-READ'
2024-01-22 23:03:00,035 - DeyeAtConnector - DEBUG - Received AT response in 1. attempt: b'192.168.178.150,402A8F39E514,2799381839'
2024-01-22 23:03:00,036 - DeyeAtConnector - DEBUG - Sending AT command: b'+ok'
2024-01-22 23:03:00,137 - DeyeAtConnector - DEBUG - Sending AT command: b'AT+INVDATA=8,0103000300**6e**3
-> Timeout starts.

check with deye-logger-at-cmd -> same.
main -t 192.168.178.128:48899 -xmb 00030070

pi@raspberrypi:~/git/deye-logger-at-cmd $ ./build/main -t 192.168.178.150:48899 -xmb **00030040**
2024/01/23 00:13:41 * Connecting :0 -> 192.168.178.150:48899...
2024/01/23 00:13:46 +ok=01038032333130323532313839000081020000151500003385C3650000C35000000201000000FF000A180117000B3100011388051400000000000000000000000000FF07D016760000000003E8FFFF000000000001000000010000000000000000C350000000001964000000000000000100010002000000FA000003E00000010E03B6F011
2024/01/23 00:13:47 
pi@raspberrypi:~/git/deye-logger-at-cmd $ ./build/main -t 192.168.178.150:48899 -xmb **00030070**
2024/01/23 00:13:52 * Connecting :0 -> 192.168.178.150:48899...
2024/01/23 00:13:58 read udp 192.168.178.128:34164->192.168.178.150:48899: i/o **timeout**

**Expected behavior**
if this a bug in this tool? the logger? or my mistake?
or does a workarround help ( reduce the big call with 1..2 shorter one)

**Hardware (please complete the following information):**
 - Inverter model: deye-sun-5k-sg03lp1

**Software (please complete the following information):**
 - Logger firmware version: v1.1.00.10
 - deye-inverter-mqtt version: config.env ghcr.io/kbialek/deye-inverter-mqtt from today... is it logged?
 - config.env (remove passwords before pasting):
 #
DEYE_LOGGER_IP_ADDRESS=192.168.178.150
DEYE_LOGGER_PORT=48899 
DEYE_LOGGER_SERIAL_NUMBER=2799381840
DEYE_LOGGER_PROTOCOL=at

MQTT_HOST=192.168.178.128
MQTT_PORT=1883
MQTT_USERNAME=xxx
MQTT_PASSWORD=xxx
MQTT_TOPIC_PREFIX=deye2
DEYE_METRIC_GROUPS=deye_hybrid,deye_hybrid_battery

kbialek, 2024-01-24:
hi @pfostenberg 

I implemented a change that splits register ranges longer than 0x40 into smaller chunks. Let's see if this will let the logger to respond on time. Here is a beta image containing this change. Try it out `ghcr.io/kbialek/deye-inverter-mqtt:2024.01.2-beta-1`

pfostenberg, 2024-01-24:
Hi,
first of all ... thanx for the response - and so quick -:) more thanx... 

looks a litte bit better.... first call works...next timeout...

Any idear?
Split smaller?
Delay?
or is there a register that kill all?
how can I help?

2024-01-24 21:36:23,669 - DeyeDaemon - INFO - Reading start
2024-01-24 21:36:23,670 - DeyeDaemon - INFO - Reading registers [metrics group: deye_hybrid_battery, range: 0003-0042]
2024-01-24 21:36:23,672 - DeyeAtConnector - DEBUG - Sending AT command: b'WIFIKIT-214028-READ'
2024-01-24 21:36:23,774 - DeyeAtConnector - DEBUG - Received AT response in 1. attempt: b'192.168.178.150,402A8F39E514,2799381839'
2024-01-24 21:36:23,775 - DeyeAtConnector - DEBUG - Sending AT command: b'+ok'
2024-01-24 21:36:23,876 - DeyeAtConnector - DEBUG - Sending AT command: b'AT+INVDATA=8,010300030040b43a\n'
2024-01-24 21:36:24,979 - DeyeAtConnector - DEBUG - Received AT response in 1. attempt: b'+ok=01\x1003\x1080\x1032\x1033\x1031\x1030\x1032\x1035\x1032\x1031\x1038\x1039\x1000\x1000\x1081\x1002\x1000\x1000\x1015\x1015\x1000\x1000\x1033\x1085\x10C3\x1065\x1000\x1000\x10C3\x1050\x1000\x1000\x1002\x1001\x1000\x1000\x1000\x10FF\x1000\x100A\x1018\x1001\x1018\x1016\x1022\x101C\x1000\x1001\x1013\x1088\x1005\x1014\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x10FF\x1007\x10D0\x1016\x1076\x1000\x1000\x1000\x1000\x1003\x10E8\x10FF\x10FF\x1000\x1000\x1000\x1000\x1000\x1001\x1000\x1000\x1000\x1001\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x10C3\x1050\x1000\x1000\x1000\x1000\x1019\x1064\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1001\x1000\x1001\x1000\x1002\x1000\x10C8\x1000\x10FA\x1000\x1000\x1005\x108F\x1000\x1000\x1001\x1091\x1005\x1050\x102E\x109C\x10\r\n\r\n'
2024-01-24 21:36:24,980 - DeyeAtConnector - DEBUG - Extracted Modbus response 01038032333130323532313839000081020000151500003385c3650000c35000000201000000ff000a18011816221c00011388051400000000000000000000000000ff07d016760000000003e8ffff000000000001000000010000000000000000c35000000000196400000000000000010001000200c800fa0000058f0000019105502e9c
2024-01-24 21:36:24,982 - DeyeAtConnector - DEBUG - Sending AT command: b'AT+Q\n'
2024-01-24 21:36:25,084 - DeyeDaemon - INFO - Reading registers [metrics group: deye_hybrid_battery, range: 0043-0070]
2024-01-24 21:36:25,085 - DeyeAtConnector - DEBUG - Sending AT command: b'WIFIKIT-214028-READ'
2024-01-24 21:36:26,188 - DeyeAtConnector - DEBUG - Connection response timeout
2024-01-24 21:36:27,191 - DeyeAtConnector - DEBUG - Connection response timeout
2024-01-24 21:36:28,192 - DeyeAtConnector - DEBUG - Connection response timeout
2024-01-24 21:36:29,194 - DeyeAtConnector - DEBUG - Connection response timeout
2024-01-24 21:36:30,196 - DeyeAtConnector - DEBUG - Connection response timeout
2024-01-24 21:36:30,197 - DeyeAtConnector - WARNING - Too many connection timeouts
2024-01-24 21:36:30,197 - DeyeAtConnector - DEBUG - Sending AT command: b'+ok'
2024-01-24 21:36:30,299 - DeyeAtConnector - DEBUG - Sending AT command: b'AT+INVDATA=8,01030043002e3402\n'
2024-01-24 21:36:32,403 - DeyeAtConnector - DEBUG - Connection response timeout
2024-01-24 21:36:33,406 - DeyeAtConnector - DEBUG - Connection response timeout

pfostenberg, 2024-01-24:
Hi, some more info...but not a solution...
after reading from 0003 40 bytes... I executed...

pi@bpi-js:~/git/deye-logger-at-cmd$ ./build/main -t 192.168.178.150:48899 -xmb 00430030
2024/01/24 21:49:58 * Connecting :0 -> 192.168.178.150:48899...
2024/01/24 21:50:01 +ok=010360055101910000000800B10098000004D100000000000000000000000000000000000000C9058F000005510000000005F7057E0000000107D0000001A00000055100000A0300000000000000000000000003700029000A00000012000000000000DFD4
2024/01/24 21:50:02

then

pi@bpi-js:~/git/deye-logger-at-cmd$ ./build/main -t 192.168.178.150:48899 -xmb 00430040
2024/01/24 21:50:07 * Connecting :0 -> 192.168.178.150:48899...
2024/01/24 21:50:13 read udp 192.168.178.84:55023->192.168.178.150:48899: i/o timeout

then no read at all.... 

tried a second one -:) I have two in paralell -:)

pi@bpi-js:~/git/deye-logger-at-cmd$ ./build/main -t 192.168.178.151:48899 -xmb 00030040
2024/01/24 21:54:46 * Connecting :0 -> 192.168.178.151:48899...
2024/01/24 21:54:49 +ok=01038032333130313930363433000081020000151500003385C3650000C35000000201000000FF000A18011816342F00011388051400000000000000000000000000FF07D016760000000003E8FFFF000000000001000000010000000000000000C35000000000196400000000000000010001000200C600FA00000583000000760544054F
2024/01/24 21:54:50
pi@bpi-js:~/git/deye-logger-at-cmd$ ./build/main -t 192.168.178.151:48899 -xmb 00030001
2024/01/24 21:54:56 * Connecting :0 -> 192.168.178.151:48899...
2024/01/24 21:55:02 read udp 192.168.178.84:33558->192.168.178.151:48899: i/o timeout

some kind of protection?

maybe someone has an idea.

BR
Andreas

pfostenberg, 2024-01-24:
more testing....
It looks that length 20 or 10 seems to work....

pi@bpi-js:~/git/deye-logger-at-cmd$ ./build/main -t 192.168.178.151:48899 -xmb 00030020
2024/01/24 22:12:14 * Connecting :0 -> 192.168.178.151:48899...
2024/01/24 22:12:17 +ok=01034032333130313930363433000081020000151500003385C3650000C35000000201000000FF000A180118170A3800011388051400000000000000000000000000FFC300
2024/01/24 22:12:18
pi@bpi-js:~/git/deye-logger-at-cmd$ ./build/main -t 192.168.178.151:48899 -xmb 00230020
2024/01/24 22:12:21 * Connecting :0 -> 192.168.178.151:48899...
2024/01/24 22:12:24 +ok=01034007D016760000000003E8FFFF000000000001000000010000000000000000C35000000000196400000000000000010001000200C700FA0000058400000076054594A1
2024/01/24 22:12:25
pi@bpi-js:~/git/deye-logger-at-cmd$ ./build/main -t 192.168.178.151:48899 -xmb 00430020
2024/01/24 22:12:31 * Connecting :0 -> 192.168.178.151:48899...
2024/01/24 22:12:34 +ok=010340054500760000000000BE000E0000054900000000000000000000000000000000000000C70584000005450000000005A805790000000107D00000008B00000545EB53
2024/01/24 22:12:35
pi@bpi-js:~/git/deye-logger-at-cmd$ ./build/main -t 192.168.178.151:48899 -xmb 00530010
2024/01/24 22:12:45 * Connecting :0 -> 192.168.178.151:48899...
2024/01/24 22:12:48 +ok=010320000000C70584000005450000000005A905790000000107D00000008B00000545B794
2024/01/24 22:12:49
pi@bpi-js:~/git/deye-logger-at-cmd$ ./build/main -t 192.168.178.151:48899 -xmb 00630010
2024/01/24 22:12:52 * Connecting :0 -> 192.168.178.151:48899...
2024/01/24 22:12:55 +ok=0103200000000100000000000000000000000000C80012003700000013000000000000644A
2024/01/24 22:12:56

can you make a try  version or add a length in the config... thinks we need that for the newer one...

BR

Andreas

kbialek, 2024-01-25:
There you have it. `ghcr.io/kbialek/deye-inverter-mqtt:2024.01.2-beta-2`
Config variable name: `DEYE_LOGGER_MAX_REG_RANGE_LENGTH`

pfostenberg, 2024-01-25:
Hi,
YES -:) now the deye-sun-5k-sg03lp1 works! 

with:

DEYE_LOGGER_IP_ADDRESS=192.168.178.150
DEYE_LOGGER_PORT=48899
DEYE_LOGGER_PROTOCOL=at
DEYE_LOGGER_MAX_REG_RANGE_LENGTH=16

Now checking the data readed...
then start to read my slave inverter -:) 
Will report when finished.

But I think the fix can be applied.

Thanx... Can I support you work?

log:
2024-01-25 19:12:57,786 - DeyeDaemon - INFO - Reading start
2024-01-25 19:12:57,788 - DeyeDaemon - INFO - Reading registers [metrics group: deye_hybrid_battery, range: 0003-0012]
2024-01-25 19:12:57,789 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 1)
2024-01-25 19:12:57,790 - DeyeAtConnector - DEBUG - Sending AT command: b'WIFIKIT-214028-READ'
2024-01-25 19:12:57,894 - DeyeAtConnector - DEBUG - Received AT response in 1. attempt: b'192.168.178.150,402A8F39E514,2799381839'
2024-01-25 19:12:57,895 - DeyeAtConnector - DEBUG - Sending AT command: b'+ok'
2024-01-25 19:12:57,996 - DeyeAtConnector - DEBUG - Sending AT command: b'AT+INVDATA=8,010300030010b406\n'
2024-01-25 19:12:59,099 - DeyeAtConnector - DEBUG - Received AT response in 1. attempt: b'+ok=01\x1003\x1020\x1032\x1033\x1031\x1030\x1032\x1035\x1032\x1031\x1038\x1039\x1000\x1000\x1081\x1002\x1000\x1000\x1015\x1015\x1000\x1000\x1033\x1085\x10C3\x1065\x1000\x1000\x10C3\x1050\x1000\x1000\x1002\x1001\x100E\x1064\x10\r\n\r\n'
2024-01-25 19:12:59,101 - DeyeAtConnector - DEBUG - Extracted Modbus response 01032032333130323532313839000081020000151500003385c3650000c350000002010e64
2024-01-25 19:12:59,102 - DeyeAtConnector - DEBUG - Sending AT command: b'AT+Q\n'
2024-01-25 19:12:59,204 - DeyeDaemon - INFO - Reading registers [metrics group: deye_hybrid_battery, range: 0013-0022]
2024-01-25 19:12:59,205 - DeyeAtConnector - DEBUG - Sending AT command: b'WIFIKIT-214028-READ'
2024-01-25 19:12:59,306 - DeyeAtConnector - DEBUG - Received AT response in 1. attempt: b'192.168.178.150,402A8F39E514,2799381839'
2024-01-25 19:12:59,306 - DeyeAtConnector - DEBUG - Sending AT command: b'+ok'
2024-01-25 19:12:59,407 - DeyeAtConnector - DEBUG - Sending AT command: b'AT+INVDATA=8,010300130010b5c3\n'
2024-01-25 19:13:00,509 - DeyeAtConnector - DEBUG - Received AT response in 1. attempt: b'+ok=01\x1003\x1020\x1000\x1000\x1000\x10FF\x1000\x100A\x1018\x1001\x1019\x1014\x100A\x100F\x1000\x1001\x1013\x1088\x1005\x1014\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x10FF\x103D\x1070\x10\r\n\r\n'
2024-01-25 19:13:00,510 - DeyeAtConnector - DEBUG - Extracted Modbus response 010320000000ff000a180119140a0f00011388051400000000000000000000000000ff3d70
2024-01-25 19:13:00,511 - DeyeAtConnector - DEBUG - Sending AT command: b'AT+Q\n'
2024-01-25 19:13:00,612 - DeyeDaemon - INFO - Reading registers [metrics group: deye_hybrid_battery, range: 0023-0032]
2024-01-25 19:13:00,613 - DeyeAtConnector - DEBUG - Sending AT command: b'WIFIKIT-214028-READ'
2024-01-25 19:13:00,715 - DeyeAtConnector - DEBUG - Received AT response in 1. attempt: b'192.168.178.150,402A8F39E514,2799381839'
2024-01-25 19:13:00,715 - DeyeAtConnector - DEBUG - Sending AT command: b'+ok'
2024-01-25 19:13:00,817 - DeyeAtConnector - DEBUG - Sending AT command: b'AT+INVDATA=8,010300230010b5cc\n'
2024-01-25 19:13:01,919 - DeyeAtConnector - DEBUG - Received AT response in 1. attempt: b'+ok=01\x1003\x1020\x1007\x10D0\x1016\x1076\x1000\x1000\x1000\x1000\x1003\x10E8\x10FF\x10FF\x1000\x1000\x1000\x1000\x1000\x1001\x1000\x1000\x1000\x1001\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x10C3\x1050\x1083\x1085\x10\r\n\r\n'
2024-01-25 19:13:01,920 - DeyeAtConnector - DEBUG - Extracted Modbus response 01032007d016760000000003e8ffff000000000001000000010000000000000000c3508385
2024-01-25 19:13:01,921 - DeyeAtConnector - DEBUG - Sending AT command: b'AT+Q\n'
2024-01-25 19:13:02,022 - DeyeDaemon - INFO - Reading registers [metrics group: deye_hybrid_battery, range: 0033-0042]
2024-01-25 19:13:02,023 - DeyeAtConnector - DEBUG - Sending AT command: b'WIFIKIT-214028-READ'
2024-01-25 19:13:02,125 - DeyeAtConnector - DEBUG - Received AT response in 1. attempt: b'192.168.178.150,402A8F39E514,2799381839'
2024-01-25 19:13:02,126 - DeyeAtConnector - DEBUG - Sending AT command: b'+ok'
2024-01-25 19:13:02,227 - DeyeAtConnector - DEBUG - Sending AT command: b'AT+INVDATA=8,010300330010b409\n'

pfostenberg, 2024-01-25:
the most values are fine..only the AC volt seems to be wrong..
[image]
but need more time to track down.

Is there a easy way to change the config and run?
tried the local one described in [CONTRIBUTING.md](https://github.com/kbialek/deye-inverter-mqtt/blob/main/CONTRIBUTING.md) but get stuck in.
make py-install-dependencies
looped for 30 minutes... on my PI... then lost the connection...

kbialek, 2024-01-25:
> Thanx... Can I support you work?

Support is always welcome. There is a paypal link at the top of the [README](https://github.com/kbialek/deye-inverter-mqtt/blob/main/README.md)

> the most values are fine..only the AC volt seems to be wrong.. 

New inverter model might require new registers mapping.

> Is there a easy way to change the config and run?
> tried the local one described in [CONTRIBUTING.md](https://github.com/kbialek/deye-inverter-mqtt/blob/main/CONTRIBUTING.md) but get stuck in.
> make py-install-dependencies
> looped for 30 minutes... on my PI... then lost the connection...

Sad to hear that. I develop it on my laptop (intel cpu)
And you can also run it directly from your desktop, even without building a docker image. Just call `make run`. Of course you must put `config.env` file at the root of the project.

closed 2024-01-28 by kbialek