kbialek/deye-inverter-mqtt #156: Help updating deyr-inverter-mqtt

JensHoRi, 2024-03-10:
I have had an older version (1 year old) of deye-inverter-mqtt running (as a service)

After firmware of inverters (three of them) deye doesn't work anymore. believe because of at-protocol.

Now I looked updating deye-inverter-mqtt to new version on ubuntu 22.04.01 LTS.

Actually dey-inverter-mqtt is in /usr/local/share/deye-inverter-mqtt

I looked for an installation-guide, but didn't find it here.

Could it be possible to add a step-by-step installation guide?
What do I need to do? Just download all files to /usr/local/share/deye-inverter-mqtt ? Or start a makefile?

I'm not so familiar with python scripts with or without docker. Sorry.

kbialek, 2024-03-11:
hi @JensHoRi 

If the old version works for you, then to perform an update, it should be enough to replace the python scripts with the latest version from the `main` branch.

Of course you will have to update the configuration variables to switch Modbus protocol to AT. I recommend to read this https://github.com/kbialek/deye-inverter-mqtt/releases/tag/2023.09.1

JensHoRi, 2024-03-11:
Well, I tried. In my version all *.py files are in main folder, while in actual version they are in src.

Let's see if it works when daylight ist back and inverters are running. I put them now all in main folder overwriting the view py scripts.

closed 2024-04-25 by github-actions[bot]