kbialek/deye-inverter-mqtt #160: Add Power Input at GenPort on Microinverter Input use

finfin82, 2024-03-19:
**I want to see what comes from Microinverter/Gen Input Port**
Im using the deye_sg04lp3 and have my genport configured to be used as microinverter input for some small additional solar plants that push AC-Power to this port.

**So it would be cool to see what comes from this solar-satelites**
Im not sure if there is a modbus address, to read it out. if it is so, it should be easy to add.
The Volts + Watts are interesting here. (any data we can get)

I can easily test suggested modbus addresses.

the configuration to use the genport as microinverter input is in position 133
[image]

i didnt find any register that could match with the needed information.

Thank you in advance!
finfin

philipphenkel, 2024-03-21:
I'm reading the GEN port power from register 667, see [deye-sun-12k-sg04lp3.yaml](https://github.com/philipphenkel/esphome-config/blob/main/devices/deye-sun-12k-sg04lp3.yaml)

- platform: modbus_controller
   modbus_controller_id: ${modbus_controller_id}
   name: "${entity_name_prefix}Gen Power"
   register_type: holding
   address: 667
   unit_of_measurement: "W"
   accuracy_decimals: 0
   device_class: power
   state_class: "measurement"
   value_type: S_WORD

finfin82, 2024-03-22:
Pretty Cool - maybe we just can add this to the main yaml so its available for everyone or guess an additional yaml would be helpful for different gen-port use-cases?

cheers andy

kbialek, 2024-03-26:
A new metric group `deye_sg04lp3_gen` it is. Are you gonna come up with a PR? @finfin82

finfin82, 2024-03-26:
Well I can give it a try, my fork is working for me, I'm not home at python... Some connections are still confusing me. Guess you would be way more faster... But there are other issues too...
May be I will be faster...

You doing such a nice job man!

So it's your decision...

Hope I will meet your guidelines

Greetings Andy

kbialek, 2024-03-26:
No worries. I'm asking for the sake of clarity. Is there anything more I need to know besides of the information provided in https://github.com/kbialek/deye-inverter-mqtt/issues/160#issuecomment-2013777350 ?

finfin82, 2024-03-26:
Well guess, power volt etc. From the gen port would be great. Perfect would be, if you can find the KWh generated via gen port too, would be interesting how much came from the gen port. The inverter KWh includes the gen port as far as I know.

finfin82, 2024-03-27:
@kbialek as you can see here: https://github.com/kbialek/deye-inverter-mqtt/compare/main...finfin82:deye-inverter-mqtt:add-gen-powers
i added volt and power so when we add a new metric group we can catch as much as possible?

kbialek, 2024-04-01:
Thanks for your proposal @finfin82 

I cannot use it directly, because the sensors implementation for this inverter model is generated automatically out of yaml descriptors. Meanwhile, I found the documentation of generator port registers. So, here is a beta image with the new metric group `deye_sg04lp3_generator` added. Try it out `ghcr.io/kbialek/deye-inverter-mqtt:2024.04.1-beta-1`

The documentation is here https://github.com/kbialek/deye-inverter-mqtt/blob/feature/sg04lp3-gen-metric-group/docs/metric_group_deye_sg04lp3_generator.md

finfin82, 2024-04-01:
hey @kbialek wow seems to working perfect, no sun atm, but looks like it should work properly. can confirm 100% tomorrow.
there does not seem to be an address for kwh dayly from gen-port only?!
didnt found one too...

guess /ac/generator/* could match more the actual topic convention?!

greetings

kbialek, 2024-04-07:
> guess /ac/generator/* could match more the actual topic convention?!

yup, will do

kbialek, 2024-04-10:
New beta build is ready for testing `ghcr.io/kbialek/deye-inverter-mqtt:2024.04.1-beta-2`

kbialek, 2024-04-11:
> well it would be nice to catch the kWh generated by genPort too

What do you mean? It's there. https://github.com/kbialek/deye-inverter-mqtt/blob/release/2024.04.1/docs/metric_group_deye_sg04lp3_generator.md

finfin82, 2024-04-11:
Sure but it's the actual power (w) not the daily generated (KWh) . This is the last value why I have to call the Solarman platform parallel.

Greetings Andy

kbialek, 2024-04-11:
Ah, you meant the energy, sorry, I didn't read carefully. Yeah, you'd have to calculate it on your own.

philipphenkel, 2024-04-11:
The daily generator production (energy) is available at address 536.

  - platform: modbus_controller
    modbus_controller_id: ${modbus_controller_id}
    name: "${entity_name_prefix}Daily Generator Production"
    register_type: holding
    address: 536
    unit_of_measurement: "kWh"
    state_class: "total_increasing"
    device_class: energy
    accuracy_decimals: 1
    filters:
      - multiply: 0.1
    value_type: U_WORD

finfin82, 2024-04-12:
Hey thanks! @kbialek let's try this... I will try to read it manually later today and confirm the source...
This would be awesome

finfin82, 2024-04-12:
bingo, thats it! just needs to be multiplied with 0,1 as @philipphenkel metioned.
Thanks to all!
in our paper its called generator working hours, guess it got lost in translation.

Greetings Andy

kbialek, 2024-04-13:
`ac/generator/daily_energy` metric is added in `ghcr.io/kbialek/deye-inverter-mqtt:2024.04.1-beta-4`

finfin82, 2024-04-13:
working perfect, think i cant drop solarman polling now ;-)

kbialek, 2024-04-14:
> working perfect, think i cant drop solarman polling now ;-)

Sounds great!. As usual I kindly encourage you to support my work and to buy me a coffee :smile:

closed 2024-04-14 by kbialek