kbialek/deye-inverter-mqtt #162: Support multiple batteries if possible

c0ldtech, 2024-04-09:
Currently only one batterie is supported, i.e. you only get the overall values if you have multiple batteries. 

For example:

Battery 1 SOC is 20%
Battery 2 SOC is 40%

The value shown for battery SOC will be 30%.

In the Solman app it's possible to view the actual values for every single battery.

I don't know if the values per battery are even exposed through modbus, so it might not be possible to actually get the values.
If testers are needed to try stuff out, I'd be happy to help!

Thanks for all the work so far!

kbialek, 2024-04-09:
What is your inverter model?

c0ldtech, 2024-04-09:
Sorry, completely forgot to mention this. It's a Deye SUN 10K-SG04LP3 with two Deye SE-G5.1 Pro batteries.

kbialek, 2024-04-09:
Try out metric group [deye_sg01hp3_battery](https://github.com/kbialek/deye-inverter-mqtt/blob/main/docs/metric_group_deye_sg01hp3_battery.md) instead of [deye_sg04lp3_battery](https://github.com/kbialek/deye-inverter-mqtt/blob/main/docs/metric_group_deye_sg04lp3_battery.md). SG01HP3 and SG04LP3 have similar register layout.

c0ldtech, 2024-04-09:
Unfortunately, that doesn't seem to have worked. The values that were shown as overall values for both batteries under "battery" are now shown under "battery/1" (and power and voltage are wrong by a factor of 10), values for "battery/2" are all 0.

kbialek, 2024-04-15:
I found this, maybe it will work. Try to read the registers using the command documented here https://github.com/kbialek/deye-inverter-mqtt?tab=readme-ov-file#reading-and-writing-raw-register-values

[image]
[image]

c0ldtech, 2024-04-15:
So I should do "r 600" to get the value for Module Voltage? I do get values, but none of the registers I tried make sense, the values do not match with anything expected. But it's good to know that I can try some stuff out this way, I'll try to do some trial and error if I find the time.

kbialek, 2024-04-15:
> So I should do "r 600" to get the value for Module Voltage?

Yes, exactly

In the doc I have, the 1st battery pack address is 600, the 2nd pack address is 614, the 3rd starts at 628, and so on, up until pack 15 starting at 800. Good luck!

kbialek, 2024-04-15:
Perhaps your batteries are mapped as packs other than 1 and 2? How is the solarman app showing the metrics?

c0ldtech, 2024-04-16:
I already tried the values for pack 15, but basically everything has a value of 0 there. 

I'm currently inclined to think that either they do not have the packs exposed through modbus, or it is just not supported for my battery configuration (even though I am using Deye batteries).

closed 2024-05-31 by github-actions[bot]