kbialek/deye-inverter-mqtt #171: Inverter Frequency is more detailed as read 0.01 instead of 0.1

JensHoRi, 2024-05-03:
Reading the registers directly I saw, Frequency is shown as 49.89 Hz. 

deye-inverter-mqtt only shows 49.9 Hz.

Maybe row in [deye-inverter-mqtt](https://github.com/kbialek/deye-inverter-mqtt/tree/main)/[src](https://github.com/kbialek/deye-inverter-mqtt/tree/main/src/deye_sensor.py needs to be changed to 0.01f in row print_format?

I'm no programmer, so here as in idea instead of send a pull request maybe causes an error

kbialek, 2024-05-05:
Yes, I can confirm that. I own string and micro inverters, and they both report frequency with scale of 2. However, later the value is sent over MQTT using the `0.1f` print format, which is the default for all metrics. This conversion is lossy and introduces accuracy error, but I do not think it needs fixing. Or do you have a use case which requires higher accuracy?

JensHoRi, 2024-05-05:
That’s the reason I put it on enhancement ;)

closed 2024-06-19 by github-actions[bot]