kbialek/deye-inverter-mqtt #173: Support for Python 3.12

CarstenGrohmann, 2024-05-31:
Hello,

ArchLinux replaced Python 3.11 with 3.12.

Please update [requirements.txt](https://github.com/kbialek/deye-inverter-mqtt/blob/main/requirements.txt) and [requirements-dev.txt](https://github.com/kbialek/deye-inverter-mqtt/blob/main/requirements-dev.txt) to support developing with Python 3.12.

I had to switch to PyYAML 6.0.1 to install all required packages:

```
--- a/requirements-dev.txt
+++ b/requirements-dev.txt
@@ -479,7 +479,8 @@ pytest-mock==3.10.0 ; python_version >= "3.10" and python_version < "4.0" \
 pytest==7.3.1 ; python_version >= "3.10" and python_version < "4.0" \
     --hash=sha256:3799fa815351fea3a5e96ac7e503a96fa51cc9942c3753cda7651b93c1cfa362 \
     --hash=sha256:434afafd78b1d78ed0addf160ad2b77a30d35d4bdf8af234fe621919d9ed15e3
-pyyaml==6.0 ; python_version >= "3.10" and python_version < "4.0" \
+pyyaml==6.0.1 ; python_version >= "3.10" and python_version < "4.0" \
+    --hash=sha256:bfdf460b1736c775f2ba9f6a92bca30bc2095067b8a9d77876d1fad6cc3b4a43 \
     --hash=sha256:01b45c0191e6d66c470b6cf1b9531a771a83c1c4208272ead47a3ae4f2f603bf \
     --hash=sha256:0283c35a6a9fbf047493e3a0ce8d79ef5030852c51e9d911a27badfde0605293 \
     --hash=sha256:055d937d65826939cb044fc8c9b08889e8c743fdc6a32b33e2390f66013e449b \
```

Thank you,
Carsten

kbialek, 2024-06-02:
hi @CarstenGrohmann 

I understand, that from your point of view, it would be convenient if I bump python version to 3.12, which is shipped with your OS. 

However, if I look from my own perspective I can say that:
1. The software runs well on python 3.10
2. Python 3.10 EOL is set to 2026-10

Actually, that was one of the reasons, why I chose to run this tool using docker. It makes it independent from OS libs, and that prevents a lot of compatibility issues from appearing.

If you wish to develop plugins, use python env, which is imo, the right way to go. This project uses a combination of pyenv and poetry, which I recommend.

Python version that's shipped with the OS, is good for OS tools, and that's it.

Cheers

CarstenGrohmann, 2024-06-02:
Hi @kbialek,

Thank you for the detailed answer. I'm also fine with Python 3.10 and will rebuild the virtual environment with it.

What do you think about adding a small hint to the [README.md](https://github.com/kbialek/deye-inverter-mqtt/blob/main/README.md)? and adapting the requirements files?

https://github.com/kbialek/deye-inverter-mqtt/blob/bc41f2af9e27f33a2747bb158c071da081b100d8/requirements-dev.txt#L1

Regards,
Carsten

kbialek, 2024-06-02:
> What do you think about adding a small hint to the [README.md](https://github.com/kbialek/deye-inverter-mqtt/blob/main/README.md)

Please check CONTRIBUTING.md

CarstenGrohmann, 2024-06-02:
Thank you for your support,
Carsten

closed 2024-06-02 by CarstenGrohmann