kbialek/deye-inverter-mqtt #175: Problem setting TimeOfUse Time

finfin82, 2024-06-08:
Im writing a script, that analyses tibber prices  and trys to set timeofuse wiht the outcome.
While doing so im getting stuck setting times:
Here are the rules im trying to set (i limited to set time at this point)
Rule#1 unload 00 oClock for 9 hours
Price: 0.2667 Price Difference: 0.0688
deye/timeofuse/time/1/command 0000-0900
deye/timeofuse/voltage/1/command 49.2
deye/timeofuse/power/1/command 6100
deye/timeofuse/enabled/1/command 0

Rule#2 keep 09 oClock for 4 hours
Price: 0.2043 Price Difference: 0.1312
deye/timeofuse/time/2/command 0900-1300
deye/timeofuse/voltage/2/command 49.2
deye/timeofuse/power/2/command 50
deye/timeofuse/enabled/2/command 0

Rule#3 load 13 oClock for 3 hours
Price: 0.1919 Price Difference: 0.1436
deye/timeofuse/time/3/command 1300-1600
deye/timeofuse/voltage/3/command 55.6
deye/timeofuse/power/3/command 50
deye/timeofuse/enabled/3/command 1

Rule#4 keep 16 oClock for 2 hours
Price: 0.2165 Price Difference: 0.119
deye/timeofuse/time/4/command 1600-1800
deye/timeofuse/voltage/4/command 49.2
deye/timeofuse/power/4/command 50
deye/timeofuse/enabled/4/command 0

Rule#5 unload 18 oClock for 6 hours
Price: 0.3102 Price Difference: 0.0253
deye/timeofuse/time/5/command 1800-0000
deye/timeofuse/voltage/5/command 49.2
deye/timeofuse/power/5/command 6100
deye/timeofuse/enabled/5/command 0

These are my config settings:
DEYE_METRIC_GROUPS=deye_sg04lp3,deye_sg04lp3_ups,deye_sg04lp3_battery,deye_sg04lp3_timeofuse,deye_sg04lp3_generator
DEYE_FEATURE_MQTT_PUBLISHER=true
DEYE_FEATURE_SET_TIME=false
DEYE_FEATURE_TIME_OF_USE=true

so my inverter is the deye 12klp3

here is the logging output:

[image]

let me know if you need any aditional information?!
Greetings Andy

kbialek, 2024-06-08:
hi @finfin82 

You should send the start-time instead of time-range. For example, send `900` instead of `0900-1300`

finfin82, 2024-06-08:
well did you change that?
guess it was different in the beginning?!
The end will be defined by the next time-setting?
thank you very much!

finfin82, 2024-06-08:
yep, sorry for my blindness :-)
missunderstood the documentation there...
My code seems to work also... some tiny edge-cases pretty hacky but looking good for the winter :-)
thank you very much!

kbialek, 2024-06-08:
Cool :slightly_smiling_face:

closed 2024-06-08 by kbialek