kbialek/deye-inverter-mqtt #178: After Deye update to Version 2.27, I got error messages like DeyeModbus - ERROR - Response frame has invalid starting byte

PoseidonFDW, 2024-06-17:
docker run -d --name deye-mqtt-1 \
--network="nas-net" \
--ip="172.18.0.107" \
--env "DEYE_LOGGER_IP_ADDRESS=192.168.178.59" \
--env "DEYE_LOGGER_PORT=8899" \
--env "DEYE_LOGGER_SERIAL_NUMBER=4146xxx" \
--env "MQTT_HOST=192.168.178.11" \
--env "MQTT_PORT=1883" \
--env "MQTT_TOPIC_PREFIX=deye1" \
--env "MQTT_TLS_ENABLED=false" \
--env "LOG_LEVEL=WARNING" \
--env "DEYE_DATA_READ_INTERVAL=60" \
--env "DEYE_METRIC_GROUPS=string" \
ghcr.io/kbialek/deye-inverter-mqtt:latest

kbialek, 2024-06-17:
Read https://github.com/kbialek/deye-inverter-mqtt?tab=readme-ov-file#installation It tells what to do with v2 firmware

PoseidonFDW, 2024-06-17:
I use the firmware "MW3_16U_5406_2.31.bin", and the "deye-inverter-mqtt:latest" does not write any Protocol after start.
loglevel is DEBUG.

PoseidonFDW, 2024-06-17:
My last config:
docker run -d --name deye-mqtt \
--network="nas-net" \
--ip="172.18.0.107" \
--env "DEYE_LOGGER_IP_ADDRESS=192.168.178.59" \
--env "DEYE_LOGGER_PROTOCOL=at" \
--env "DEYE_LOGGER_PORT=48899" \
--env "DEYE_LOGGER_SERIAL_NUMBER=4146117309" \
--env "MQTT_HOST=192.168.178.11" \
--env "MQTT_PORT=1883" \
--env "MQTT_TOPIC_PREFIX=deye1" \
--env "MQTT_TLS_ENABLED=false" \
--env "LOG_LEVEL=WARNING" \
--env "DEYE_DATA_READ_INTERVAL=60" \
--env "DEYE_METRIC_GROUPS=string" \
ghcr.io/kbialek/deye-inverter-mqtt:latest

kbialek, 2024-06-18:
Show the logs

closed 2024-07-17 by kbialek