kbialek/deye-inverter-mqtt #187: Error. Modbus device address does not match

karolisju, 2024-07-08:
Hello, 
help is appreciated or some hint where to look.

**Hardware :**
 - Inverter model: deye sun-12k-sg04lp3-eu

**Software :**
 - Logger firmware version: LSW3_15_FFFF_1.0.9E
 - deye-inverter-mqtt version: latest

For a couple of days trying to figure out where is a problem, errors: 

1. “DeyeMqttPublisher - ERROR - Unknown MQTT publishing error: Message publish failed: The client is not currently connected.” (randomly) 
2. “DeyeModbusTcp - ERROR - Modbus device address does not match.”
3. “DeyeSetTimeProcessor - WARNING - Failed to set logger time”

I guess the issue/error is related to writing inverter settings. First of all tried the function timeofuse (according to documentation), but there are no errors and nothing changes in the inverter. loglevel=debug

After enabling the feature flag “DEYE_FEATURE_SET_TIME=true” saw errors that Time could not be set and I guess due to Modbus error.
Tried different protocols tcp and at (Inverter metrics receiving over MQTT) 

Config.env

```
DEYE_LOGGER_IP_ADDRESS=10.0.1.54
DEYE_LOGGER_PORT=8899
DEYE_LOGGER_SERIAL_NUMBER=2799835885
DEYE_LOGGER_PROTOCOL=tcp
DEYE_LOGGER_MAX_REG_RANGE_LENGTH=256

MQTT_HOST=10.0.1.200
MQTT_PORT=1883
MQTT_TOPIC_PREFIX=deye

LOG_LEVEL=DEBUG
LOG_STREAM=STDOUT
PLUGINS_DIR=plugins
PLUGINS_ENABLED=deye_plugin_sample,deye_plugin_custom
DEYE_DATA_READ_INTERVAL=60
DEYE_METRIC_GROUPS=deye_sg04lp3,deye_sg04lp3_battery,deye_sg04lp3_ups,deye_sg04lp3_timeofuse,deye_sg04lp3_generator,settings

DEYE_FEATURE_MQTT_PUBLISHER=true
DEYE_FEATURE_TIME_OF_USE=true
DEYE_FEATURE_ACTIVE_POWER_REGULATION=true
DEYE_FEATURE_SET_TIME=true
```

Logs:

```

deye-inverter-mqtt-py3.12) (base) @Karolis-MacBook-Air deye-original % sh local-run.sh
[local-run] Using bash
2024-07-08 22:01:00,090 - DeyeDaemon - INFO - Please help me build the list of compatible inverters. https://github.com/kbialek/deye-inverter-mqtt/issues/41
2024-07-08 22:01:00,093 - DeyePluginLoader - INFO - Loading plugin: 'deye_plugin_sample'
2024-07-08 22:01:00,093 - DeyeConnectorFactory - INFO - Creating Modbus/TCP Logger connector
2024-07-08 22:01:00,093 - DeyeProcessorFactory - INFO - Feature "Report metrics over MQTT": enabled
2024-07-08 22:01:00,093 - DeyeProcessorFactory - INFO - Feature "Set inverter time once online": enabled
2024-07-08 22:01:00,093 - DeyeProcessorFactory - INFO - Feature "Time-of-use configuration over MQTT": enabled
2024-07-08 22:01:00,093 - DeyeProcessorFactory - INFO - Feature "Active power regulation over MQTT": enabled
2024-07-08 22:01:00,093 - DeyePluginLoader - INFO - Loading custom event processor: 'sample_publisher'
2024-07-08 22:01:00,105 - paho.mqtt.client - DEBUG - Sending CONNECT (u0, p0, wr1, wq1, wf1, c1, k60) client_id=b'deye-inverter-2799835885'
2024-07-08 22:01:00,109 - paho.mqtt.client - DEBUG - Received CONNACK (0, 0)
2024-07-08 22:01:01,111 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r1, m1), 'b'deye/status'', ... (6 bytes)
2024-07-08 22:01:01,111 - DeyeMqttClient - INFO - Successfully connected to MQTT Broker located at 10.0.1.200:1883
2024-07-08 22:01:01,111 - DeyeMqttClient - INFO - Subscribing to topic: deye/timeofuse/selling/command
2024-07-08 22:01:01,111 - paho.mqtt.client - DEBUG - Sending SUBSCRIBE (d0, m2) [(b'deye/timeofuse/selling/command', 1)]
2024-07-08 22:01:01,111 - DeyeMqttClient - INFO - Subscribing to topic: deye/timeofuse/time/1/command
2024-07-08 22:01:01,111 - paho.mqtt.client - DEBUG - Sending SUBSCRIBE (d0, m3) [(b'deye/timeofuse/time/1/command', 1)]
2024-07-08 22:01:01,112 - DeyeMqttClient - INFO - Subscribing to topic: deye/timeofuse/time/2/command
[... 306 lines cut ...]
{'name': 'timeofuse/enabled/1', 'value': 0}
{'name': 'timeofuse/enabled/2', 'value': 0}
{'name': 'timeofuse/enabled/3', 'value': 0}
{'name': 'timeofuse/enabled/4', 'value': 0}
{'name': 'timeofuse/enabled/5', 'value': 0}
{'name': 'timeofuse/enabled/6', 'value': 0}
{'name': 'ac/generator/a/voltage', 'value': 0.8}
{'name': 'ac/generator/b/voltage', 'value': 0.1}
{'name': 'ac/generator/c/voltage', 'value': 0.4}
{'name': 'ac/generator/a/power', 'value': 0}
{'name': 'ac/generator/b/power', 'value': 0}
{'name': 'ac/generator/c/power', 'value': 0}
{'name': 'ac/generator/total_power', 'value': 0}
{'name': 'ac/generator/daily_energy', 'value': 0.0}
{'name': 'settings/active_power_regulation', 'value': 0.0}
2024-07-08 22:01:12,610 - DeyeInverterState - INFO - Reading completed
2024-07-08 22:01:12,799 - paho.mqtt.client - DEBUG - Sending CONNECT (u0, p0, wr1, wq1, wf1, c1, k60) client_id=b'deye-inverter-2799835885'
2024-07-08 22:01:12,804 - paho.mqtt.client - DEBUG - Received CONNACK (0, 0)
2024-07-08 22:01:14,852 - paho.mqtt.client - DEBUG - Sending CONNECT (u0, p0, wr1, wq1, wf1, c1, k60) client_id=b'deye-inverter-2799835885'
2024-07-08 22:01:14,857 - paho.mqtt.client - DEBUG - Received CONNACK (0, 0)
2024-07-08 22:01:16,898 - paho.mqtt.client - DEBUG - Sending CONNECT (u0, p0, wr1, wq1, wf1, c1, k60) client_id=b'deye-inverter-2799835885'
2024-07-08 22:01:16,903 - paho.mqtt.client - DEBUG - Received CONNACK (0, 0)
^Z

```

kbialek, 2024-07-09:
Can it be that you have some wifi networking issues? The erros are reported for both MQTT and Modbus, but these connections are unrelated with each other.

karolisju, 2024-07-09:
Thank you for replay.

I did another test. started ping command from PC where debugging, to Inverter and to MQTT broker.  Launched local-run.sh (this time haven't got DeyeMqttPublisher error)

- MQTT broker response time 3-5ms (stable)
- Inverter response time 3-12ms (sometimes getting request timeout, it could be about 1%)

so it makes sense why randomly getting DeyeMqttPublisher - ERROR, maybe it is possible to increase timeout or retry?

and any clue about this “DeyeModbusTcp - ERROR - Modbus device address does not match.”? (this error always repeats)

kbialek, 2024-07-17:
> and any clue about this “DeyeModbusTcp - ERROR - Modbus device address does not match.”? (this error always repeats)

It indicates that the Modbus response contains unexpected values. Perhaps your inverter model uses different Modbus registers for date/time data?

closed 2024-08-31 by github-actions[bot]