kbialek/deye-inverter-mqtt #189: day_energy and total_energy reports 0

CarstenGrohmann, 2024-07-29:
**Describe the bug**

I use your software for a few months and it runs stable. 

Unfortunatelly it reports unexpected 0s in the morning and the evening e.g. for day_enery and total_energy.

Both zeros are illogical for the total_energy, as the total_energy is a steadily increasing value that should not be 0 except at the very beginning.

The situation is similar with day_energy. Zero values are only expected in the morning after the time reset. In the evening, after the values have risen steadily throughout the day, a 0 value is illogical and unexpected.

I prefer to send no values rather than a series of zeros.

In addition, the 0 for day_energy in the evening breaks the calculation of my self-used solar energy as provided by Home Assistant.

What do you think about this behaviour?

**Expected behavior**
Continuous line that is interrupted overnight because the inverter and logger are offline overnight.
The line does not go down to 0 either in the morning or in the evening.

**Screenshots**
[image]

[image]

**Hardware (please complete the following information):**
 - Inverter model: Deye SUN300G3 EU 230

**Software (please complete the following information):**
 - Logger firmware version: 1.57
 - deye-inverter-mqtt version: [2024.06.2](https://github.com/kbialek/deye-inverter-mqtt/releases/tag/2024.06.2)

### Logs
**Logger is online for the first time this day:**

```
2024-07-20 03:28:00,909 - DeyeDaemon - DEBUG - Invoking action
2024-07-20 03:28:00,910 - DeyeInverterState - INFO - Reading start
2024-07-20 03:28:00,910 - DeyeInverterState - INFO - Reading registers [metrics group: {'micro'}, range: 003c-0074]
2024-07-20 03:28:00,939 - DeyeTcpConnector - INFO - Re-connected to socket on IP 192.168.178.42
2024-07-20 03:28:00,939 - DeyeTcpConnector - DEBUG - Request frame: a517001045000029acebf60200000000000000000000000000000103003c003945d4b615
2024-07-20 03:28:01,114 - DeyeTcpConnector - DEBUG - Received response frame in 1. attempt: a585001015000129acebf60201c07a2f012d0000000000000001037200000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000078ab9415
2024-07-20 03:28:01,116 - DeyeInverterState - INFO - Reading registers [metrics group: {'settings_micro', 'settings'}, range: 0028-0028]
2024-07-20 03:28:01,120 - DeyeTcpConnector - DEBUG - Request frame: a517001045000029acebf602000000000000000000000000000001030028000104025715
2024-07-20 03:28:01,175 - DeyeTcpConnector - DEBUG - Received response frame in 1. attempt: a515001015000229acebf60201c07a2f012d000000000000000103020064b9af5e15
2024-07-20 03:28:01,177 - DeyeInverterState - DEBUG - Production today: 0.0
2024-07-20 03:28:01,177 - DeyeInverterState - DEBUG - Production Total: 0.0
2024-07-20 03:28:01,178 - DeyeInverterState - DEBUG - Phase1 Voltage: 0.0
2024-07-20 03:28:01,179 - DeyeInverterState - DEBUG - Phase1 Current: 0.0
2024-07-20 03:28:01,180 - DeyeInverterState - DEBUG - Phase1 Power: 0.0
2024-07-20 03:28:01,181 - DeyeInverterState - DEBUG - AC Freq: 0.0
2024-07-20 03:28:01,181 - DeyeInverterState - DEBUG - Uptime: 0.0
2024-07-20 03:28:01,181 - DeyeInverterState - DEBUG - PV1 Voltage: 0.0
2024-07-20 03:28:01,182 - DeyeInverterState - DEBUG - PV1 Current: 0.0
2024-07-20 03:28:01,182 - DeyeInverterState - DEBUG - PV1 Power: 0.0
2024-07-20 03:28:01,183 - DeyeInverterState - DEBUG - PV1 Production today: 0.0
2024-07-20 03:28:01,183 - DeyeInverterState - DEBUG - PV1 Total: 0.0
2024-07-20 03:28:01,183 - DeyeInverterState - DEBUG - PV2 Voltage: 0.0
2024-07-20 03:28:01,184 - DeyeInverterState - DEBUG - PV2 Current: 0.0
2024-07-20 03:28:01,184 - DeyeInverterState - DEBUG - PV2 Power: 0.0
2024-07-20 03:28:01,185 - DeyeInverterState - DEBUG - PV2 Production today: 0.0
2024-07-20 03:28:01,185 - DeyeInverterState - DEBUG - PV2 Total: 0.0
2024-07-20 03:28:01,185 - DeyeInverterState - DEBUG - PV3 Voltage: 0.0
2024-07-20 03:28:01,186 - DeyeInverterState - DEBUG - PV3 Current: 0.0
2024-07-20 03:28:01,186 - DeyeInverterState - DEBUG - PV3 Power: 0.0
2024-07-20 03:28:01,186 - DeyeInverterState - DEBUG - PV3 Production today: 0.0
2024-07-20 03:28:01,187 - DeyeInverterState - DEBUG - PV3 Total: 0.0
2024-07-20 03:28:01,187 - DeyeInverterState - DEBUG - PV4 Voltage: 0.0
2024-07-20 03:28:01,187 - DeyeInverterState - DEBUG - PV4 Current: 0.0
2024-07-20 03:28:01,188 - DeyeInverterState - DEBUG - PV4 Power: 0.0
[... 213 lines cut ...]
2024-07-20 18:50:01,531 - DeyeInverterState - DEBUG - PV3 Current: 0.0
2024-07-20 18:50:01,531 - DeyeInverterState - DEBUG - PV3 Power: 0.0
2024-07-20 18:50:01,532 - DeyeInverterState - DEBUG - PV3 Production today: 0.0
2024-07-20 18:50:01,532 - DeyeInverterState - DEBUG - PV3 Total: 0.0
2024-07-20 18:50:01,532 - DeyeInverterState - DEBUG - PV4 Voltage: 0.0
2024-07-20 18:50:01,533 - DeyeInverterState - DEBUG - PV4 Current: 0.0
2024-07-20 18:50:01,533 - DeyeInverterState - DEBUG - PV4 Power: 0.0
2024-07-20 18:50:01,533 - DeyeInverterState - DEBUG - PV4 Production today: 0.0
2024-07-20 18:50:01,533 - DeyeInverterState - DEBUG - PV4 Total: 0.0
2024-07-20 18:50:01,534 - DeyeInverterState - DEBUG - DC Total Power: 0.0
2024-07-20 18:50:01,534 - DeyeInverterState - DEBUG - Operating Power: 0.0
2024-07-20 18:50:01,534 - DeyeInverterState - DEBUG - AC Active Power: 0.0
2024-07-20 18:50:01,535 - DeyeInverterState - DEBUG - Radiator temperature: -10.0
2024-07-20 18:50:01,535 - DeyeInverterState - DEBUG - Active power regulation: 100.0
2024-07-20 18:50:01,536 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deyemqtt/logger_status', value: 'online'
2024-07-20 18:50:01,536 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m63124), 'b'deyemqtt/logger_status'', ... (6 bytes)
2024-07-20 18:50:01,538 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 63124)
2024-07-20 18:50:01,541 - DeyeMqttClient - INFO - Logger is online
2024-07-20 18:50:01,541 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deyemqtt/day_energy', value: '0.0'
2024-07-20 18:50:01,542 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m63125), 'b'deyemqtt/day_energy'', ... (3 bytes)
2024-07-20 18:50:01,544 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 63125)
2024-07-20 18:50:01,547 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deyemqtt/total_energy', value: '0.0'
2024-07-20 18:50:01,549 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m63126), 'b'deyemqtt/total_energy'', ... (3 bytes)
2024-07-20 18:50:01,551 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 63126)
```

kbialek, 2024-08-01:
@CarstenGrohmann hi! This software only publishes the values read from the inverter. Your inverter returns zeros, and that is what is published.

Let's look at this response frame:
```
a585001015000929acebf602019f7b2f010c0100000000000001037200000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000078ab5c15
```

It contains this ModbusRTU frame inside:
```
01037200000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000078ab
```

Even without decoding, it is obvious that it contains zeros only. It's worth noting that the CRC is valid, which indicates, that the communication is working correctly.

CarstenGrohmann, 2024-08-04:
My converter has no connection to the internet and `DEYE_FEATURE_SET_TIME` is set to `true`.

I'll update the firmware to the newest 2.32 releae and check, if this solves this issue.

nifoc, 2024-08-08:
I'm sorry for what is essentially a `+1`, but I'm noticing the same behavior.

At some point during the day `day_energy` suddenly becomes `0` and then returns to the previous (correct) measurement during the next poll.

Maybe this can be handled so that it simply sends nothing instead of `0` in cases where `0` doesn't make any sense?

kbialek, 2024-08-09:
> Maybe this can be handled so that it simply sends nothing instead of `0` in cases where `0` doesn't make any sense?

The question is how to define "where `0` doesn't make any sense"?

Options:
#### Ignore Modbus response frames having zeros only
* Pros: Easy to implement
* Cons: Prone to false positives (zeroed frame may be valid in some situations)
#### Publish measurement having value equal zero only when zero is read two times in a row or more.
* Pros: More fine grained control over the reported values.
* Cons: Risk of losing legitimate zero measurement.

What do you think? Or maybe you have other proposal?

This feature would have to be put behind a feature flag, and be disabled by default.

CarstenGrohmann, 2024-08-15:
I had trouble with the newest firmware and rolled back to MW3_16U_5406_1.63. The firmware 1.63 reports the same zero frames as reported above.

Therefore, I would prefer the first option as it prevents data from being pushed like in the "Logger is online for the first time this day" and "First zero value in the evening" sections.

@nifoc: In such a case, are all your values 0 or only day_energy?

CarstenGrohmann, 2024-08-31:
Short update as I tested a small hack to filter 0 frames based on our discussion.

I added this temporary workaround to my docker container:

```
diff --git a/src/deye_modbus.py b/src/deye_modbus.py
index 1178712..13666d4 100644
--- a/src/deye_modbus.py
+++ b/src/deye_modbus.py
@@ -47,6 +47,9 @@ class DeyeModbus:
         modbus_crc.reverse()
 
         modbus_resp_frame = self.connector.send_request(modbus_frame + modbus_crc)
+        if "0" * 80 in modbus_resp_frame.hex():
+            self.__log.debug("Ignore empty frame: %s", modbus_resp_frame.hex())
+            return {}
         if modbus_resp_frame is None:
             return {}
         return self.__parse_modbus_read_holding_registers_response(modbus_resp_frame, first_reg, last_reg)
```

```
2024-08-31 04:45:50,980 - DeyeModbus - DEBUG - Ignore empty frame: 0103720000000000001292000000000000000000001292000000000000094200000000000000000000138800000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000400000000000000000000012a00000000000000000000000000001517
2024-08-31 04:46:51,252 - DeyeModbus - DEBUG - Ignore empty frame: 0103720000000000001292000000000000000000001292000000000000094200000000000000000000138800000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000400000000000000000000012a00000000000000000000000000001517
```

and in combination with a Utility Meter:

```
utility_meter:
  energy:
    name: "Production today (daily reset)"
    source: sensor.deye_inverter_mqtt_production_today
    cycle: daily
```

Home Assistent generates diagrams without any outliers in the morning or in the afternoon:

[image]

kbialek, 2024-09-03:
As there is no consensus about the best strategy here, my proposal is to implement them one-by-one, as needed, so the end-user can choose the one that suits him best. Yours will be added first @CarstenGrohmann

kosl, 2024-09-05:
Let me note that I am experiencing similar sporadic "spikes" readings from Deye SUN12k SG04LP3 in my [fork](https://github.com/kosl/inverter-connect/tree/ha_services_upgrade), where I observe regular +ok=... response with valid CRC. I do not use block reads but single parameter and the reads are not zero but usually close to. I thought that some delay (1 second) between AT+INVDATA=... send and +ok=... recv would solve the problem of getting correct value(s) to the logger. So, there is some bug in the firmware that deyecloud.com is resolving somewhat better that us with AT+ protocol and I am looking for a root cause of such behaviour.

Bhoft, 2024-09-10:
@CarstenGrohmann 
Hi can you add the HA sensor settings of this sensor?
> sensor.deye_inverter_mqtt_production_today

I have trouble with my daily sensor values which my small inverter  see https://github.com/kbialek/deye-inverter-mqtt/issues/197

CarstenGrohmann, 2024-09-10:
I have no special/manual settings for the `sensor.deye_inverter_mqtt_production_today` sensor as I use my plugin [Deye MQTT HA Plugin](https://sr.ht/~carstengrohmann/deye-mqtt-ha-plugin/) to announce the MQTT values to HA. In addition,  I use several UtilityMeter sensor is used to reset most values at midnight.

Bhoft, 2024-09-11:
Hi Carsten
I saw that you use a Utility Meter for the daily sensor. 
But i guess that doesn't solve my issue that my inverter sends it's last known value (before it got offline) for `day_energy` or your `production_today` via mqtt when it goes back online,
He resets to 0 when the time is set, but sadly he first sends sensor data via mqqt before that.

I just installed your plugin, but that wouldn't change the issue as it only changes the mqtt topics and auto creates the mqtt sensors (which i have done manually before). Nevertheless a plugin to automatically create the sensors is a great benefit.

But as i said an additional utility meter wouldn't change a thing if the sensor which is used has wrong values.
But I will see this tomorrow when the inverter comes back online.

The sensor has wrong values from 0:00 - 7:20 which would be then used for the utility meter as well.
(image of my day_energy sensor)
[image]

Btw do you plan to enable the `active_power_regulation` via mqtt as well?
Btw2 why are you using `state_class:total` and not `total_increasing` for your total / daily sensors?

kbialek, 2024-09-11:
@Bhoft My microinverters behave in the same way. I will think how to solve this in a systemic way.
[image]

Bhoft, 2024-09-11:
@kbialek 
good that i am not the only one 👍 
I just checked your code and if you aggregrate over multiple inverters you are already checking if now isn't the last aggregation date.
https://github.com/kbialek/deye-inverter-mqtt/blob/8a56a28eee905bff2e3b5e0cf9d54b80db76df88/src/deye_multi_inverter_data_aggregator.py#L43-L46

Maybe something like this would also work for single inverter usage?

CarstenGrohmann, 2024-09-11:
> But as i said an additional utility meter wouldn't change a thing if the sensor which is used has wrong values.

The utility meter cannot change the values of the origin sensor. The utility meter creates a new sensor with changed values and you can use the new sensor instead of the original sensor.

[image]

From my perspective the utility meter is a feasible workaround.

CarstenGrohmann, 2024-09-11:
> Btw do you plan to enable the `active_power_regulation` via mqtt as well?

Yes, I'll do after I added support for another set of topics.

kosl, 2024-09-12:
With 3 Utility meters (battery and production) I have been able to correctly restart the sensors at midnight for the Energy card. 

Further, I needed to add several sensor templates to correctly recompute for Sunsynk-Power-Flow-Card
- Deye Inverter MQTT Essential Power (computed)
- Deye Inverter MQTT Internal CT total power (computed)
- Deye Inverter MQTT Inverter Power (computed)
- Deye Inverter MQTT Non Essential Power (computed)
- Deye Inverter MQTT Tesla (computed)
- Solar power (computed)
- SolarEdge Meter1 Power actual (inverted)

[image]

where some missing sensors (e.g. frequency) are taken from another inverter (Solaredge).

Btw, I would love to see R/W registers such as peak shaving (191), limit_control (142), (off) grid state, battery limits,  ... to be implemented for automated (timed)  control of the inverter such as https://github.com/klatremis/esphome-for-deye is!

kbialek, 2024-09-12:
hi!

I'm currently testing this beta image `ghcr.io/kbialek/deye-inverter-mqtt:2024.09.1-beta-2`
It contains these PRs
* https://github.com/kbialek/deye-inverter-mqtt/pull/198
* https://github.com/kbialek/deye-inverter-mqtt/pull/199

I encourage you to give it try

@CarstenGrohmann You will need to add `DEYE_DATA_FILTERS=ignore_zeroed_frames` to the config to activate this functionality

Bhoft, 2024-09-13:
> I'm currently testing this beta image `ghcr.io/kbialek/deye-inverter-mqtt:2024.09.1-beta-2` It contains these PRs

I have installed the beta version yesterday and have an error in the log (multiple times)
```
2024-09-13 05:59:04,549 - DeyeDaemon - ERROR - Unexpected error during runner execution
Traceback (most recent call last):
  File "/opt/deye_inverter_mqtt/deye_daemon.py", line 53, in __invoke_action
    self.__action()
  File "/opt/deye_inverter_mqtt/deye_inverter_state.py", line 60, in read_from_logger
    processor.process(events)
  File "/opt/deye_inverter_mqtt/plugins/deye_plugin_ha_discovery.py", line 237, in process
    self.publish_sensor_information(topic, event.observation)
  File "/opt/deye_inverter_mqtt/plugins/deye_plugin_ha_discovery.py", line 167, in publish_sensor_information
    "unit_of_measurement": observation.sensor.unit,
AttributeError: 'DailyResetSensor' object has no attribute 'unit'
```

Besides that the day energy value is now 0 when the inverter comes online.
```
2024-09-13 05:14:04,299 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/day_energy', value: '0.0'
```
same for the PV1 values
```
2024-09-13 05:14:04,323 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv1/day_energy', value: '0.0'
```

CarstenGrohmann, 2024-09-13:
The AttributeError exception is raised from my code. It looks like the `unit` attribute is missing from `DailyResetSensor` even if the class is derived from `Sensor` class. 

```
2024-09-13 11:08:31,622 - DeyeDaemon - ERROR - Unexpected error during runner execution
Traceback (most recent call last):
  File "/opt/deye_inverter_mqtt/deye_daemon.py", line 53, in __invoke_action
    self.__action()
  File "/opt/deye_inverter_mqtt/deye_inverter_state.py", line 60, in read_from_logger
    processor.process(events)
  File "/opt/deye_inverter_mqtt/plugins/deye_plugin_ha_discovery.py", line 320, in process
    self.publish_sensor_information(topic, event.observation)
  File "/opt/deye_inverter_mqtt/plugins/deye_plugin_ha_discovery.py", line 242, in publish_sensor_information
    "unit_of_measurement": observation.sensor.unit,
AttributeError: 'DailyResetSensor' object has no attribute 'unit'
```

@kbialek: What do you think about this issue?

kbialek, 2024-09-13:
A new beta image `2024.09.1-beta-3` is there
1. I fixed the error reported by @CarstenGrohmann I was actually expecting some errors to appear. The first implementation was quite dirty.
2. I've added a log message `DeyeDaemon - INFO - Active data filters: ignore_zeroed_frames`

CarstenGrohmann, 2024-09-13:
I've updated my docker image and published a development branch with new metric and total_increasing state class for day_energy in https://github.com/CarstenGrohmann/deye-mqtt-ha-plugin/tree/more_inverters.

Today it was raining. I'll monitor all changes during the days next week with more sun.

Thank you for this fast update.

Bhoft, 2024-09-14:
> A new beta image `2024.09.1-beta-3` is there

I have installed that beta and have these results:

- carstens addon is working again (the old version haven't tested the new version yet)
- there is an issue with the "total" values

Issue with total values:

These are the Log lines when my micro inverter came online  today the total values are 0 for the first 3 MQTT publishes.
```
2024-09-14 05:13:50,598 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/total_energy', value: '0.0'
2024-09-14 05:14:48,080 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/total_energy', value: '0.0'
2024-09-14 05:16:48,082 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/total_energy', value: '0.0'
2024-09-14 05:17:48,160 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/total_energy', value: '201.3'
```

Same for other total values like PV1 Total energy
```
2024-09-14 05:13:50,647 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv1/total_energy', value: '0.0'
2024-09-14 05:14:48,105 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv1/total_energy', value: '0.0'
2024-09-14 05:16:48,103 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv1/total_energy', value: '0.0'
2024-09-14 05:17:48,202 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv1/total_energy', value: '44.4'
```

Which results in a wrong value 
[image]

**which was btw already the case in the first beta. I just didn't saw it but can see it in the statistics**

kbialek, 2024-09-14:
> ### Issue with total values:
>
> These are the Log lines when my micro inverter came online today the total values are 0 for the first 3 MQTT publishes.
>
> ```
> 2024-09-14 05:13:50,598 - DeyeMqttClient - [...]

@Bhoft I do not think that this behavior is caused by the changes I added in the beta image. I belive that what you see is caused by zeroed frames returned by the logger "just" right after it has powered on in the morning. You may try to add `DEYE_DATA_FILTERS=ignore_zeroed_frames` config variable to mitigate this.

Bhoft, 2024-09-14:
> @Bhoft I do not think that this behavior is caused by the changes I added in the beta image. I belive that what you see is caused by zeroed frames returned by the logger "just" right after it has [...]

You are right i just extended the history in my statistics and it also occur way before the beta.  I just didn`t saw this because the default range on the history was only the last 2 days and i didn't checked before that time range.

Bhoft, 2024-09-15:
I have added the ignore_zeroed_frames  filter yesterday but still got the Zero value publishes for total values this morning

relevant log entries
```
2024-09-14 13:22:33,170 - DeyeDaemon - INFO - Active data filters: ignore_zeroed_frames
2024-09-15 05:10:45,896 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv1/total_energy', value: '0.0'
2024-09-15 05:11:45,738 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv1/total_energy', value: '0.0'
2024-09-15 05:12:45,747 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv1/total_energy', value: '0.0'
2024-09-15 05:14:45,734 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv1/total_energy', value: '45.4'
```

and register values at that times
```
2024-09-15 05:10:45,447 - DeyeInverterState - INFO - Reading registers [metrics group: {'micro'}, range: 003c-0074]
2024-09-15 05:10:45,455 - DeyeTcpConnector - INFO - Re-connected to socket on IP 192.168.5.149
2024-09-15 05:10:45,456 - DeyeTcpConnector - DEBUG - Request frame: a5170010450000067c63e50200000000000000000000000000000103003c003945d4ca15
2024-09-15 05:10:45,632 - DeyeTcpConnector - DEBUG - Received response frame in 1. attempt: a5850010150003067c63e50201c6e6b52e2f00000090853038010372000000000000000000000000000000000000000000000000000008ca0000000000000000000000000000000000000000000000000000000000000000000008ff08f3138400000000000000000000000002920002001d000000000000000000000000000000000000000000000000000000004e27be15
2024-09-15 05:10:45,632 - DeyeInverterState - INFO - Reading registers [metrics group: {'settings_micro', 'settings'}, range: 0028-0028]
2024-09-15 05:10:45,637 - DeyeTcpConnector - DEBUG - Request frame: a5170010450000067c63e502000000000000000000000000000001030028000104026b15
2024-09-15 05:10:45,821 - DeyeTcpConnector - DEBUG - Received response frame in 1. attempt: a5150010150004067c63e50201c6e6b52e30000000908530380103020064b9af1915

2024-09-15 05:11:45,447 - DeyeInverterState - INFO - Reading registers [metrics group: {'micro'}, range: 003c-0074]
2024-09-15 05:11:45,452 - DeyeTcpConnector - DEBUG - Request frame: a5170010450000067c63e50200000000000000000000000000000103003c003945d4ca15
2024-09-15 05:11:45,637 - DeyeTcpConnector - DEBUG - Received response frame in 1. attempt: a585001015000d067c63e5020102e7b52e6b0000008f853038010372000000000000000000000000000000000000000000000000000008d400000000000000000000000000000000000000000000000000000000000000000000090108f5138100000000000000000000000002920002001d00000000000000000000000000000000000000000000000000000000de883d15
2024-09-15 05:11:45,637 - DeyeInverterState - INFO - Reading registers [metrics group: {'settings_micro', 'settings'}, range: 0028-0028]
2024-09-15 05:11:45,643 - DeyeTcpConnector - DEBUG - Request frame: a5170010450000067c63e502000000000000000000000000000001030028000104026b15
2024-09-15 05:11:45,707 - DeyeTcpConnector - DEBUG - Received response frame in 1. attempt: a515001015000e067c63e5020102e7b52e6c0000008f8530380103020064b9af9b15

2024-09-15 05:12:45,447 - DeyeInverterState - INFO - Reading registers [metrics group: {'micro'}, range: 003c-0074]
2024-09-15 05:12:45,453 - DeyeTcpConnector - DEBUG - Request frame: a5170010450000067c63e50200000000000000000000000000000103003c003945d4ca15
2024-09-15 05:12:45,635 - DeyeTcpConnector - DEBUG - Received response frame in 1. attempt: a585001015000f067c63e502013ee7b52ea70000008f853038010372000000000000000000000000000000000000000000000000000008ca00000000000000000000000000000000000000000000000000000000000000000000090308f7138600000000000000000000000002920002001d00000000000000000000000000000000000000000000000000000000d8f41c15
2024-09-15 05:12:45,635 - DeyeInverterState - INFO - Reading registers [metrics group: {'settings_micro', 'settings'}, range: 0028-0028]
2024-09-15 05:12:45,639 - DeyeTcpConnector - DEBUG - Request frame: a5170010450000067c63e502000000000000000000000000000001030028000104026b15
2024-09-15 05:12:45,695 - DeyeTcpConnector - DEBUG - Received response frame in 1. attempt: a5150010150010067c63e502013ee7b52ea80000008f8530380103020064b9af1515

2024-09-15 05:13:45,447 - DeyeInverterState - INFO - Reading registers [metrics group: {'micro'}, range: 003c-0074]
2024-09-15 05:13:45,452 - DeyeTcpConnector - DEBUG - Request frame: a5170010450000067c63e50200000000000000000000000000000103003c003945d4ca15
2024-09-15 05:13:45,635 - DeyeTcpConnector - DEBUG - Received response frame in 1. attempt: a5850010150012067c63e5020179e7b52ee30000008f853038010372000000000000000000000000000000000000000000000000000008ca00000000000000000000000000000000000000000000000000000000000000000000090108f5138200000000000000000000000002920002001d000000000000000000000000000000000000000000000000000000008a045015
2024-09-15 05:13:45,635 - DeyeInverterState - INFO - Reading registers [metrics group: {'settings_micro', 'settings'}, range: 0028-0028]
2024-09-15 05:13:45,640 - DeyeTcpConnector - DEBUG - Request frame: a5170010450000067c63e502000000000000000000000000000001030028000104026b15
2024-09-15 05:13:46,007 - DeyeTcpConnector - DEBUG - Received response frame in 1. attempt: a5150010150013067c63e502017ae7b52ee40000008f8530380103020064b9af9015

2024-09-15 05:14:45,448 - DeyeInverterState - INFO - Reading registers [metrics group: {'micro'}, range: 003c-0074]
2024-09-15 05:14:45,452 - DeyeTcpConnector - DEBUG - Request frame: a5170010450000067c63e50200000000000000000000000000000103003c003945d4ca15
[... cut ...]
2024-09-15 05:14:45,700 - DeyeTcpConnector - DEBUG - Received response frame in 1. attempt: a5150010150015067c63e50201b6e7b52e200100008f8530380103020064b9af0b15
```

*"Ignoring zeroed frame"* wasn't found in the log files, so i guess the skipping doesn't work in my case?

What i found strange is that i don't see it today in the popup statistics (where it was visible yesterday)
[image]

but in the detailed statistics 
[image]

but i guess it has something to do with the resolution of the timeline or some usage of function like mean.
I just mentioning this because it seems you can't rely on the display of the statistics in the popup.

kbialek, 2024-09-16:
@CarstenGrohmann @Bhoft  I've built a new beta image `2024.09.1-beta-4`

1. I decided to remove https://github.com/kbialek/deye-inverter-mqtt/pull/198 - it's based on a heuristic which does not  work reliably.
2. Added https://github.com/kbialek/deye-inverter-mqtt/pull/200 - values will be published only when total energy sensor value is higher than zero
3. I kept https://github.com/kbialek/deye-inverter-mqtt/pull/199

CarstenGrohmann, 2024-09-17:
I've updated the docker image to beta-4. The last version beta-3 runs without any anormalities:

```
2024-09-16 16:20:38,398 - DeyeModbus - DEBUG - Ignoring zeroed frame: 01037200000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000078ab
2024-09-16 16:24:39,445 - DeyeModbus - DEBUG - Ignoring zeroed frame: 01037200000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000078ab
```

kbialek, 2024-09-17:
@CarstenGrohmann Maybe I should explain it a bit more. Modbus responses shared by @Bhoft contain mostly zeros, but not only zeros, thus zeroed frames ignoring does not work for him. This convinced me that ignoring zeroed frames is only an unreliable weak heuristic. That's why I decided to replace it with "data readiness check". The service will publish the values to MQTT only when "data readiness check" is successful. 
Now what "data readiness check" actually is? It simply checks if "Production Total" value is greater than zero. This check should work for all inverters I belive.

I'm testing beta-4 now. I see that probably I did there something wrong. I will come back to you.

Bhoft, 2024-09-18:
Hi i have tested the beta-4 now for two days and can say that the Data readiness check seems to work.
I have no publishes for the `deye/total_energy` in the mornings with Zero.

e.g. first connect this morning and its publishes
```
2024-09-18 05:16:21,099 - DeyeTcpConnector - INFO - Re-connected to socket on IP 192.168.5.149
2024-09-18 05:16:21,099 - DeyeTcpConnector - DEBUG - Request frame: a5170010450000067c63e50200000000000000000000000000000103003c003945d4ca15
2024-09-18 05:16:21,311 - DeyeTcpConnector - DEBUG - Received response frame in 1. attempt: a5850010150005067c63e50201fbddb72e360100002a843238010372000d00000000084a0000000200020003000201d7000001b9000008de02700000000001eb0000138800000000000000000000000000000000000000000000091409081380000000000000000000000000028f0002001d001000000000000000000000012d0000012e0000012d0000012e00005768d115
2024-09-18 05:16:21,311 - DeyeInverterState - INFO - Reading registers [metrics group: {'settings_micro', 'settings'}, range: 0028-0028]
2024-09-18 05:16:21,321 - DeyeTcpConnector - DEBUG - Request frame: a5170010450000067c63e502000000000000000000000000000001030028000104026b15
2024-09-18 05:16:21,382 - DeyeTcpConnector - DEBUG - Received response frame in 1. attempt: a5150010150006067c63e50201fbddb72e360100002a8432380103020064b9afeb15

2024-09-18 05:16:21,386 - DeyeInverterState - DEBUG - Data readiness observations: [total_energy@2024-09-18 05:16:21.382964:212.20000000000002]
2024-09-18 05:16:21,386 - DeyeInverterState - DEBUG - Data readiness check result: True
2024-09-18 05:16:21,387 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/logger_status', value: 'online'
2024-09-18 05:16:21,392 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/day_energy', value: '1.3'
2024-09-18 05:16:21,398 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/total_energy', value: '212.2'
```
So everthing is fine 👍 

The only strange thing is that the debug message of the readiness check is always true, so even when offline
```
2024-09-18 05:14:27,235 - DeyeTcpConnector - DEBUG - Could not open socket on IP 192.168.5.149: [Errno 113] Host is unreachable
2024-09-18 05:14:27,236 - DeyeInverterState - DEBUG - Data readiness observations: []
2024-09-18 05:14:27,236 - DeyeInverterState - DEBUG - Data readiness check result: True
2024-09-18 05:14:27,236 - DeyeInverterState - DEBUG - No observation events received (offline)
```

I couldn't find a single line in my logs with `Data readiness check result: False` or `Data readiness observations: [total_energy@timestamp:0]`

What i also find a bit strange is I find many entries in the log with `0000000000002`.
```
...
Data readiness observations: [total_energy@2024-09-17 14:50:21.082968:212.0]
Data readiness observations: [total_energy@2024-09-17 14:51:21.089482:212.0]
Data readiness observations: [total_energy@2024-09-17 14:52:21.079735:212.10000000000002]
Data readiness observations: [total_energy@2024-09-17 14:53:21.086853:212.10000000000002]
....100 of times
Data readiness observations: [total_energy@2024-09-18 05:49:21.348573:212.20000000000002]
Data readiness observations: [total_energy@2024-09-18 05:50:21.348877:212.20000000000002]
Data readiness observations: [total_energy@2024-09-18 05:51:21.350037:212.3]
Data readiness observations: [total_energy@2024-09-18 05:52:21.354751:212.3]
Data readiness observations: [total_energy@2024-09-18 05:53:21.351361:212.3]

...
Data readiness observations: [total_energy@2024-09-18 07:23:21.379221:212.5]
Data readiness observations: [total_energy@2024-09-18 07:24:21.377085:212.5]
Data readiness observations: [total_energy@2024-09-18 07:25:21.687514:212.60000000000002]
Data readiness observations: [total_energy@2024-09-18 07:53:21.396904:212.70000000000002]
.. also multiple times
Data readiness observations: [total_energy@2024-09-18 07:54:21.431255:212.70000000000002]
Data readiness observations: [total_energy@2024-09-18 07:55:21.384730:212.8]
Data readiness observations: [total_energy@2024-09-18 07:56:21.436309:212.8]
```

The `0000000000002` seems a bit strange in my opinion. 
There are multiple entries and its always ends with `0000000000002`. 
For me it looks like something isn't read correctly from the registers?

Reading before where no `0000000000002` is added
```
2024-09-18 07:24:21,122 - DeyeInverterState - INFO - Reading registers [metrics group: {'micro'}, range: 003c-0074]
2024-09-18 07:24:21,127 - DeyeTcpConnector - DEBUG - Request frame: a5170010450000067c63e50200000000000000000000000000000103003c003945d4ca15
2024-09-18 07:24:21,305 - DeyeTcpConnector - DEBUG - Received response frame in 1. attempt: a5850010150027067c63e502017bfbb72eab020000aa843238010372000200000000084d0000000000000000000101d8000001b9000008e802700000000e01ed000013880000000000000000000000000cc20000000000000dac091e090f1382000000000000000000000000028f0002001d00000000000000000000000001b4000a01c900080164000801ac00351ccb6a15
2024-09-18 07:24:21,305 - DeyeInverterState - INFO - Reading registers [metrics group: {'settings_micro', 'settings'}, range: 0028-0028]
2024-09-18 07:24:21,313 - DeyeTcpConnector - DEBUG - Request frame: a5170010450000067c63e502000000000000000000000000000001030028000104026b15
2024-09-18 07:24:21,376 - DeyeTcpConnector - DEBUG - Received response frame in 1. attempt: a5150010150028067c63e502017bfbb72eab020000aa8432380103020064b9afa115
2024-09-18 07:24:21,377 - DeyeInverterState - DEBUG - Production today: 0.2
[... cut ...]
2024-09-18 07:25:21,690 - DeyeInverterState - DEBUG - Radiator temperature: 24.4
2024-09-18 07:25:21,690 - DeyeInverterState - DEBUG - Active power regulation: 100.0
2024-09-18 07:25:21,691 - DeyeInverterState - DEBUG - Data readiness observations: [total_energy@2024-09-18 07:25:21.687514:212.60000000000002]
2024-09-18 07:25:21,691 - DeyeInverterState - DEBUG - Data readiness check result: True
```

It's still published as 212.6
```
2024-09-18 07:25:21,700 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/total_energy', value: '212.6'
```

kbialek, 2024-09-18:
@Bhoft 

> I couldn't find a single line in my logs with `Data readiness check result: False`

Interesting. Please watch it for few more days.

> The `0000000000002` seems a bit strange in my opinion.
> There are multiple entries and its always ends with `0000000000002`.
> For me it looks like something isn't read correctly from the registers?

This is how floating point number encoding manifests. It's not an issue. You can ignore it.

Bhoft, 2024-09-19:
@kbialek 
> > I couldn't find a single line in my logs with `Data readiness check result: False`
> 
> Interesting. Please watch it for few more days.

Checked again this morning and there were 4 readiness checks with false and 0.0.

```
2024-09-19 05:11:27,652 - DeyeInverterState - DEBUG - Data readiness observations: []
2024-09-19 05:11:27,652 - DeyeInverterState - DEBUG - Data readiness check result: True
2024-09-19 05:12:21,749 - DeyeInverterState - DEBUG - Data readiness observations: [total_energy@2024-09-19 05:12:21.746710:0.0]
2024-09-19 05:12:21,780 - DeyeInverterState - DEBUG - Data readiness check result: False
2024-09-19 05:13:21,750 - DeyeInverterState - DEBUG - Data readiness observations: [total_energy@2024-09-19 05:13:21.746222:0.0]
2024-09-19 05:13:21,750 - DeyeInverterState - DEBUG - Data readiness check result: False
2024-09-19 05:14:21,752 - DeyeInverterState - DEBUG - Data readiness observations: [total_energy@2024-09-19 05:14:21.749380:0.0]
2024-09-19 05:14:21,752 - DeyeInverterState - DEBUG - Data readiness check result: False
2024-09-19 05:15:22,554 - DeyeInverterState - DEBUG - Data readiness observations: [total_energy@2024-09-19 05:15:22.549701:0.0]
2024-09-19 05:15:22,554 - DeyeInverterState - DEBUG - Data readiness check result: False
2024-09-19 05:16:21,771 - DeyeInverterState - DEBUG - Data readiness observations: [total_energy@2024-09-19 05:16:21.763551:216.9]
2024-09-19 05:16:21,771 - DeyeInverterState - DEBUG - Data readiness check result: True
```

I checked again older logs and the check with false did occur on the 17th as well (also 4 times).
```
2024-09-17 05:19:13,757 - DeyeInverterState - DEBUG - Data readiness check result: True
2024-09-17 05:20:13,650 - DeyeInverterState - DEBUG - Data readiness observations: [total_energy@2024-09-17 05:20:13.647024:0.0]
2024-09-17 05:20:13,650 - DeyeInverterState - DEBUG - Data readiness check result: False
2024-09-17 05:21:13,661 - DeyeInverterState - DEBUG - Data readiness observations: [total_energy@2024-09-17 05:21:13.658232:0.0]
2024-09-17 05:21:13,661 - DeyeInverterState - DEBUG - Data readiness check result: False
2024-09-17 05:22:13,626 - DeyeInverterState - DEBUG - Data readiness observations: [total_energy@2024-09-17 05:22:13.623930:0.0]
2024-09-17 05:22:13,626 - DeyeInverterState - DEBUG - Data readiness check result: False
2024-09-17 05:23:13,733 - DeyeInverterState - DEBUG - Data readiness observations: [total_energy@2024-09-17 05:23:13.729732:0.0]
2024-09-17 05:23:13,733 - DeyeInverterState - DEBUG - Data readiness check result: False
2024-09-17 05:24:13,631 - DeyeInverterState - DEBUG - Data readiness observations: [total_energy@2024-09-17 05:24:13.628597:210.9]
2024-09-17 05:24:13,631 - DeyeInverterState - DEBUG - Data readiness check result: True
```

**I have no idea why this didn't occure yesterday on the 18th.**

The only thing i see that the first `true` total energy reading has these `0000000000002` 
```
2024-09-18 05:15:27,268 - DeyeInverterState - DEBUG - Data readiness observations: []
2024-09-18 05:15:27,269 - DeyeInverterState - DEBUG - Data readiness check result: True
2024-09-18 05:16:21,386 - DeyeInverterState - DEBUG - Data readiness observations: [total_energy@2024-09-18 05:16:21.382964:212.20000000000002]
2024-09-18 05:16:21,386 - DeyeInverterState - DEBUG - Data readiness check result: True
```

kbialek, 2024-09-19:
> Checked again this morning and there were 4 readiness checks with false and 0.0.

Thanks for the feedback @Bhoft This means that the solution works as expected.

> I have no idea why this didn't occure yesterday on the 18th.

I suppose, that it's caused by a timing issue in the inverter's firmware. Probably the memory locations that are read over Modbus are being initialized after some delay. Now, it depends when we query the values. We do that either before or after the initialization. If we do before, then we get zeros. If we do after then we get the expected values.

Having said that, I'm getting confident that the new version can be released. What about you @CarstenGrohmann . Did you have a chance to test beta-4?

CarstenGrohmann, 2024-09-19:
I activated beta-4 a few hours after you released it. I see the same true and false behavior as reported by @Bhoft.
My Docker logs do not contain any errors. 

I am doing fine with the current changes. My data in HA looks good. 

Please feel free to publish the changes.

BTW: Did you have a chance to read my comment in https://github.com/kbialek/deye-inverter-mqtt/pull/200#pullrequestreview-2310667825?

closed 2024-09-19 by kbialek