kbialek/deye-inverter-mqtt #190: Crashes with nearly empty plugin during "import requests"

BjoernSch, 2024-08-18:
**Hardware (please complete the following information):**
 - Inverter model:

**Software (please complete the following information):**
 - deye-inverter-mqtt version: 2024.07.1
 - config.env
 DEYE_LOGGER_1_IP_ADDRESS	192.168.50.182
DEYE_LOGGER_1_PROTOCOL	tcp
DEYE_LOGGER_1_SERIAL_NUMBER	38XXXXX
DEYE_LOGGER_COUNT	1
DEYE_METRIC_GROUPS	micro, settings_micro
MQTT_HOST	mosquitto
PLUGINS_DIR	plugins
PLUGINS_ENABLED	deye_plugin_influx_publisher

Creating an empty plugin just with "import requests" creates the following error:
2024-08-18 18:40:20,063 - DeyeDaemon - INFO - Please help me build the list of compatible inverters. https://github.com/kbialek/deye-inverter-mqtt/issues/41
Traceback (most recent call last):
  File "/opt/deye_inverter_mqtt/./deye_docker_entrypoint.py", line 59, in <module>
    main()
  File "/opt/deye_inverter_mqtt/./deye_docker_entrypoint.py", line 55, in main
    daemon_main()
  File "/opt/deye_inverter_mqtt/deye_daemon.py", line 123, in main
    daemon = DeyeDaemon(config)
  File "/opt/deye_inverter_mqtt/deye_daemon.py", line 76, in __init__
    self.__processor_factory = DeyeProcessorFactory(self.__config, self.__mqtt_client)
  File "/opt/deye_inverter_mqtt/deye_processor_factory.py", line 41, in __init__
    self.plugin_loader.load_plugins(plugin_context)
  File "/opt/deye_inverter_mqtt/deye_plugin_loader.py", line 46, in load_plugins
    discovered_plugins = {
  File "/opt/deye_inverter_mqtt/deye_plugin_loader.py", line 47, in <dictcomp>
    name: importlib.import_module(name)
  File "/usr/local/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/opt/deye_inverter_mqtt/plugins/deye_plugin_influx_publisher.py", line 1, in <module>
    import requests
  File "/usr/local/lib/python3.10/site-packages/requests/__init__.py", line 164, in <module>
    from .api import delete, get, head, options, patch, post, put, request
  File "/usr/local/lib/python3.10/site-packages/requests/api.py", line 11, in <module>
    from . import sessions
  File "/usr/local/lib/python3.10/site-packages/requests/sessions.py", line 15, in <module>
    from .adapters import HTTPAdapter
  File "/usr/local/lib/python3.10/site-packages/requests/adapters.py", line 80, in <module>
    _preloaded_ssl_context = create_urllib3_context()
  File "/usr/local/lib/python3.10/site-packages/urllib3/util/ssl_.py", line 290, in create_urllib3_context
    context.minimum_version = TLSVersion.TLSv1_2
AttributeError: '_SSLMethod' object has no attribute 'TLSv1_2'

Running Python interactive doesn´t give an error:
/opt/deye_inverter_mqtt # python
Python 3.10.13 (main, Mar 16 2024, 05:27:13) [GCC 12.2.1 20220924] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import requests
>>> requests.ssl
requests.ssl
>>> requests.ssl.OPENSSL_VERSION
'OpenSSL 3.1.4 24 Oct 2023'
>>>

kbialek, 2024-08-19:
@BjoernSch  `requests` is not bundled in the docker image. How do you provide it to the plugin?

```
❯ docker run --rm -ti --entrypoint /bin/sh deye-inverter-mqtt
/opt/deye_inverter_mqtt # python
Python 3.10.13 (main, Mar 16 2024, 05:27:13) [GCC 12.2.1 20220924] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import requests
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'requests'
```

BjoernSch, 2024-08-20:
At the moment by "pip install requests" at the shell.
Adding the import to the core files works and the plugin is also working after that.

kbialek, 2024-08-21:
> At the moment by "pip install requests" at the shell. Adding the import to the core files works and the plugin is also working after that.

So can we close this issue?

What do you mean by "core files", btw?

BjoernSch, 2024-08-22:
I would appreciate any help with this issue.

If I add the import to your deye_daemon.py for example - it works.

Adding the import only to the plugin doesn´t work. I tried to debug it but I can not find the point why it fails.

closed 2024-10-06 by github-actions[bot]