kbialek/deye-inverter-mqtt #191: Docker on Synology NAS DS920+ witth Portainer

K3vb3rt, 2024-08-20:
Hardware (please complete the following information): Synology DS920+

Inverter model: Deye SUN800-G3-EU
Software (please complete the following information): Docker/Portainer

deye-inverter-mqtt version: 2024.07.1
DEYE_LOGGER_IP_ADDRESS=192.168.1.145
DEYE_LOGGER_PORT=8899
DEYE_LOGGER_SERIAL_NUMBER=39010XXXXX

MQTT_HOST=192.168.1.144
MQTT_PORT=1883
MQTT_USERNAME=mqtt-user
MQTT_PASSWORD=ck9jPbpwXXXXXXXXX
MQTT_TOPIC_PREFIX=deye

LOG_LEVEL=INFO
DEYE_DATA_READ_INTERVAL=10
DEYE_METRIC_GROUPS=string
DEYE_FEATURE_SET_TIME=true

Every time I try to install the container in Docker or Portainer I get this error and I have to remove it manually. Any idea what I can to? It worked until I installed watchtower and it pulled the latest image.

[image]

kbialek, 2024-08-21:
No idea tbh. I can only speculate. This seems to be quite common problem. Look [here](https://stackoverflow.com/questions/72695311/failure-starting-docker-container-failed-to-create-shim-task-oci-runtime-crea). Maybe it's caused by some file permission issue. Maybe watchtower runs as a different user (possibly root) and modifies file permissions.

K3vb3rt, 2024-08-21:
I handled it by myself by completely removing my Docker (container station) and rebuilding from ground. Now it works! Thank you ;-)

closed 2024-08-21 by K3vb3rt