kbialek/deye-inverter-mqtt #193: Question about protocol tcp vs at

Bhoft, 2024-08-26:
**Hardware (please complete the following information):**
 - Inverter model: deye SUN-M160G4-EU-Q0

**Software (please complete the following information):**
 - Logger firmware version: MW3_SSL_5408_1.0B
 - deye-inverter-mqtt version: latest
 - config.env
```
MQTT_PORT=1883
MQTT_TOPIC_PREFIX=deye
DEYE_DATA_READ_INTERVAL=60
DEYE_PUBLISH_ON_CHANGE=true
DEYE_PUBLISH_ON_CHANGE_MAX_INTERVAL=360
DEYE_METRIC_GROUPS=micro,settings
DEYE_FEATURE_SET_TIME=true
DEYE_FEATURE_ACTIVE_POWER_REGULATION=true
DEYE_FEATURE_TIME_OF_USE=false
DEYE_LOGGER_PROTOCOL=at
```

Hi my inverter auto updated its firmware from MW3_16U_5406_1.59 to MW3_SSL_5408_1.0B this morning.
I realized that the `DEYE_LOGGER_PROTOCOL=at` mode doesn't work any more as it seems that the port is now blocked.

The `DEYE_LOGGER_PROTOCOL=tcp` mode is still working.
Are they any benefits from one or the other protocol?
What are the differences (maybe faster response etc.)?

Is there a pssibility to downgrade?
I have now blocked the internet access for my inverter and I hope that it doesn't is already too late.

kbialek, 2024-08-26:
> What are the differences (maybe faster response etc.)?

There is no logical difference, both protocols encapsulate Modbus frames that are essential for the communication with the inverter.
Historically, `tcp` has been implemented first (actually ported from another project). Later, `tcp` stopped working with firmware versions >= 2.x, so then I added `udp/at` protocol (which is also ported from yet another project).
Performance-wise I see no much difference between them. `tcp` seems to be slightly faster. But I didn't measure it tbh.

> Is there a pssibility to downgrade?

Probably yes. You should seek for help at one of the forums, like https://www.photovoltaikforum.com/ or https://diysolarforum.com/

dmaj, 2024-08-29:
Since the firmware update to MW3_SSL_5408_1.0B (SUNxG3, port 48899 is no longer available. This means that udp/at is probably no longer an option.

closed 2024-09-10 by Bhoft