kbialek/deye-inverter-mqtt #207: Deye Inverter Cloud free with daily reset

Kolbi, 2024-10-18:
**Is your feature request related to a problem? Please describe.**
Not really, but currently the Deye Inverter needs to be connected to the Internet.

**Describe the solution you'd like**
Here there is a cloudfree solution: [deye-microinverter-cloud-free](https://github.com/Hypfer/deye-microinverter-cloud-free/tree/master?tab=readme-ov-file#using-the-dummycloud) which also updates the time on the inverter for the daily reset, any idea how to implement this to your solution? Would be great to be cloud free and to disallow internet access.

kbialek, 2024-10-18:
This feature exists for long time already https://github.com/kbialek/deye-inverter-mqtt?tab=readme-ov-file#automatically-set-loggerinverter-time

Kolbi, 2024-10-18:
Thanks! Was not aware about this feature. Does it take the time from the docker?

kbialek, 2024-10-18:
> Does it take the time from the docker?

Yes

closed 2024-10-18 by Kolbi