kbialek/deye-inverter-mqtt #210: More Configuration Options

belowm, 2024-11-09:
First of all, thanks for this excellent project!

I'm missing a specific setting (namely the max current the allowed for grid charge).  I understand that this can be seen more of a "hard configuration value" that is not really meant to be used in automations, but I found this to be the only reliable way to take full control of grid chargingwith `Deye SG04LP3` (turning it off by setting this to `0`, turning it on by setting it to > `0`.

The corresponding modbus address is `40`, the setting can be found here: 
[image]

Below is the code to add the parameter, which works fine with `Deye SG04LP3`. However, I wasn't able to test this with the micro inverter:

```diff
diff --git a/src/deye_sensors_settings.py b/src/deye_sensors_settings.py
index 95c2e70..d91fa09 100644
--- a/src/deye_sensors_settings.py
+++ b/src/deye_sensors_settings.py
@@ -37,8 +37,18 @@ deye_settings_active_power_regulation_micro = SingleRegisterSensor(
     groups=["settings_micro"],
 )
 
-deye_settings_sensors = [deye_settings_active_power_regulation_default, deye_settings_active_power_regulation_micro]
+deye_settings_gridcharge_current = SingleRegisterSensor(
+    "Gridcharge current",
+    128,
+    1,
+    mqtt_topic_suffix="settings/gridcharge_current",
+    unit="A",
+    signed=False,
+    groups=["settings"],
+)
+
+deye_settings_sensors = [deye_settings_active_power_regulation_default, deye_settings_active_power_regulation_micro, deye_settings_gridcharge_current]
 
 deye_settings_register_ranges = [
-    SensorRegisterRange(group={"settings", "settings_micro"}, first_reg_address=40, last_reg_address=40),
-]
+    SensorRegisterRange(group={"settings", "settings_micro"}, first_reg_address=40, last_reg_address=128),
+]
\ No newline at end of file
```

So if you are willing to add this, please let me know and I'll create a merge request. 
Should such setting be out of scope of this project: Are there any plans to enhance the plugin API so that sensors can be added as part of a plugin? Thanks!

belowm, 2024-11-09:
Also added this [here](https://github.com/kbialek/deye-inverter-mqtt/compare/main...belowm:deye-inverter-mqtt:main)

closed 2024-11-12 by belowm

kbialek, 2024-11-12:
Why did you close this issue? Should I ignore this change request?

belowm, 2024-11-14:
Yes, I've closed this because I think this can be solved more elegantly by making DeyeModbus available to plugins, which I [tried to implement here](https://github.com/kbialek/deye-inverter-mqtt/pull/211). With this, I'd be able to add my specific requirement via a plugin.