kbialek/deye-inverter-mqtt #212: Losing connection to mqtt broker

Kolbi, 2024-11-13:
**Describe the bug**
Sometimes after a time (very often between 10-11am) the deye-mqtt docker container looses the connection to the mqtt broker. After restarting the deye-mqtt container everything works again.

**Expected behavior**
After a disconnect from the mqtt broker a new connection should be tried to establish. Sadly I can't find something interesting in the logs...

kbialek, 2024-11-14:
You need to enable DEBUG logs to get more info. Maybe you will catch the cause this way.

Kolbi, 2024-11-20:
Thanks, last data at around 10:24:44.

One the two lines are looking a little bit suspicious:

> 2024-11-20T09:25:06.537591243Z 2024-11-20 10:25:06,537 - paho.mqtt.client - DEBUG - Sending CONNECT (u1, p1, wr1, wq1, wf1, c1, k60) client_id=b'deye-inverter-[SNremoved]'
> 2024-11-20T09:25:06.538880301Z 2024-11-20 10:25:06,538 - paho.mqtt.client - DEBUG - Received CONNACK (0, 0)

Log (SN removed):

> 2024-11-20T09:24:44.152625702Z 2024-11-20 10:24:44,152 - DeyeDaemon - DEBUG - Invoking action
> 2024-11-20T09:24:44.152799366Z 2024-11-20 10:24:44,152 - DeyeInverterState - INFO - Reading start
> 2024-11-20T09:24:44.152881918Z 2024-11-20 10:24:44,152 - DeyeInverterState - INFO - Reading registers [metrics group: {'micro'}, range: 003c-0074]
> 2024-11-20T09:24:44.161105115Z 2024-11-20 10:24:44,160 - DeyeTcpConnector - DEBUG - Request frame: [request]
> 2024-11-20T09:24:44.344110581Z 2024-11-20 10:24:44,343 - DeyeTcpConnector - DEBUG - Received response frame in 1. attempt: [respone]
> 2024-11-20T09:24:44.344506645Z 2024-11-20 10:24:44,344 - DeyeInverterState - DEBUG - Production today: 0.3
> 2024-11-20T09:24:44.344690925Z 2024-11-20 10:24:44,344 - DeyeInverterState - DEBUG - Production Total: 3578.5
> 2024-11-20T09:24:44.344834422Z 2024-11-20 10:24:44,344 - DeyeInverterState - DEBUG - Phase1 Voltage: 238.0
> 2024-11-20T09:24:44.345076578Z 2024-11-20 10:24:44,344 - DeyeInverterState - DEBUG - Phase1 Current: 0.0
> 2024-11-20T09:24:44.345203429Z 2024-11-20 10:24:44,345 - DeyeInverterState - DEBUG - Phase1 Power: 0.0
> 2024-11-20T09:24:44.345350012Z 2024-11-20 10:24:44,345 - DeyeInverterState - DEBUG - AC Freq: 50.0
> 2024-11-20T09:24:44.345482913Z 2024-11-20 10:24:44,345 - DeyeInverterState - DEBUG - Uptime: 0.0
> 2024-11-20T09:24:44.345578593Z 2024-11-20 10:24:44,345 - DeyeInverterState - DEBUG - PV1 Voltage: 26.8
> 2024-11-20T09:24:44.345698684Z 2024-11-20 10:24:44,345 - DeyeInverterState - DEBUG - PV1 Current: 3.3
> 2024-11-20T09:24:44.345823438Z 2024-11-20 10:24:44,345 - DeyeInverterState - DEBUG - PV1 Power: 88.4
> 2024-11-20T09:24:44.345952363Z 2024-11-20 10:24:44,345 - DeyeInverterState - DEBUG - PV1 Production today: 0.0
> 2024-11-20T09:24:44.346070459Z 2024-11-20 10:24:44,345 - DeyeInverterState - DEBUG - PV1 Total: 919.3
> 2024-11-20T09:24:44.346187768Z 2024-11-20 10:24:44,346 - DeyeInverterState - DEBUG - PV2 Voltage: 37.3
> 2024-11-20T09:24:44.346287745Z 2024-11-20 10:24:44,346 - DeyeInverterState - DEBUG - PV2 Current: 2.3
> 2024-11-20T09:24:44.346425030Z 2024-11-20 10:24:44,346 - DeyeInverterState - DEBUG - PV2 Power: 85.8
> 2024-11-20T09:24:44.346557492Z 2024-11-20 10:24:44,346 - DeyeInverterState - DEBUG - PV2 Production today: 0.0
> 2024-11-20T09:24:44.346655982Z 2024-11-20 10:24:44,346 - DeyeInverterState - DEBUG - PV2 Total: 630.7
> 2024-11-20T09:24:44.346770455Z 2024-11-20 10:24:44,346 - DeyeInverterState - DEBUG - PV3 Voltage: 38.3
> 2024-11-20T09:24:44.346865210Z 2024-11-20 10:24:44,346 - DeyeInverterState - DEBUG - PV3 Current: 2.6
> 2024-11-20T09:24:44.346967013Z 2024-11-20 10:24:44,346 - DeyeInverterState - DEBUG - PV3 Power: 99.6
> 2024-11-20T09:24:44.347109854Z 2024-11-20 10:24:44,346 - DeyeInverterState - DEBUG - PV3 Production today: 0.0
> 2024-11-20T09:24:44.347183719Z 2024-11-20 10:24:44,347 - DeyeInverterState - DEBUG - PV3 Total: 944.8
> 2024-11-20T09:24:44.347313532Z 2024-11-20 10:24:44,347 - DeyeInverterState - DEBUG - PV4 Voltage: 30.3
> 2024-11-20T09:24:44.347533926Z 2024-11-20 10:24:44,347 - DeyeInverterState - DEBUG - PV4 Current: 3.4
> 2024-11-20T09:24:44.347548057Z 2024-11-20 10:24:44,347 - DeyeInverterState - DEBUG - PV4 Power: 103.0
> 2024-11-20T09:24:44.347651739Z 2024-11-20 10:24:44,347 - DeyeInverterState - DEBUG - PV4 Production today: 0.0
> 2024-11-20T09:24:44.347783244Z 2024-11-20 10:24:44,347 - DeyeInverterState - DEBUG - PV4 Total: 951.1
> 2024-11-20T09:24:44.347899072Z 2024-11-20 10:24:44,347 - DeyeInverterState - DEBUG - DC Total Power: 376.8
> 2024-11-20T09:24:44.347936260Z 2024-11-20 10:24:44,347 - DeyeInverterState - DEBUG - Operating Power: 0.0
> 2024-11-20T09:24:44.348091019Z 2024-11-20 10:24:44,347 - DeyeInverterState - DEBUG - AC Active Power: 355.0
> 2024-11-20T09:24:44.348198842Z 2024-11-20 10:24:44,348 - DeyeInverterState - DEBUG - Radiator temperature: 7.9
> 2024-11-20T09:24:44.348392108Z 2024-11-20 10:24:44,348 - DeyeInverterState - DEBUG - Data readiness observations: [total_energy@2024-11-20 10:24:44.344211:3578.5]
> 2024-11-20T09:24:44.348420133Z 2024-11-20 10:24:44,348 - DeyeInverterState - DEBUG - Data readiness check result: True
> 2024-11-20T09:24:44.348494419Z 2024-11-20 10:24:44,348 - DeyeMqttClient - DEBUG - Publishing message. topic: 'homeassistant/sensor/deye/logger_status', value: 'online'
> 2024-11-20T09:24:44.348675372Z 2024-11-20 10:24:44,348 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m46312), 'b'homeassistant/sensor/deye/logger_status'', ... (6 bytes)
> 2024-11-20T09:24:44.349489367Z 2024-11-20 10:24:44,349 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 46312)
> 2024-11-20T09:24:44.349718236Z 2024-11-20 10:24:44,349 - DeyeMqttClient - INFO - Logger is online
> 2024-11-20T09:24:44.349857059Z 2024-11-20 10:24:44,349 - DeyeMqttClient - DEBUG - Publishing message. topic: 'homeassistant/sensor/deye/day_energy', value: '0.3'
> 2024-11-20T09:24:44.350055372Z 2024-11-20 10:24:44,349 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m46313), 'b'homeassistant/sensor/deye/day_energy'', ... (3 bytes)
[... cut ...] -inverter-mqtt with HA Discovery Plugin version 2024-10-22"}}'
> 2024-11-20T09:25:45.172140735Z 2024-11-20 10:25:45,171 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m46441), 'b'homeassistant/sensor/deye_inverter_mqtt_[SNremoved]/radiator_temp/config'', ... (602 bytes)
> 2024-11-20T09:25:45.172762158Z 2024-11-20 10:25:45,172 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 46441)
> 2024-11-20T09:25:45.173017817Z 2024-11-20 10:25:45,172 - DeyeInverterState - INFO - Reading completed

Kolbi, 2024-11-25:
Seems to be daily now and always at the same time.

kbialek, 2024-11-25:
Sorry to say that, but I belive that the problem is on your end. Maybe something is off with your network or infra setup.

Kolbi, 2024-11-25:
Okay need to check. Mosquitto and your solution are running as two separate docker instances on the same machine.

Kolbi, 2024-11-28:
I reference it as bookmark for me :) https://github.com/eclipse-mosquitto/mosquitto/pull/3168

closed 2024-12-02 by kbialek

Kolbi, 2024-12-05:
Still not fixed but workaround:
`services:
  restart-notifier:
    container_name: restart-notifier
    image: index.docker.io/deduard/tools:restart-notifier-latest
    env_file:
      - stack.env
    restart: unless-stopped
    volumes:
      - /var/run/docker.sock:/var/run/docker.sock
      - /etc/localtime:/etc/localtime:ro
      - 
 ENV:
RESTART_CONTAINERS deye-mqtt-deye-mqtt-1 
CRON_SCHEDULE 30 10 * * *