kbialek/deye-inverter-mqtt #213: Non-numeric serial number

MagiX13, 2024-11-24:
For some reason, my Deye SUN-M80G3-EU-Q0 (obtained in Nov 2023, never upgrade the firmware thus running `MW3_16U_5406_2.27`) has a non-numeric serial number, ending with BD. 

I made some local changes to accommodate this, but was wondering if there are more people with this/there is a need for a pull request - I think it amounts to 3-4 lines that I had to change. I had it running with these changes since March already and it seems stable. Today I decided to update the docker repo and saw that this is still not supported and would be happy to submit a PR.

kbialek, 2024-11-25:
hi! I think it was a bad design decision to model the serial number as integer, so if you would like to submit a PR fixing this, that would be great.

thalesmaoa, 2025-01-03:
Same problem here. Serial number ending with F character. @MagiX13 , I'm sorry, but I wasn't able to find your fork. Can you provide the places you've edited?
I just got a SUN-M225G4.

kbialek, 2025-01-05:
@MagiX13 I can implement this change. Could you tell me how do you encode textual serial number into the Modbus frame? I'm referring to this code https://github.com/kbialek/deye-inverter-mqtt/blob/5cca6e13e20bfbb6868c9e8a8f5f5e15c0b40fcb/src/deye_modbus_tcp_custom.py#L47

thalesmaoa, 2025-01-06:
I use to use WVC script from @krywenko https://github.com/krywenko/WVC-inverters/blob/dfd8d103635cdf91f37f67a41c00a81fb87d6483/wvc#L21. I'm not sure how deye build its serial number to request the message. 

It is a modbus tcp? do I need to encode the serial number? if so, I can make some tests to find out.

thalesmaoa, 2025-01-06:
`bytearray.fromhex(self.loggger_config.serial_number)`?

kbialek, 2025-01-07:
@MagiX13 I have problem in understanding how this code `f"{self.config.serial_number>10}"` works :slightly_smiling_face:  Actually it fails to execute.
```
❯ python
Python 3.10.12 (main, Nov  6 2024, 20:22:13) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> sn="123F"
>>> f"{sn>10}"
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: '>' not supported between instances of 'str' and 'int'
>>> 
```
Can you check on your side?

thalesmaoa, 2025-01-07:
Yes, since I know nothing about the communication system with the deye. How it is encoded and so on, I need to know if that makes sense. 

```
In [11]: sn="1234567890"

In [12]: int(sn, 16)
Out[12]: 78187493520
```

In other words, does older versions has an in serial number and new ones hex serial number? This is the question. The code we fix it.

From @MagiX13 answer, it seems that my suggestion is correct. There must be a if statement for the serial number.

MagiX13, 2025-01-07:
Sorry, there was a typo in there, should have been
`f"{sn:>10}"`

kbialek, 2025-01-07:
Thanks @MagiX13 It's clear now that the SN is hex encoded

thalesmaoa, 2025-01-07:
No, perhaps is my inverter not compatible

Serial: `240713044f`
Int: `154737509455`

Model: `SUN-M225G4`

Config
```
DEYE_LOGGER_IP_ADDRESS=192.168.200.193
DEYE_LOGGER_PORT=8899
DEYE_LOGGER_SERIAL_NUMBER=154737509455
# DEYE_LOGGER_PROTOCOL=tcp
# DEYE_LOGGER_MAX_REG_RANGE_LENGTH=256

MQTT_HOST=192.168.200.254
# MQTT_PORT=1883
# MQTT_USERNAME=user
# MQTT_PASSWORD=password
# MQTT_TOPIC_PREFIX=deye
# MQTT_TLS_ENABLED=true
# MQTT_TLS_INSECURE=true
# MQTT_TLS_CA_CERT_PATH=certs/ca.crt
# MQTT_TLS_CLIENT_CERT_PATH=certs/client.crt
# MQTT_TLS_CLIENT_KEY_PATH=certs/client.key

LOG_LEVEL=DEBUG
LOG_STREAM=STDOUT
# PLUGINS_DIR=plugins
# PLUGINS_ENABLED=deye_plugin_sample,deye_plugin_custom
DEYE_DATA_READ_INTERVAL=5
DEYE_METRIC_GROUPS=micro,settings
# DEYE_PUBLISH_ON_CHANGE=false
# DEYE_PUBLISH_ON_CHANGE_MAX_INTERVAL=360
DEYE_METRIC_GROUPS=string

# DEYE_FEATURE_MQTT_PUBLISHER=true
# DEYE_FEATURE_TIME_OF_USE=false
# DEYE_FEATURE_ACTIVE_POWER_REGULATION=false
# DEYE_FEATURE_SET_TIME=false

## Sample multiinverter configuration with two loggers
#
# DEYE_LOGGER_COUNT=2

# DEYE_LOGGER_1_IP_ADDRESS=192.168.0.1
# DEYE_LOGGER_1_SERIAL_NUMBER=1234567890
# DEYE_LOGGER_1_PROTOCOL=at

# DEYE_LOGGER_2_IP_ADDRESS=192.168.0.2
# DEYE_LOGGER_2_SERIAL_NUMBER=1234567891
# DEYE_LOGGER_2_PROTOCOL=at
```

log
```
2025-01-07 19:05:31,339 - DeyeInverterState - INFO - Reading completed
2025-01-07 19:05:36,319 - DeyeDaemon - DEBUG - Invoking action
2025-01-07 19:05:36,319 - DeyeInverterState - INFO - Reading start
2025-01-07 19:05:36,319 - DeyeInverterState - INFO - Reading registers [metrics group: {'string'}, range: 003c-0074]
2025-01-07 19:05:36,332 - DeyeTcpConnector - DEBUG - Request frame: a51700104500004f041307240200000000000000000000000000000103003c003945d49115
2025-01-07 19:05:36,344 - DeyeTcpConnector - DEBUG - Received response frame in 1. attempt: a51000101500845cc33be72401d66b020069150000290f7b670200fc15
2025-01-07 19:05:36,344 - DeyeModbusTcp - ERROR - Unknown response error code. Error frame: 0200
2025-01-07 19:05:36,344 - DeyeInverterState - DEBUG - Data readiness observations: []
2025-01-07 19:05:36,344 - DeyeInverterState - DEBUG - Data readiness check result: True
2025-01-07 19:05:36,344 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/logger_status', value: 'offline'
2025-01-07 19:05:36,344 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m45), 'b'deye/logger_status'', ... (7 bytes)
2025-01-07 19:05:36,346 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 45)
2025-01-07 19:05:36,346 - DeyeMqttClient - INFO - Logger is offline
2025-01-07 19:05:36,346 - DeyeInverterState - INFO - Reading completed
2025-01-07 19:05:41,319 - DeyeDaemon - DEBUG - Invoking action
2025-01-07 19:05:41,319 - DeyeInverterState - INFO - Reading start
2025-01-07 19:05:41,319 - DeyeInverterState - INFO - Reading registers [metrics group: {'string'}, range: 003c-0074]
2025-01-07 19:05:41,327 - DeyeTcpConnector - DEBUG - Request frame: a51700104500004f041307240200000000000000000000000000000103003c003945d49115
2025-01-07 19:05:41,403 - DeyeTcpConnector - DEBUG - Received response frame in 1. attempt: a51000101500855cc33be72401db6b02006e150000290f7b6702000715
2025-01-07 19:05:41,403 - DeyeModbusTcp - ERROR - Unknown response error code. Error frame: 0200
2025-01-07 19:05:41,403 - DeyeInverterState - DEBUG - Data readiness observations: []
2025-01-07 19:05:41,404 - DeyeInverterState - DEBUG - Data readiness check result: True
2025-01-07 19:05:41,404 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/logger_status', value: 'offline'
2025-01-07 19:05:41,404 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m46), 'b'deye/logger_status'', ... (7 bytes)
2025-01-07 19:05:41,408 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 46)
2025-01-07 19:05:41,408 - DeyeMqttClient - INFO - Logger is offline
2025-01-07 19:05:41,408 - DeyeInverterState - INFO - Reading completed
2025-01-07 19:05:46,319 - DeyeDaemon - DEBUG - Invoking action
2025-01-07 19:05:46,319 - DeyeInverterState - INFO - Reading start
2025-01-07 19:05:46,319 - DeyeInverterState - INFO - Reading registers [metrics group: {'string'}, range: 003c-0074]
2025-01-07 19:05:46,333 - DeyeTcpConnector - DEBUG - Request frame: a51700104500004f041307240200000000000000000000000000000103003c003945d49115
2025-01-07 19:05:46,340 - DeyeTcpConnector - DEBUG - Received response frame in 1. attempt: a51000101500865cc33be72401e06b020073150000290f7b6702001215
2025-01-07 19:05:46,340 - DeyeModbusTcp - ERROR - Unknown response error code. Error frame: 0200
2025-01-07 19:05:46,340 - DeyeInverterState - DEBUG - Data readiness observations: []
2025-01-07 19:05:46,340 - DeyeInverterState - DEBUG - Data readiness check result: True
2025-01-07 19:05:46,340 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/logger_status', value: 'offline'
2025-01-07 19:05:46,340 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m47), 'b'deye/logger_status'', ... (7 bytes)
2025-01-07 19:05:46,342 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 47)
2025-01-07 19:05:46,342 - DeyeMqttClient - INFO - Logger is offline
2025-01-07 19:05:46,342 - DeyeInverterState - INFO - Reading completed
^C2025-01-07 19:05:51,147 - DeyeDaemon - DEBUG - Stopping the runner
2025-01-07 19:05:51,147 - DeyeDaemon - DEBUG - Invocation loop stopped
```

kbialek, 2025-01-07:
@thalesmaoa  The SN `240713044f` is properly placed in the request frame:
```
a5170010450000 4f04130724 0200000000000000000000000000000103003c003945d49115
               ^^^^^^^^^^
```

Bytes are reversed as expected.

What is your timezone? Is there sunlight and is the inverter producing energy?

thalesmaoa, 2025-01-07:
Yes! I'm in Brazil. GMT -3.

From my readings, it seems that it is getting a response. I still have a few minutes for a test.

```
2025-01-07 19:05:41,327 - DeyeTcpConnector - DEBUG - Request frame: a51700104500004f041307240200000000000000000000000000000103003c003945d49115
2025-01-07 19:05:41,403 - DeyeTcpConnector - DEBUG - Received response frame in 1. attempt: a51000101500855cc33be72401db6b02006e150000290f7b6702000715
2025-01-07 19:05:41,403 - DeyeModbusTcp - ERROR - Unknown response error code. Error frame: 0200
2025-01-07 19:05:41,403 - DeyeInverterState - DEBUG - Data readiness observations: []
2025-01-07 19:05:41,404 - DeyeInverterState - DEBUG - Data readiness check result: True
```

Not sure what `Error frame: 0200` means.

kbialek, 2025-01-08:
@thalesmaoa Did you try using `at` protocol btw? It works great with my microinverters.

thalesmaoa, 2025-01-08:
Yes, 

```
2025-01-08 21:16:55,372 - DeyeInverterState - INFO - Reading completed
2025-01-08 21:16:55,372 - DeyeDaemon - DEBUG - Invoking action
2025-01-08 21:16:55,372 - DeyeInverterState - INFO - Reading start
2025-01-08 21:16:55,372 - DeyeInverterState - INFO - Reading registers [metrics group: {'micro'}, range: 003c-0074]
2025-01-08 21:16:55,373 - DeyeAtConnector - DEBUG - Sending AT command: b'WIFIKIT-214028-READ'
2025-01-08 21:16:56,474 - DeyeAtConnector - DEBUG - Connection response timeout
2025-01-08 21:16:57,476 - DeyeAtConnector - DEBUG - Connection response timeout
2025-01-08 21:16:58,478 - DeyeAtConnector - DEBUG - Connection response timeout
2025-01-08 21:16:59,478 - DeyeAtConnector - DEBUG - Connection response timeout
2025-01-08 21:17:00,480 - DeyeAtConnector - DEBUG - Connection response timeout
2025-01-08 21:17:00,480 - DeyeAtConnector - WARNING - Too many connection timeouts
2025-01-08 21:17:00,480 - DeyeAtConnector - DEBUG - Sending AT command: b'+ok'
2025-01-08 21:17:00,581 - DeyeAtConnector - DEBUG - Sending AT command: b'AT+INVDATA=8,0103003c003945d4\n'
2025-01-08 21:17:02,683 - DeyeAtConnector - DEBUG - Connection response timeout
2025-01-08 21:17:03,684 - DeyeAtConnector - DEBUG - Connection response timeout
2025-01-08 21:17:04,686 - DeyeAtConnector - DEBUG - Connection response timeout
2025-01-08 21:17:05,383 - paho.mqtt.client - DEBUG - Sending PINGREQ
2025-01-08 21:17:05,409 - paho.mqtt.client - DEBUG - Received PINGRESP
2025-01-08 21:17:05,686 - DeyeAtConnector - DEBUG - Connection response timeout
2025-01-08 21:17:06,687 - DeyeAtConnector - DEBUG - Connection response timeout
2025-01-08 21:17:06,687 - DeyeAtConnector - WARNING - Too many connection timeouts
2025-01-08 21:17:06,688 - DeyeInverterState - INFO - Reading registers [metrics group: {'settings', 'settings_micro'}, range: 0028-0028]
2025-01-08 21:17:06,688 - DeyeAtConnector - DEBUG - Sending AT command: b'WIFIKIT-214028-READ'
2025-01-08 21:17:07,789 - DeyeAtConnector - DEBUG - Connection response timeout
2025-01-08 21:17:08,790 - DeyeAtConnector - DEBUG - Connection response timeout
2025-01-08 21:17:09,791 - DeyeAtConnector - DEBUG - Connection response timeout
2025-01-08 21:17:10,792 - DeyeAtConnector - DEBUG - Connection response timeout
2025-01-08 21:17:11,794 - DeyeAtConnector - DEBUG - Connection response timeout
2025-01-08 21:17:11,794 - DeyeAtConnector - WARNING - Too many connection timeouts
2025-01-08 21:17:11,794 - DeyeAtConnector - DEBUG - Sending AT command: b'+ok'
2025-01-08 21:17:11,894 - DeyeAtConnector - DEBUG - Sending AT command: b'AT+INVDATA=8,0103002800010402\n'
2025-01-08 21:17:13,996 - DeyeAtConnector - DEBUG - Connection response timeout
2025-01-08 21:17:14,997 - DeyeAtConnector - DEBUG - Connection response timeout
2025-01-08 21:17:15,998 - DeyeAtConnector - DEBUG - Connection response timeout
2025-01-08 21:17:17,000 - DeyeAtConnector - DEBUG - Connection response timeout
2025-01-08 21:17:18,001 - DeyeAtConnector - DEBUG - Connection response timeout
2025-01-08 21:17:18,001 - DeyeAtConnector - WARNING - Too many connection timeouts
2025-01-08 21:17:18,001 - DeyeInverterState - DEBUG - Data readiness observations: []
2025-01-08 21:17:18,001 - DeyeInverterState - DEBUG - Data readiness check result: True
2025-01-08 21:17:18,001 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/logger_status', value: 'offline'
2025-01-08 21:17:18,001 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m4), 'b'deye/logger_status'', ... (7 bytes)
2025-01-08 21:17:18,012 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 4)
2025-01-08 21:17:18,013 - DeyeMqttClient - INFO - Logger is offline
2025-01-08 21:17:18,013 - DeyeInverterState - INFO - Reading completed
2025-01-08 21:17:18,013 - DeyeDaemon - DEBUG - Invoking action
2025-01-08 21:17:18,013 - DeyeInverterState - INFO - Reading start
2025-01-08 21:17:18,013 - DeyeInverterState - INFO - Reading registers [metrics group: {'micro'}, range: 003c-0074]
2025-01-08 21:17:18,013 - DeyeAtConnector - DEBUG - Sending AT command: b'WIFIKIT-214028-READ'
2025-01-08 21:17:19,114 - DeyeAtConnector - DEBUG - Connection response timeout
2025-01-08 21:17:20,114 - DeyeAtConnector - DEBUG - Connection response timeout
2025-01-08 21:17:21,115 - DeyeAtConnector - DEBUG - Connection response timeout
2025-01-08 21:17:22,117 - DeyeAtConnector - DEBUG - Connection response timeout
2025-01-08 21:17:23,118 - DeyeAtConnector - DEBUG - Connection response timeout
2025-01-08 21:17:23,118 - DeyeAtConnector - WARNING - Too many connection timeouts
2025-01-08 21:17:23,118 - DeyeAtConnector - DEBUG - Sending AT command: b'+ok'
2025-01-08 21:17:23,218 - DeyeAtConnector - DEBUG - Sending AT command: b'AT+INVDATA=8,0103003c003945d4\n'
^C2025-01-08 21:17:24,671 - DeyeDaemon - DEBUG - Stopping the runner
2025-01-08 21:17:25,321 - DeyeAtConnector - DEBUG - Connection response timeout
2025-01-08 21:17:26,323 - DeyeAtConnector - DEBUG - Connection response timeout
2025-01-08 21:17:27,324 - DeyeAtConnector - DEBUG - Connection response timeout
2025-01-08 21:17:28,325 - DeyeAtConnector - DEBUG - Connection response timeout
2025-01-08 21:17:29,326 - DeyeAtConnector - DEBUG - Connection response timeout
2025-01-08 21:17:29,327 - DeyeAtConnector - WARNING - Too many connection timeouts
2025-01-08 21:17:29,327 - DeyeInverterState - INFO - Reading registers [metrics group: {'settings', 'settings_micro'}, range: 0028-0028]
[... cut ...]
2025-01-08 21:17:40,642 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/logger_status', value: 'offline'
2025-01-08 21:17:40,642 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m5), 'b'deye/logger_status'', ... (7 bytes)
2025-01-08 21:17:40,645 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 5)
2025-01-08 21:17:40,645 - DeyeMqttClient - INFO - Logger is offline
2025-01-08 21:17:40,645 - DeyeInverterState - INFO - Reading completed
2025-01-08 21:17:40,645 - DeyeDaemon - DEBUG - Invocation loop stopped
```

Not working.

thalesmaoa, 2025-01-10:
@kbialek , what is your suggestion for debug?

kbialek, 2025-01-11:
> @kbialek , what is your suggestion for debug?

None :disappointed: 

You may also try out the new `mbtcp` protocol which I've added just recently for other inverter model.

thalesmaoa, 2025-01-11:
I found the problem, but not sure yet how to solve it.

Short answer: serial number is wrong.

I was playing with awesome work from jedie https://github.com/jedie/inverter-connect and I could retrieve my serial using AT.

`240713044f` = `3879453532`

thalesmaoa, 2025-01-11:
Just to add.
I found that:

Inverter serial number
240713044F

Device serial number
3879453532

The last one is firmware relatted.

thalesmaoa, 2025-01-12:
Sorry for not making it clear.
It does work with TCP and AT.

The answer for this issue is not that.

What if someone has a inverter serial number with letters?
Don't use the value labelled in the inverter. The one next to Qr code.

Which is the correct number?
Just look into the device information using http. There you will find two serial numbers.

The first one, called inverter serial number, is the same labelled next to the Qr code. Just ignore this number.

The second one, called device serial number is the correct one and the one you should use.

kbialek, 2025-01-12:
> It does work with TCP and AT.

:+1: 

> The first one, called inverter serial number, is the same labelled next to the Qr code. Just ignore this number.
> 
> The second one, called device serial number is the correct one and the one you should use.

That's correct. Always use device serial number aka logger serial number.

closed 2025-02-26 by github-actions[bot]

MrMartty, 2025-07-22:
@thalesmaoa did you manage to solve the problem?

thalesmaoa, 2025-07-23:
Yes. It is in the history. Just use the firmware number