kbialek/deye-inverter-mqtt #215: Not working with a DEYE 10k-g06p3 inverter

lpcvoid, 2024-12-07:
**Describe the bug**

Hello, I have tried this project with my deye 10k-g06p3. It doesn't work. Here's what happens:

```
2024-12-07 11:26:04,445 - DeyeDaemon - INFO - Please help me build the list of compatible inverters. https://github.com/kbialek/deye-inverter-mqtt/issues/41
2024-12-07 11:26:04,477 - DeyeConnectorFactory - INFO - Creating Modbus/TCP Logger connector
2024-12-07 11:26:04,478 - DeyeProcessorFactory - INFO - Feature "Report metrics over MQTT": enabled
2024-12-07 11:26:04,478 - DeyeProcessorFactory - INFO - Feature "Set inverter time once online": disabled
2024-12-07 11:26:04,478 - DeyeProcessorFactory - INFO - Feature "Time-of-use configuration over MQTT": disabled
2024-12-07 11:26:04,478 - DeyeActivePowerRegulationEventProcessor - ERROR - Active power regulation sensor not found. Enable appropriate settings metric group.
2024-12-07 11:26:04,478 - DeyeProcessorFactory - INFO - Feature "Active power regulation over MQTT": disabled
2024-12-07 11:26:04,484 - paho.mqtt.client - DEBUG - Sending CONNECT (u1, p1, wr1, wq1, wf1, c1, k60) client_id=b'deye-inverter-2405078035'
2024-12-07 11:26:04,485 - paho.mqtt.client - DEBUG - Received CONNACK (0, 0)
2024-12-07 11:26:05,485 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r1, m1), 'b'deye/status'', ... (6 bytes)
2024-12-07 11:26:05,485 - DeyeMqttClient - INFO - Successfully connected to MQTT Broker located at localhost:1883
2024-12-07 11:26:05,486 - DeyeDaemon - DEBUG - Starting executing the runner at intervals of 5 seconds
2024-12-07 11:26:05,486 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 1)
2024-12-07 11:26:07,486 - DeyeDaemon - DEBUG - Invoking action
2024-12-07 11:26:07,486 - DeyeInverterState - INFO - Reading start
2024-12-07 11:26:07,487 - DeyeInverterState - INFO - Reading registers [metrics group: {'string'}, range: 003c-0040]
2024-12-07 11:26:07,489 - DeyeTcpConnector - DEBUG - Request frame: a517001045000013945a8f0200000000000000000000000000000103003c000545c54d15
2024-12-07 11:26:17,497 - DeyeTcpConnector - DEBUG - Connection response timeout
2024-12-07 11:26:27,507 - DeyeTcpConnector - DEBUG - Connection response timeout
2024-12-07 11:26:37,513 - DeyeTcpConnector - DEBUG - Connection response timeout
2024-12-07 11:26:47,521 - DeyeTcpConnector - DEBUG - Connection response timeout
2024-12-07 11:26:57,529 - DeyeTcpConnector - DEBUG - Connection response timeout
2024-12-07 11:26:57,530 - DeyeTcpConnector - WARNING - Too many connection timeouts
2024-12-07 11:26:57,530 - DeyeInverterState - INFO - Reading registers [metrics group: {'string'}, range: 0041-0045]
2024-12-07 11:26:57,532 - DeyeTcpConnector - DEBUG - Request frame: a517001045000013945a8f020000000000000000000000000000010300410005d5ddfa15
2024-12-07 11:27:04,541 - paho.mqtt.client - DEBUG - Sending PINGREQ
2024-12-07 11:27:04,542 - paho.mqtt.client - DEBUG - Received PINGRESP
2024-12-07 11:27:07,541 - DeyeTcpConnector - DEBUG - Connection response timeout
```

I am using the "string" group, like so (docker-compose config):

```
deye-mqtt:
    image: ghcr.io/kbialek/deye-inverter-mqtt
    container_name: deye-mqtt
    environment:
      - MQTT_HOST=localhost
      - MQTT_PORT=1883
      - MQTT_USERNAME=blabla
      - MQTT_PASSWORD=blabla
      - MQTT_TOPIC_PREFIX=deye
      - DEYE_LOGGER_PROTOCOL=tcp
      - DEYE_LOGGER_IP_ADDRESS=192.168.30.5
      - DEYE_LOGGER_PORT=502
      - DEYE_LOGGER_MAX_REG_RANGE_LENGTH=5
      - DEYE_METRIC_GROUPS=string
      - DEYE_LOGGER_SERIAL_NUMBER=blabla
      - DEYE_DATA_READ_INTERVAL=5
      - LOG_LEVEL=DEBUG
```
I can, though, access the register using 

`mbpoll -m tcp -a 4 -r 81 -t 4 -c 1 192.168.30.5`

Which gives me:

```
# mbpoll -m tcp -a 4 -r 81 -t 4 -c 1 192.168.30.5
mbpoll 1.0-0 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright © 2015-2019 Pascal JEAN, https://github.com/epsilonrt/mbpoll
This program comes with ABSOLUTELY NO WARRANTY.
This is free software, and you are welcome to redistribute it
under certain conditions; type 'mbpoll -w' for details.

Protocol configuration: Modbus TCP
Slave configuration...: address = [4]
                        start reference = 81, count = 1
Communication.........: 192.168.30.5, port 502, t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave 4... Ctrl-C to stop)
[81]: 	3161
-- Polling slave 4... Ctrl-C to stop)
[81]: 	3146
-- Polling slave 4... Ctrl-C to stop)
[81]: 	3146
-- Polling slave 4... Ctrl-C to stop)
[81]: 	3146
```
This corresponds to the display reading (314W), it's a cloudy day :)

I can also telnet to the modbus gateway (a Waveshare PoE Gateway, [Link](https://www.amazon.de/dp/B0BN61G4VF)) from inside the container.

**To Reproduce**
Setup the project like described above.

**Expected behavior**
Not sure. I guess the stdout prints should look more successful :)

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware (please complete the following information):**
 - Inverter model: DEYE 10k-g06p3 (EU version)
 - Gateway: Waveshare PoE Gateway, [Link](https://www.amazon.de/dp/B0BN61G4VF)

**Software (please complete the following information):**
 - Logger firmware version:
 - deye-inverter-mqtt version: [latest](https://github.com/kbialek/deye-inverter-mqtt/pkgs/container/deye-inverter-mqtt/299335807?tag=latest)
 - config.env (remove passwords before pasting): See above

kbialek, 2024-12-07:
I think Deye simplified the protocol and replaced its custom frame format with plain modbus/tcp.

Please try this:
1. Use beta image `ghcr.io/kbialek/deye-inverter-mqtt:2024.10.1-beta-2`
2. Set `DEYE_LOGGER_PROTOCOL=slimtcp`

And let me know what the result is

lpcvoid, 2024-12-07:
Great, thanks for your help. I will test tomorrow when the inverter is back online.

lpcvoid, 2024-12-08:
Didn't work. Here's the log output:

```
2024-12-08 08:38:40,550 - DeyeDaemon - INFO - Please help me build the list of compatible inverters. https://github.com/kbialek/deye-inverter-mqtt/issues/41
2024-12-08 08:38:40,581 - DeyeConnectorFactory - INFO - Creating Slim Modbus/TCP Logger connector
2024-12-08 08:38:40,582 - DeyeProcessorFactory - INFO - Feature "Report metrics over MQTT": enabled
2024-12-08 08:38:40,582 - DeyeProcessorFactory - INFO - Feature "Set inverter time once online": disabled
2024-12-08 08:38:40,582 - DeyeProcessorFactory - INFO - Feature "Time-of-use configuration over MQTT": disabled
2024-12-08 08:38:40,582 - DeyeActivePowerRegulationEventProcessor - ERROR - Active power regulation sensor not found. Enable appropriate settings metric group.
2024-12-08 08:38:40,582 - DeyeProcessorFactory - INFO - Feature "Active power regulation over MQTT": disabled
2024-12-08 08:38:40,588 - paho.mqtt.client - DEBUG - Sending CONNECT (u1, p1, wr1, wq1, wf1, c1, k60) client_id=b'deye-inverter-2405078035'
2024-12-08 08:38:40,588 - paho.mqtt.client - DEBUG - Received CONNACK (0, 0)
2024-12-08 08:38:41,589 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r1, m1), 'b'deye/status'', ... (6 bytes)
2024-12-08 08:38:41,590 - DeyeMqttClient - INFO - Successfully connected to MQTT Broker located at localhost:1883
2024-12-08 08:38:41,590 - DeyeDaemon - DEBUG - Starting executing the runner at intervals of 5 seconds
2024-12-08 08:38:41,591 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 1)
2024-12-08 08:38:45,591 - DeyeDaemon - DEBUG - Invoking action
2024-12-08 08:38:45,591 - DeyeInverterState - INFO - Reading start
2024-12-08 08:38:45,591 - DeyeInverterState - INFO - Reading registers [metrics group: {'string'}, range: 003c-0040]
2024-12-08 08:38:45,593 - DeyeTcpConnector - DEBUG - Request frame: 0103003c000545c5
2024-12-08 08:38:55,601 - DeyeTcpConnector - DEBUG - Connection response timeout
2024-12-08 08:39:05,609 - DeyeTcpConnector - DEBUG - Connection response timeout
2024-12-08 08:39:15,617 - DeyeTcpConnector - DEBUG - Connection response timeout
2024-12-08 08:39:25,627 - DeyeTcpConnector - DEBUG - Connection response timeout
2024-12-08 08:39:35,633 - DeyeTcpConnector - DEBUG - Connection response timeout
2024-12-08 08:39:35,634 - DeyeTcpConnector - WARNING - Too many connection timeouts
2024-12-08 08:39:35,634 - DeyeInverterState - INFO - Reading registers [metrics group: {'string'}, range: 0041-0045]
2024-12-08 08:39:35,636 - DeyeTcpConnector - DEBUG - Request frame: 010300410005d5dd
2024-12-08 08:39:40,649 - paho.mqtt.client - DEBUG - Sending PINGREQ
2024-12-08 08:39:40,650 - paho.mqtt.client - DEBUG - Received PINGRESP
2024-12-08 08:39:45,645 - DeyeTcpConnector - DEBUG - Connection response timeout
```

kbialek, 2024-12-08:
Could you capture mbpoll traffic using Wireshark?

lpcvoid, 2024-12-08:
Absolutely, here it is.

[502.zip](https://github.com/user-attachments/files/18052774/502.zip)

Here's what I ran, it may help to correspond the frames in the pcap:

```
# mbpoll -m tcp -a 1 -r 81 -t 4 -c 1 192.168.30.5
mbpoll 1.0-0 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright © 2015-2019 Pascal JEAN, https://github.com/epsilonrt/mbpoll
This program comes with ABSOLUTELY NO WARRANTY.
This is free software, and you are welcome to redistribute it
under certain conditions; type 'mbpoll -w' for details.

Protocol configuration: Modbus TCP
Slave configuration...: address = [1]
                        start reference = 81, count = 1
Communication.........: 192.168.30.5, port 502, t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave 1... Ctrl-C to stop)
[81]: 	673
-- Polling slave 1... Ctrl-C to stop)
[81]: 	674
-- Polling slave 1... Ctrl-C to stop)
[81]: 	674
-- Polling slave 1... Ctrl-C to stop)
[81]: 	674
-- Polling slave 1... Ctrl-C to stop)
[81]: 	674
-- Polling slave 1... Ctrl-C to stop)
[81]: 	678
-- Polling slave 1... Ctrl-C to stop)
[81]: 	678
-- Polling slave 1... Ctrl-C to stop)
[81]: 	672
-- Polling slave 1... Ctrl-C to stop)
[81]: 	672
-- Polling slave 1... Ctrl-C to stop)
[81]: 	684
^C--- 192.168.30.5 poll statistics ---
10 frames transmitted, 10 received, 0 errors, 0.0% frame loss

everything was closed.
Have a nice day !
# mbpoll -m tcp -a 1 -r 80 -t 4 -c 1 192.168.30.5
mbpoll 1.0-0 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright © 2015-2019 Pascal JEAN, https://github.com/epsilonrt/mbpoll
This program comes with ABSOLUTELY NO WARRANTY.
This is free software, and you are welcome to redistribute it
under certain conditions; type 'mbpoll -w' for details.

Protocol configuration: Modbus TCP
Slave configuration...: address = [1]
                        start reference = 80, count = 1
Communication.........: 192.168.30.5, port 502, t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave 1... Ctrl-C to stop)
[80]: 	5000
-- Polling slave 1... Ctrl-C to stop)
[80]: 	5000
-- Polling slave 1... Ctrl-C to stop)
[80]: 	5000
-- Polling slave 1... Ctrl-C to stop)
[80]: 	5000
-- Polling slave 1... Ctrl-C to stop)
[80]: 	5000
^C--- 192.168.30.5 poll statistics ---
5 frames transmitted, 5 received, 0 errors, 0.0% frame loss

everything was closed.
Have a nice day !
# mbpoll -m tcp -a 1 -r 64 -t 4 -c 1 192.168.30.5
mbpoll 1.0-0 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright © 2015-2019 Pascal JEAN, https://github.com/epsilonrt/mbpoll
This program comes with ABSOLUTELY NO WARRANTY.
This is free software, and you are welcome to redistribute it
under certain conditions; type 'mbpoll -w' for details.

Protocol configuration: Modbus TCP
Slave configuration...: address = [1]
                        start reference = 64, count = 1
Communication.........: 192.168.30.5, port 502, t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave 1... Ctrl-C to stop)
[64]: 	6541
-- Polling slave 1... Ctrl-C to stop)
[64]: 	6541
-- Polling slave 1... Ctrl-C to stop)
[64]: 	6541
-- Polling slave 1... Ctrl-C to stop)
[64]: 	6541
-- Polling slave 1... Ctrl-C to stop)
[64]: 	6541
^C--- 192.168.30.5 poll statistics ---
5 frames transmitted, 5 received, 0 errors, 0.0% frame loss

everything was closed.
Have a nice day !
```

Thanks for looking into this :)

kbialek, 2024-12-11:
Please try this:

1. Use beta image `ghcr.io/kbialek/deye-inverter-mqtt:2024.12.1-beta-1`
2. Set `DEYE_LOGGER_PROTOCOL=mbtcp`

lpcvoid, 2024-12-12:
Hi, thank you! It works flawlessly now, I see all stats in my mqtt broker, and they all look correct. I bought you a beer, thank you again for your work :)

Issue can be closed.

kbialek, 2024-12-12:
> Hi, thank you! It works flawlessly now, I see all stats in my mqtt broker, and they all look correct. I bought you a beer, thank you again for your work :)

Thanks :beers: Let me know if it's running stable

> Issue can be closed.

Not so fast. I need to document it and make a proper release first.

lpcvoid, 2024-12-13:
> > Issue can be closed.
> 
> Not so fast. I need to document it and make a proper release first.

You where right, I found an issue. Not sure if it's the fault of my inverter, it seems more like a general logic error.

When the inverter shuts itself down (at night), deye-inverter-mqtt doesn't set respective values to 0 in mqtt. Instead, the last (very low) value stays active within mqtt. This means that any software using those mqtt values assume that the inverter is still providing power.

The only value updated for me is `logger_status`, which is set to "offline".

Is this a known issue, or a setup problem on my end?

kbialek, 2024-12-15:
> You were right, I found an issue. Not sure if it's the fault of my inverter, it seems more like a general logic error.
> 
> When the inverter shuts itself down (at night), deye-inverter-mqtt doesn't set respective values to 0 in mqtt. Instead, the last (very low) value stays active within mqtt. This means that any software using those mqtt values assume that the inverter is still providing power.

None of that. What you see are the latest values sent by the inverter before it shuts down.

closed 2025-01-05 by kbialek