kbialek/deye-inverter-mqtt #216: Integrate an EnergyOptimizationSystem as Plugin

finfin82, 2024-12-09:
It would be cool to have some kind of optimization system that can handle with hourly changing prices, respect solar prediction to build a charging discharging + keep load state for the battery.

for forecast source we could use [api.akkudoktor](https://api.akkudoktor.net/)

so what needs to be done
- [ ] get hourly prices
- [ ] get hourly forecast
- [ ] analyse the data
- [ ] find the best battery-state plan 6 different times (fort the time of use)
- [ ] set those plans using the existing time of use implementation

i have a private project doing this. it works once a day 0:01 it runs and sets my states.
how can i send you my code to review and maybe porting this to a plugin?
@kbialek would be awesome to have such a plugin

questions welcome cheers
Andy

kbialek, 2024-12-09:
hi! If you want to make your plugin public, then well, make it public :smile: at github or any other platform of your choice. Then send me a link and I will add it here https://github.com/kbialek/deye-inverter-mqtt?tab=readme-ov-file#list-of-public-plugins

finfin82, 2024-12-10:
well its not a plugin at the moment, its some kind of spaghetti-code, but its working. i was hoping you will find some time to make it a plugin and we can optimize it together.

kbialek, 2024-12-10:
I can maybe review your code and provide some feedback, but will not
actively participate in coding for sure. Sorry, I have no time for that.

finfin82, 2024-12-10:
Thx for your feedback, I will give it a try.

tonobo, 2025-01-06:
To actively support the plugin, you likely need support for bi-directional charging. @kbialek, I'm interested in the SG04LP3 inverters, but they need to support external regulation. I only want to instruct the inverter to either send 10 kW to the battery or 10 kW to the grid, ideally within milliseconds. Can you share some insights?

kbialek, 2025-01-11:
@tonobo Are you really gonna write to the inverter few times a second?

1. I'm unsure if the inverter will manage to adapt to the new settings so quickly.
2. Frequent writes may wear out inverter's settings memory.

Imo, it's best to match inverter model to the specific requirements that you have. I guess you are aware of Time-Of-Use feature that's present in some hybrid inverters.

tonobo, 2025-01-11:
Hey, i've no idea at all to the deye inverters, that's why i'm askin. With victron inverters it is common to do so. So probably it's better to emulate battery and grid meter serial communication.

closed 2025-02-25 by github-actions[bot]