kbialek/deye-inverter-mqtt #220: Additional sensors for string inverter (tested and works)

kingp0dd, 2025-01-29:
These are tested and confirmed to be working:

https://github.com/kbialek/deye-inverter-mqtt/issues/41#issuecomment-2621091175

kbialek, 2025-02-03:
Beta image `ghcr.io/kbialek/deye-inverter-mqtt:2025.02.1-beta-1` is ready for testing

kbialek, 2025-02-03:
Updated docs https://github.com/kbialek/deye-inverter-mqtt/blob/46027403aeb725b20a6b27da174b25195d3a49b7/docs/metric_group_string.md

MQTT topic names are aligned with similar topics used for other inverter types.

kbialek, 2025-02-05:
Fixed image `ghcr.io/kbialek/deye-inverter-mqtt:2025.02.1-beta-2` is ready for testing.

closed 2025-02-05 by kbialek

kingp0dd, 2025-03-23:
> Fixed image `ghcr.io/kbialek/deye-inverter-mqtt:2025.02.1-beta-2` is ready for testing.

Just got back.

I can confirm that `ghcr.io/kbialek/deye-inverter-mqtt:2025.02.1-beta-2`  works. I'm just confused on how to convert grid power. In SolarmanPV, I get value of around -310W, but in beta-2:

[image]