kbialek/deye-inverter-mqtt #225: Serve multiple configurations for one logger

aperiodicchain, 2025-02-11:
Hi,
Below config is deployed in my docker compose stack and works fine.
```.env
# config.env
DEYE_DATA_READ_INTERVAL=15
DEYE_METRIC_GROUPS=deye_sg01hp3,deye_sg01hp3_battery,deye_sg01hp3_bms,deye_sg01hp3_ups,deye_sg01hp3_generator
LOG_LEVEL=INFO
LOG_STREAM=STDOUT
PLUGINS_DIR=plugins
PLUGINS_ENABLED=deye_plugin_stdout_publisher
...
```

Now, I want to also read `DEYE_METRIC_GROUPS=deye_sg01hp3_timeofuse` occasionally (mostly ad-hoc and later also to write time-of-use on adhoc basis as calculated by my energy optimizer) to mqtt, without adding it to my general config.
Like so (I am aware that --run-once does not exist, just to bring my intention across):

```
docker run --rm --env-file time-of-use-config.env ghcr.io/kbialek/deye-inverter-mqtt --run-once
```

`cli_main()` seems to support r/w on a specific register, but won't publish a metric_group to mqtt
Do you have any suggestions how to achieve this (maybe deye_daemon can already do this)?

**Hardware (please complete the following information):**
 - Inverter model: deye_sg01hp3

**Software (please complete the following information):**
 - Logger firmware version: 
 - deye-inverter-mqtt version: own mqtt server in separate docker-stack

kbialek, 2025-02-11:
> Now, I want to also read `DEYE_METRIC_GROUPS=deye_sg01hp3_timeofuse` occasionally (mostly ad-hoc and later also to write time-of-use on adhoc basis as calculated by my energy optimizer) to mqtt, without adding it to my general config.

Can you explain why you want to do it this way?

aperiodicchain, 2025-02-13:
> > Now, I want to also read `DEYE_METRIC_GROUPS=deye_sg01hp3_timeofuse` occasionally (mostly ad-hoc and later also to write time-of-use on adhoc basis as calculated by my energy optimizer) to mqtt, without adding it to my general config.
> 
> Can you explain why you want to do it this way?

I am not determined to do it that way, maybe there is a better one.

I change timeofuse settings manually on the display only very infrequently, a few times per month, so it would be total overkill to query timeofuse group at my standard interval of 15 seconds.
Besides, Deye seems to be struggling to serve the 15 seconds interval, I observe intervals between 15-25 seconds, so I do not want to    enlarge my metrics_goup even more.

I should also add that, next to mqtt, I am using a custom plugin to write all readings as JSONL to ssd so I keep a long term archive of my data for energy aggregation/optimization using Spark, which does not "like" the DEYE_PUBLISH_ON_CHANGE.
The optimization needs to know the timeofuse settings (and later allow writing registers, too).

So for the moment I am looking for an (convenient) ad-hoc way to get all timeofuse metric group.
What I tried so far:

- using a bash script to loop through all timeofuse-registers one-by-one. Works, but not a good design also does not send mqtt/jsonl so not really a solution.
- spanning a 2nd docker loading timeofuse.config, Works, but too much overhead and prone to errors.

Hence, a `--run-once` flag would be optimal, maybe you have a better suggestion?

kbialek, 2025-02-15:
> Besides, Deye seems to be struggling to serve the 15 seconds interval, I observe intervals between 15-25 seconds, so I do not want to enlarge my metrics_goup even more.

Why is this, quite short interval needed? From my experience, I can say that the inverter/logger does not even update its internal state that frequently. Hence, my recommendation is to query metrics at most once a minute.

aperiodicchain, 2025-02-16:
Thanks @kbialek, this frequency helps me identify the power consumer more stable by its pattern (I have many small consumers with similar consumption but characteristic patterns).
However I can confirm that my Deye/logger is serving me updated power values (L1/L2/L3) every 15 seconds (the aggregated energy values are updated only every 10 Minutes which is fine).
I will lower the frequency if necessary but it has been working stable for a year now.

Regarding the timeofuse use case (to query it on ad-hoc basis only) do you have any other suggestion/idea? Else I can try to set a flag in the Config and then add a line to the deye_daemon

```python
if self.__run_once:
    self.__log.debug("Running once")
    for interval_runner in self.__interval_runners:
        interval_runner._IntervalRunner__invoke_action()
    self.stop(None, None)
```
It doesn't seem to be a general applicable use case, so I won't push it back to your repo.

kbialek, 2025-02-16:
What about starting a second instance of the service to use it for timeofuse exclusively? The only problem I see is that the logger's serial number is used [to construct MQTT clientId](https://github.com/kbialek/deye-inverter-mqtt/blob/691ccf9b84d87e41a8bcdd84a5f494bff2aa7fe8/src/deye_mqtt.py#L40), hence an option to overwrite it through a config variable needs to be implemented.

aperiodicchain, 2025-03-12:
Yes, that work. As you mentioned there seems to be a collision with the mqqt_client (shows very briefly "offline") but the json is written to ssd so it is a working solution for me, thanks!

closed 2025-04-26 by github-actions[bot]