kbialek/deye-inverter-mqtt #226: Add New Metric Group for Deye Sun-6K-SG05LP1-EU Inverter

fpucorbier, 2025-02-13:
Hi,

sorry for my english.

I'm using the Deye Sun-6K-SG05LP1-EU inverter and have tried the existing metric groups, but I'm unable to find all the necessary information such as instantaneous production, total production, and total consumption. 

I couldn't find a list of metric groups for this specific inverter model online.

Could you please add a new metric group specifically for this inverter model?

thank you !!!

kbialek, 2025-02-15:
Recently I've added these metrics in response to this [request](https://github.com/kbialek/deye-inverter-mqtt/issues/220):
* Daily Energy Sold
* Total Energy Sold
* Daily Energy Bought
* Total Energy Bought

See https://github.com/kbialek/deye-inverter-mqtt/blob/main/docs/metric_group_string.md

Check if they are working for you.

fpucorbier, 2025-02-17:
Thank you for your quick response!!! Sorry for the late reply, but I've just started using a Deye inverter.

it works for Daily Energy Sold, Total Energy Sold and Daily Energy Bought.
But for Total Energy Bought, I'm getting an inconsistent figure. 
Could you please confirm this for me? 

Thank you.

kbialek, 2025-02-17:
> But for Total Energy Bought, I'm getting an inconsistent figure.
> Could you please confirm this for me?

No, I cannot. I do not own this inverter model. You need to figure it out by yourself.

closed 2025-04-03 by github-actions[bot]