kbialek/deye-inverter-mqtt #228: Problem with wattage over 6553.5 (probably an overflow error)

lpcvoid, 2025-03-02:
**Describe the bug**
If the total power of Inverter is over 6553.5W, then the power is returned as an offset of that value. I guess the uint16 overflows in this case. I bet that `deye-inverter-mqtt` needs to consider another additional register as part of the wattage calculation.

**To Reproduce**
Steps to reproduce the behavior:

Wait until sun shines very brightly :)

This is a log excerpt which shows the behavior:

2025-03-02 10:02:23,311 - DeyeDaemon - DEBUG - Invoking action
2025-03-02 10:02:23,311 - DeyeInverterState - INFO - Reading start
2025-03-02 10:02:23,312 - DeyeInverterState - INFO - Reading registers [metrics group: {'string'}, range: 003c-0045]
2025-03-02 10:02:23,314 - DeyeTcpConnector - DEBUG - Request frame: 03e9000000060103003c000a
2025-03-02 10:02:23,398 - DeyeTcpConnector - DEBUG - Received response frame in 1. attempt: 03e9000000170103140034000000e93733000000000000000000000000
2025-03-02 10:02:23,399 - DeyeInverterState - INFO - Reading registers [metrics group: {'string'}, range: 0046-004f]
2025-03-02 10:02:23,400 - DeyeTcpConnector - DEBUG - Request frame: 03ea0000000601030046000a
2025-03-02 10:02:23,498 - DeyeTcpConnector - DEBUG - Received response frame in 1. attempt: 03ea00000017010314103c100e100c0960094509440061006000611384
2025-03-02 10:02:23,498 - DeyeInverterState - INFO - Reading registers [metrics group: {'string'}, range: 0050-0059]
2025-03-02 10:02:23,500 - DeyeTcpConnector - DEBUG - Request frame: 03eb0000000601030050000a
2025-03-02 10:02:23,599 - DeyeTcpConnector - DEBUG - Received response frame in 1. attempt: 03eb000000170103140f9a00018e7c0000100200018c2c0000ffbbffff
2025-03-02 10:02:23,600 - DeyeInverterState - INFO - Reading registers [metrics group: {'string'}, range: 005a-0063]
2025-03-02 10:02:23,601 - DeyeTcpConnector - DEBUG - Request frame: 03ec000000060103005a000a
2025-03-02 10:02:23,699 - DeyeTcpConnector - DEBUG - Received response frame in 1. attempt: 03ec0000001701031405910591000403e80000000013d5000100100000
2025-03-02 10:02:23,700 - DeyeInverterState - INFO - Reading registers [metrics group: {'string'}, range: 0064-006d]
2025-03-02 10:02:23,702 - DeyeTcpConnector - DEBUG - Request frame: 03ed0000000601030064000a
2025-03-02 10:02:23,800 - DeyeTcpConnector - DEBUG - Received response frame in 1. attempt: 03ed0000001701031400000000000000000000000000000000000018b7
2025-03-02 10:02:23,801 - DeyeInverterState - INFO - Reading registers [metrics group: {'string'}, range: 006e-0074]
2025-03-02 10:02:23,803 - DeyeTcpConnector - DEBUG - Request frame: 03ee000000060103006e0007
2025-03-02 10:02:23,894 - DeyeTcpConnector - DEBUG - Received response frame in 1. attempt: 03ee0000001101030e001119ce005a0000000000000000
2025-03-02 10:02:23,895 - DeyeInverterState - INFO - Reading registers [metrics group: {'string'}, range: 00c6-00cf]
2025-03-02 10:02:23,897 - DeyeTcpConnector - DEBUG - Request frame: 03ef00000006010300c6000a
2025-03-02 10:02:23,991 - DeyeTcpConnector - DEBUG - Received response frame in 1. attempt: 03ef000000170103140000000002083733000000000000000000000000
2025-03-02 10:02:23,992 - DeyeInverterState - INFO - Reading registers [metrics group: {'string'}, range: 00d0-00d2]
2025-03-02 10:02:23,994 - DeyeTcpConnector - DEBUG - Request frame: 03f000000006010300d00003
2025-03-02 10:02:24,077 - DeyeTcpConnector - DEBUG - Received response frame in 1. attempt: 03f000000009010306000000000000
2025-03-02 10:02:24,078 - DeyeInverterState - DEBUG - Production today: 5.2
2025-03-02 10:02:24,078 - DeyeInverterState - DEBUG - Production Total: 1413.1
2025-03-02 10:02:24,078 - DeyeInverterState - DEBUG - Grid L1 Voltage: 240.0
2025-03-02 10:02:24,078 - DeyeInverterState - DEBUG - Grid L1 Current: 9.7
2025-03-02 10:02:24,079 - DeyeInverterState - DEBUG - Grid L1 Power: 2328.0
2025-03-02 10:02:24,079 - DeyeInverterState - DEBUG - Grid L2 Voltage: 237.3
2025-03-02 10:02:24,079 - DeyeInverterState - DEBUG - Grid L2 Current: 9.6
2025-03-02 10:02:24,079 - DeyeInverterState - DEBUG - Grid L2 Power: 2278.1
2025-03-02 10:02:24,079 - DeyeInverterState - DEBUG - Grid L3 Voltage: 237.2
2025-03-02 10:02:24,079 - DeyeInverterState - DEBUG - Grid L3 Current: 9.7
2025-03-02 10:02:24,079 - DeyeInverterState - DEBUG - Grid L3 Power: 2300.8
2025-03-02 10:02:24,079 - DeyeInverterState - DEBUG - Grid L12 Voltage: 415.6
2025-03-02 10:02:24,079 - DeyeInverterState - DEBUG - Grid L23 Voltage: 411.0
2025-03-02 10:02:24,080 - DeyeInverterState - DEBUG - Grid L31 Voltage: 410.8
2025-03-02 10:02:24,080 - DeyeInverterState - DEBUG - Grid Power: 0.0
2025-03-02 10:02:24,080 - DeyeInverterState - DEBUG - Load Power: 0.0
2025-03-02 10:02:24,080 - DeyeInverterState - DEBUG - AC Freq: 50.0
2025-03-02 10:02:24,080 - DeyeInverterState - DEBUG - Uptime: 233.0
2025-03-02 10:02:24,080 - DeyeInverterState - DEBUG - Daily Load Consumption: 5.2
2025-03-02 10:02:24,080 - DeyeInverterState - DEBUG - Total Load Consumption: 1413.1
2025-03-02 10:02:24,080 - DeyeInverterState - DEBUG - Daily Energy Sold: 0.0
2025-03-02 10:02:24,080 - DeyeInverterState - DEBUG - Total Energy Sold: 0.0
2025-03-02 10:02:24,081 - DeyeInverterState - DEBUG - Daily Energy Bought: 0.0
2025-03-02 10:02:24,081 - DeyeInverterState - DEBUG - Total Energy Bought: 0.0
2025-03-02 10:02:24,081 - DeyeInverterState - DEBUG - PV1 Voltage: 632.7
2025-03-02 10:02:24,081 - DeyeInverterState - DEBUG - PV1 Current: 1.7
2025-03-02 10:02:24,081 - DeyeInverterState - DEBUG - PV1 Power: 1075.6
2025-03-02 10:02:24,081 - DeyeInverterState - DEBUG - PV2 Voltage: 660.6
2025-03-02 10:02:24,082 - DeyeInverterState - DEBUG - PV2 Current: 9.0
2025-03-02 10:02:24,082 - DeyeInverterState - DEBUG - PV2 Power: 5945.4
2025-03-02 10:02:24,082 - DeyeInverterState - DEBUG - PV3 Voltage: 0.0
2025-03-02 10:02:24,082 - DeyeInverterState - DEBUG - PV3 Current: 0.0
2025-03-02 10:02:24,082 - DeyeInverterState - DEBUG - PV3 Power: 0.0
2025-03-02 10:02:24,082 - DeyeInverterState - DEBUG - PV4 Voltage: 0.0
[... 145 lines cut ...]

**Hardware (please complete the following information):**
 - Inverter model: Deye SUN-10K-G06

**Software (please complete the following information):**
 - Logger firmware version: none
 - deye-inverter-mqtt version: `head`
 - config.env (remove passwords before pasting):
    environment:
      - MQTT_HOST=localhost
      - MQTT_PORT=1883
      - MQTT_USERNAME=xxx
      - MQTT_PASSWORD=xxx
      - MQTT_TOPIC_PREFIX=deye
      - DEYE_LOGGER_PROTOCOL=mbtcp
      - DEYE_LOGGER_IP_ADDRESS=192.168.30.5
      - DEYE_LOGGER_PORT=502
      - DEYE_LOGGER_MAX_REG_RANGE_LENGTH=10
      - DEYE_METRIC_GROUPS=string
      - DEYE_LOGGER_SERIAL_NUMBER=xxx
      - DEYE_DATA_READ_INTERVAL=10
      - LOG_LEVEL=DEBUG

**Additional context**
Add any other context about the problem here.

kbialek, 2025-03-16:
Are you able to figure out which register it is?

lpcvoid, 2025-03-16:
I will try in the next few days if it's sunny enough and get back to you.

lpcvoid, 2025-03-17:
I had a brief moment in time for testing today.

So it seems that the address for current power is 80, with a 32bit value being read (so I guess register 80 + 81, each 16 bit). Then you need to swap second uint16 with the first: ABCD -> CDAB  (each letter being 8bit).

closed 2025-05-01 by github-actions[bot]