kbialek/deye-inverter-mqtt #229: DEYE SUN-12K-SG02LP1-EU-AM3 - Different factor in readings

aletmp, 2025-03-05:
**Hardware (please complete the following information):**
 - Inverter model: DEYE SUN-12K-SG02LP1-EU-AM3

**Software (please complete the following information):**
 - Logger firmware version: LSW3_32U_5406_1.07
 - deye-inverter-mqtt version:
 - config.env (remove passwords before pasting):
LOG_LEVEL=INFO
LOG_STREAM=STDOUT
DEYE_DATA_READ_INTERVAL=3
DEYE_METRIC_GROUPS=deye_hybrid,deye_hybrid_battery
DEYE_FEATURE_MQTT_PUBLISHER=true

The issue is the following, in this Deye model, I've identified so far 2 issues about the scale, where they are different:

name: "Total Grid Power"
      class: "power"
      state_class: "measurement"
      uom: "W"
      scale: 10 // this instead of 1
      rule: 2
      registers: [0x00A9]
      icon: 'mdi:transmission-tower'

    - name: "Battery Current"
      class: "current"
      state_class: "measurement"
      uom: "A"
      scale: 0.1 // this instead of 0.01
      rule: 2
      registers: [0x00BF]
      icon: 'mdi:battery'

How do I modify the code in the running docker image in my local so it can apply?

kbialek, 2025-03-05:
> How do I modify the code in the running docker image in my local so it can apply?

That's not possible. I'd rather prefer to add a new metric group for sg02lp1 as a clone of deye_hybrid with appropriate scale factors.

Similar issue is reported here https://github.com/davidrapan/ha-solarman/issues/281

aletmp, 2025-03-06:
I agree we probably need a new metric for the SG02LP1 series. However, I believe is a potential problem from Deye, where they, for some reason, messed around with the modbus for this new models. Previous version of Logger firmware fixed a scale issue already in the Inverter itself. The issue was in the "Load Power (W)"

I will investigate in detail in the next few days what modbus records are causing problems. Sould I update that information in this thread or in the existing one you linked?

kbialek, 2025-03-06:
> Sould I update that information in this thread or in the existing one you linked?

Here please

closed 2025-04-20 by github-actions[bot]