kbialek/deye-inverter-mqtt #231: Is changing active_power_regulation still working?

ferby09, 2025-03-13:
**Hardware (please complete the following information):**
 - Inverter model: SUN600G3/EU230

**Software (please complete the following information):**
 - Logger firmware version: MW3_SSL_5408_1.0B
 - deye-inverter-mqtt version: latest
 - config.env (remove passwords before pasting):
 `DEYE_LOGGER_IP_ADDRESS=10.xxx.xxx.27
# DEYE_LOGGER_PORT=8899
DEYE_LOGGER_SERIAL_NUMBER=4176xxxxxx
# DEYE_LOGGER_PROTOCOL=tcp
# DEYE_LOGGER_MAX_REG_RANGE_LENGTH=256

MQTT_HOST=10.xxx.xxx.4
# MQTT_PORT=1883
# MQTT_USERNAME=
# MQTT_PASSWORD=
# MQTT_TOPIC_PREFIX=deye
# MQTT_TLS_ENABLED=true
MQTT_TLS_INSECURE=true
# MQTT_TLS_CA_CERT_PATH=certs/ca.crt
# MQTT_TLS_CLIENT_CERT_PATH=certs/client.crt
# MQTT_TLS_CLIENT_KEY_PATH=certs/client.key

LOG_LEVEL=DEBUG
# LOG_STREAM=STDOUT
# PLUGINS_DIR=plugins
# PLUGINS_ENABLED=deye_plugin_sample,deye_plugin_custom
# DEYE_DATA_READ_INTERVAL=60
DEYE_PUBLISH_ON_CHANGE=true
# DEYE_PUBLISH_ON_CHANGE_MAX_INTERVAL=360
DEYE_METRIC_GROUPS=micro,settings_micro

DEYE_FEATURE_MQTT_PUBLISHER=true
# DEYE_FEATURE_TIME_OF_USE=false
DEYE_FEATURE_ACTIVE_POWER_REGULATION=true
# DEYE_FEATURE_SET_TIME=false`

The mosquitto broker is running on my Unraid server and I'm trying to publish with a raspberry pi where deye-inverter-mqtt is running in docker.
The command I'm using is 
`mosquitto_pub -h 10.xxx.xxx.4 -t deye/settings/active_power_regulation -m "120"`
The command doesn't return an error.

My docker log looks like this after starting the container and publishing:
`2025-03-13 10:45:38,613 - DeyeDaemon - INFO - Please help me build the list of compatible inverters. https://github.com/kbialek/deye-inverter-mqtt/issues/41
2025-03-13 10:45:38,618 - DeyeConnectorFactory - INFO - Creating proprietary Deye Modbus/TCP Logger connector
2025-03-13 10:45:38,625 - DeyeProcessorFactory - INFO - Feature "Report metrics over MQTT": enabled
2025-03-13 10:45:38,627 - DeyeProcessorFactory - INFO - Feature "Set inverter time once online": disabled
2025-03-13 10:45:38,630 - DeyeProcessorFactory - INFO - Feature "Time-of-use configuration over MQTT": disabled
2025-03-13 10:45:38,633 - DeyeProcessorFactory - INFO - Feature "Active power regulation over MQTT": enabled
2025-03-13 10:45:38,664 - paho.mqtt.client - DEBUG - Sending CONNECT (u0, p0, wr1, wq1, wf1, c1, k60) client_id=b'deye-inverter-4176773089'
2025-03-13 10:45:38,681 - paho.mqtt.client - DEBUG - Received CONNACK (0, 0)
2025-03-13 10:45:38,684 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r1, m1), 'b'deye/status'', ... (6 bytes)
2025-03-13 10:45:38,686 - DeyeMqttClient - INFO - Successfully connected to MQTT Broker located at 10.20.30.4:1883
2025-03-13 10:45:38,689 - DeyeMqttClient - INFO - Subscribing to topic: deye/settings/active_power_regulation/command
2025-03-13 10:45:38,690 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 1)
2025-03-13 10:45:38,693 - paho.mqtt.client - DEBUG - Sending SUBSCRIBE (d0, m2) [(b'deye/settings/active_power_regulation/command', 1)]
2025-03-13 10:45:38,703 - DeyeDaemon - DEBUG - Starting executing the runner at intervals of 60 seconds
2025-03-13 10:45:38,707 - paho.mqtt.client - DEBUG - Received SUBACK
2025-03-13 10:45:44,710 - DeyeDaemon - DEBUG - Invoking action
2025-03-13 10:45:44,711 - DeyeInverterState - INFO - Reading start
2025-03-13 10:45:44,712 - DeyeInverterState - INFO - Reading registers [metrics group: {'micro'}, range: 003c-0074]
2025-03-13 10:45:44,720 - DeyeTcpConnector - DEBUG - Request frame: a5170010450000e17ff4f80200000000000000000000000000000103003c003945d44c15
2025-03-13 10:45:44,912 - DeyeTcpConnector - DEBUG - Received response frame in 1. attempt: a5850010150014e17ff4f80201e214b2005201000077a220670103720005000000002c9e0000000200020000000014fe000016090000096000000000000a00000000138800000000000000000000000009a60000000000000c44000000000000000000000000000000000000000000000000000000000000000000000000015000240149002600000000000000003cd82815
2025-03-13 10:45:44,915 - DeyeInverterState - INFO - Reading registers [metrics group: {'settings_micro', 'settings'}, range: 0028-0028]
2025-03-13 10:45:44,925 - DeyeTcpConnector - DEBUG - Request frame: a5170010450000e17ff4f80200000000000000000000000000000103002800010402ed15
2025-03-13 10:45:44,994 - DeyeTcpConnector - DEBUG - Received response frame in 1. attempt: a5150010150015e17ff4f80201e214b2005201000077a220670103020064b9af0b15
2025-03-13 10:45:44,998 - DeyeInverterState - DEBUG - Production today: 0.5
2025-03-13 10:45:45,000 - DeyeInverterState - DEBUG - Production Total: 1142.2
2025-03-13 10:45:45,001 - DeyeInverterState - DEBUG - Grid L1 Voltage: 240.0
[... 111 lines cut ...]
2025-03-13 10:45:45,423 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv4/power', value: '0.0'
2025-03-13 10:45:45,426 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m28), 'b'deye/dc/pv4/power'', ... (3 bytes)
2025-03-13 10:45:45,432 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 28)
2025-03-13 10:45:45,435 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv4/day_energy', value: '0.0'
2025-03-13 10:45:45,439 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m29), 'b'deye/dc/pv4/day_energy'', ... (3 bytes)
2025-03-13 10:45:45,444 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 29)
2025-03-13 10:45:45,447 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv4/total_energy', value: '0.0'
2025-03-13 10:45:45,448 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m30), 'b'deye/dc/pv4/total_energy'', ... (3 bytes)
2025-03-13 10:45:45,503 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 30)
2025-03-13 10:45:45,505 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/total_power', value: '246.0'
2025-03-13 10:45:45,506 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m31), 'b'deye/dc/total_power'', ... (5 bytes)
2025-03-13 10:45:45,511 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 31)
2025-03-13 10:45:45,513 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/operating_power', value: '0.0'
2025-03-13 10:45:45,514 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m32), 'b'deye/operating_power'', ... (3 bytes)
2025-03-13 10:45:45,519 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 32)
2025-03-13 10:45:45,520 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/ac/active_power', value: '247.0'
2025-03-13 10:45:45,521 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m33), 'b'deye/ac/active_power'', ... (5 bytes)
2025-03-13 10:45:45,526 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 33)
2025-03-13 10:45:45,527 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/radiator_temp', value: '21.4'
2025-03-13 10:45:45,532 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m34), 'b'deye/radiator_temp'', ... (4 bytes)
2025-03-13 10:45:45,539 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 34)
2025-03-13 10:45:45,540 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/settings/active_power_regulation', value: '100.0'
2025-03-13 10:45:45,541 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m35), 'b'deye/settings/active_power_regulation'', ... (5 bytes)
2025-03-13 10:45:45,546 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 35)
2025-03-13 10:45:45,547 - DeyeInverterState - INFO - Reading completed`

ferby09, 2025-03-16:
Can I set active_power_regulation by directly addressing the register in the docker?
I tried the following:
```bash
sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 40
2025-03-16 12:29:04,009 - root - INFO - Connecting to logger at IP 10.20.30.27
2025-03-16 12:29:04,015 - DeyeConnectorFactory - INFO - Creating proprietary
Deye Modbus/TCP Logger connector
2025-03-16 12:29:04,137 - DeyeTcpConnector - DEBUG - Request frame: a5170010450000e17ff4f80200000000000000000000000000000103002800010402ed15
2025-03-16 12:29:04,199 - DeyeTcpConnector - DEBUG - Received response frame
in 1. attempt: a5150010150002e17ff4f80201b806b40012000000000000000103020064b9afe115
2025-03-16 12:29:04,202 - root - INFO - int: 100, l: 100, h: 0
```

Is this the right register?

ferby09, 2025-03-16:
Seems like
`sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt w 40 120`
did work.

closed 2025-03-16 by ferby09

kbialek, 2025-03-16:
Cool you found the solution.

Meanwhile I checked if active power regulation is still working over mqtt, and yes, it is.
```
❯ mosquitto_sub -F '@Y-@m-@dT@H:@M:@S@z : %r%q : %t : %p' -t 'home/pv/1/8/#'
2025-03-16T17:18:07+0100 : 00 : home/pv/1/8/logger_status : online
2025-03-16T17:18:07+0100 : 00 : home/pv/1/8/settings/active_power_regulation : 109.0
2025-03-16T17:18:46+0100 : 00 : home/pv/1/8/settings/active_power_regulation/command : 108
...
2025-03-16T17:20:07+0100 : 00 : home/pv/1/8/settings/active_power_regulation : 108.0
2025-03-16T17:20:15+0100 : 00 : home/pv/1/8/settings/active_power_regulation/command : 109
...
2025-03-16T17:21:07+0100 : 00 : home/pv/1/8/settings/active_power_regulation : 109.0
```