kbialek/deye-inverter-mqtt #232: broken template conversion

simcup, 2025-03-15:
hey,

for some reason, every time i ran systemd/install.sh the created unit file was missing an "/" between the working directory and the path to whatever was referenced

might wanne check that out, was installing it bare metal on a raspberry pi 4 (bookworm soruces), but seems to me that is something in the converting of the template

kbialek, 2025-03-16:
Thanks for reminding me that I should remove this systemd integration from the repo. It was contributed by one of the users. I'm not willing to actively support it.

simcup, 2025-03-16:
well technically it's two (see my other issue) typos, which is why i didn't fork anything. for this one its install.sh line 56, put a "/" after the ${DEYE_DIR}

kbialek, 2025-03-16:
Sorry, already removed. I'd be happy to link a gist describing this installation method, if you are okay with maintaining it on your own.

closed 2025-04-30 by github-actions[bot]