kbialek/deye-inverter-mqtt #235: DeyeMqttPublisher - ERROR - Unknown MQTT publishing error: Message publish failed: The client is not currently connected.

solarc3po, 2025-03-30:
After update to new docker image I get without changing docker config or anything the foolowing error (see attached). 
In MQTT Explorer the dey inverter is for 2-4 sec online then Status=offline
Some DC data send but not more

** Docker config
  DeyeM01:
    image: basdorf_deye_20250322 #badorf_20241206_deye #badorf_20240802c_deye #basdorf_deye_20240710 
    restart: always
    networks:
      - iot-net
    labels:
        - "conky=no"
        - "stack=iot"            
    environment:
      - DEYE_LOGGER_IP_ADDRESS=192.172.1.135
      - DEYE_LOGGER_PORT=8899
      - DEYE_LOGGER_SERIAL_NUMBER=1234567890 #change here

      - MQTT_HOST=192.168.100.148
      - MQTT_PORT=1883
      - MQTT_USERNAME=mqtt
      - MQTT_PASSWORD=super pw
      - MQTT_TOPIC_PREFIX=wr/m01

      - LOG_LEVEL=INFO
      - DEYE_DATA_READ_INTERVAL=30
      - DEYE_METRIC_GROUPS=deye_sg04lp3, deye_sg04lp3_battery

**Hardware (please complete the following information):**
Deye deye_sg04lp3 10K

**Software (please complete the following information):**

get in docker following error

2025-03-30 16:22:30,283 - DeyeDaemon - INFO - Please help me build the list of compatible inverters. https://github.com/kbialek/deye-inverter-mqtt/issues/41
2025-03-30 16:22:30,283 - DeyeConnectorFactory - INFO - Creating Modbus/TCP Logger connector
2025-03-30 16:22:30,283 - DeyeProcessorFactory - INFO - Feature "Report metrics over MQTT": enabled
2025-03-30 16:22:30,283 - DeyeProcessorFactory - INFO - Feature "Set inverter time once online": disabled
2025-03-30 16:22:30,283 - DeyeProcessorFactory - INFO - Feature "Time-of-use configuration over MQTT": disabled
2025-03-30 16:22:30,283 - DeyeActivePowerRegulationEventProcessor - ERROR - Active power regulation sensor not found. Enable appropriate settings metric group.
2025-03-30 16:22:30,283 - DeyeProcessorFactory - INFO - Feature "Active power regulation over MQTT": disabled
2025-03-30 16:22:31,289 - DeyeMqttClient - INFO - Successfully connected to MQTT Broker located at 192.168.100.148:1883
2025-03-30 16:22:52,290 - DeyeInverterState - INFO - Reading start
2025-03-30 16:22:52,290 - DeyeInverterState - INFO - Reading registers [metrics group: {'deye_sg04lp3'}, rnge: 0202-022e]
2025-03-30 16:22:52,463 - DeyeInverterState - INFO - Reading registers [metrics group: {'deye_sg04lp3_battery'}, range: 024a-024f]
2025-03-30 16:22:53,554 - DeyeInverterState - INFO - Reading registers [metrics group: {'deye_sg04lp3'}, range: 0256-027c]
2025-03-30 16:22:54,714 - DeyeInverterState - INFO - Reading registers [metrics group: {'deye_sg04lp3'}, range: 02a0-02a7]
2025-03-30 16:22:55,804 - DeyeMqttPublisher - ERROR - Unknown MQTT publishing error: Message publish failed: The client is not currently connected.
2025-03-30 16:22:55,804 - DeyeInverterState - INFO - Reading completed

kbialek, 2025-03-30:
Be sure you have exactly one instance of the service running

solarc3po, 2025-03-31:
yes, the stack on the other Host was still running. Thanks.

SebStaeubert, 2025-03-31:
I have the same issue with 2 micro inverters (deye sun600, sun800).
Sometimes this message:
```
DeyeMqttPublisher - ERROR - [1] Unknown MQTT publishing error: Message publish failed: The client is not currently c
onnected.
```

And deye/status is almost everytime "offline".
```console
root@openhabian:~# mosquitto_sub -h 192.168.18.165 -u openhabian -P KVze7nT3 -v -t deye/status
deye/status offline
deye/status offline
deye/status offline
deye/status offline
deye/status offline
deye/status offline
deye/status offline
deye/status offline
```
Very rare the status changes to "online".

Perhaps because of this message:
```
deye-mqtt-1  | 2025-03-31 09:48:07,708 - DeyeInverterState - INFO - [1] Reading start
deye-mqtt-1  | 2025-03-31 09:48:07,708 - DeyeInverterState - INFO - [1] Reading registers [metrics group: {'micro'}, range: 003c-0074]
deye-mqtt-1  | 2025-03-31 09:48:07,937 - DeyeInverterState - INFO - [1] No changes found in received data, or the logger is offline, skipping
deye-mqtt-1  | 2025-03-31 09:48:07,938 - DeyeInverterState - INFO - [1] Reading completed
```

kbialek, 2025-03-31:
@SebStaeubert Share your config file(s) here.

SebStaeubert, 2025-04-01:
```ini
MQTT_HOST=192.168.--.---
MQTT_PORT=1883
MQTT_USERNAME=----
MQTT_PASSWORD=----
MQTT_TOPIC_PREFIX=deye

DEYE_DATA_READ_INTERVAL=30
DEYE_PUBLISH_ON_CHANGE=true
DEYE_PUBLISH_ON_CHANGE_MAX_INTERVAL=330
DEYE_METRIC_GROUPS=micro

DEYE_FEATURE_MQTT_PUBLISHER=true
DEYE_FEATURE_TIME_OF_USE=true
DEYE_FEATURE_ACTIVE_POWER_REGULATION=false
DEYE_FEATURE_MULTI_INVERTER_DATA_AGGREGATOR=true

DEYE_LOGGER_COUNT=2

# SUN600
DEYE_LOGGER_1_IP_ADDRESS=192.168.--.---
DEYE_LOGGER_1_SERIAL_NUMBER=41--------

# SUN800
DEYE_LOGGER_2_IP_ADDRESS=192.168.--.---
DEYE_LOGGER_2_SERIAL_NUMBER=41--------
```

kbialek, 2025-04-02:
Looks okay. I also assume that you have a single instance running. Please upload the logs collected for a whole day.

SebStaeubert, 2025-04-03:
I added the log.txt, which startet at 31.03.2025 and ended 01.04.2025.
BUT, the container still runs and also work (reads data and sends it via mqtt), but the logging freezes! Probably as a cause of an exception.

[log.txt](https://github.com/user-attachments/files/19582025/log.txt)

Bhoft, 2025-04-18:
Hi i just wanted to also create an issue, which might be related to this.
I have used an older version `ghcr.io/kbialek/deye-inverter-mqtt:2024.09.1-beta-4` for a longer time and didn't watched my solar charts during the winter days..

Now as the weather gets better and better i have checked my solar view in home assistant more frequently.
And i always got the issue that after some days the mqtt topic `deye/status` stayed on offline.

I haven't noticed it when i used my own mqtt sensors. But since i am using the [deye-mqtt-ha-plugin](https://sr.ht/~carstengrohmann/deye-mqtt-ha-plugin/) the issues started.

I have updated to the latest version of the deye-inverter-mqtt and deye-mqtt-ha-plugin.
When i restart the portainer everything seems to work for multiple days until it doesn't.

The issue is the plugin has e.g. this setup for his sensors
``` 
unique_id: deye_mqtt_inverter_1234_pv1_power
force_update: true
device_class: power
availability_topic: deye/status
state_topic: deye/dc/pv1/power
device:
  identifiers:
    - deye_inverter_mqtt_1234
  manufacturer: deye
  model: 2404020884 SN:1234
  serial_number: '1234'
  sw_version: deye-inverter-mqtt with HA Discovery Plugin version 2025-03-23
  name: deye Inverter MQTT
expire_after: 600
state_class: measurement
unit_of_measurement: W
name: PV1 Power
```

All sensors are using the availability topic `deye/status` which should be online when the inverter is online.
The strange thing is after some days it stays `offline` when the sun goes up and powers the inverter.

Sensor data however are correctly transfered via mqtt. Only the status stays offline.
Thats why my own mqtt sensors which used the logger_status as availability_topic weren't affected.

When I restart my portainer it will work again for some days.

[image]

These are my env settings in portainer
```
DEYE_DATA_FILTERS 	ignore_zeroed_frames
DEYE_DATA_READ_INTERVAL 	60
DEYE_FEATURE_ACTIVE_POWER_REGULATION 	true
DEYE_FEATURE_SET_TIME 	true
DEYE_FEATURE_TIME_OF_USE 	false
DEYE_HA_PLUGIN_EXPIRE_AFTER 	600
DEYE_HA_PLUGIN_HA_MQTT_PREFIX 	homeassistant
DEYE_HA_PLUGIN_INVERTER_MANUFACTURER 	deye
DEYE_HA_PLUGIN_INVERTER_MODEL 	2404020884
DEYE_LOGGER_IP_ADDRESS 	192.168.5.149
DEYE_LOGGER_PROTOCOL 	tcp
DEYE_LOGGER_SERIAL_NUMBER 	1234
DEYE_METRIC_GROUPS 	micro,settings_micro
DEYE_PUBLISH_ON_CHANGE 	true
DEYE_PUBLISH_ON_CHANGE_MAX_INTERVAL 	360
LANG 	C.UTF-8
LOG_LEVEL 	DEBUG
MQTT_HOST 	192.168.5.140
MQTT_PASSWORD 	testpassword
MQTT_PORT 	1883
MQTT_TOPIC_PREFIX 	deye
MQTT_USERNAME 	mqtt-user
PATH 	/usr/local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
PLUGINS_ENABLED 	deye_plugin_ha_discovery
TZ 	Europe/Berlin
```

The issue which I also have is I can find multiple log entries for `Publishing message. topic: 'deye/logger_status'` which can be used as a filter in the portainer log to display related data.
But couldn't find any for `Publishing message. topic: 'deye/status'`.

After container restart however i could find this:
```
2025-04-18 18:50:16,812 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r1, m1), 'b'deye/status'', ... (6 bytes)
```
But it would be easier if the status also gets his Publishing message for debuging.
Btw i couldn't find this entry in the logs of the last 5 days where it didn't work.
So it seems that the status isn't published correctly.

Bhoft, 2025-04-19:
I have investigated my issue more closely.
The `deye/status` will only be set to "online" when the mqtt client is connected.
So in my case it's stange that it wasn't connected but somehow still updated the other sensor data via mqtt successfully (so it must be connected otherwise the tool couldn't send the inverter data).

Checking the mqtt client code i see that the status should be set to offline after an 3 seconds timeout.
```
        self.__status_topic = f"{config.mqtt.topic_prefix}/{config.mqtt.availability_topic}"
        self.__mqtt_client.will_set(self.__status_topic, "offline", retain=True, qos=1)
        self.__config = config.mqtt
        self.__mqtt_timeout = 3  # seconds
```

And `on connect` to `online`
```
            self.__mqtt_client.publish(self.__status_topic, "online", retain=True, qos=1)
            self.__log.info(
                "Successfully connected to MQTT Broker located at %s:%d", self.__config.host, self.__config.port
            )
```

Could it be that perhaps an update and or restart of home assitant produced the state that the new mqtt client connected first (and set status = online) before the old client timed out and set the status back to `offline`?

I have checked in Home assistant and its somehow strange that on day was an core update
[image]

Other things which might be related are maybe backups of my portainer system.
I have added now a notification for me which informs me when the status is offline. 
So i could hopefully narrow the issue down.

kbialek, 2025-04-19:
@Bhoft 
1. Are there any exceptions in the log?
2. What do you use as MQTT broker? I guess it's mosquitto, but please confirm it.
3. How is your MQTT broker managed? Does it start/stop together with HA?

Bhoft, 2025-04-19:
@kbialek 
1. no i can't find anything
```
sudo journalctl -g "Failed to connect to MQTT Broker located" CONTAINER_NAME=deye-inverter-mqtt-deye-mqtt
-- No entries --

sudo journalctl -g "MQTT outgoing queue is full" CONTAINER_NAME=deye-inverter-mqtt-deye-mqtt
-- No entries --

sudo journalctl -g "Unknown MQTT publishing error" CONTAINER_NAME=deye-inverter-mqtt-deye-mqtt
-- No entries --

sudo journalctl -g "MQTT connection error" CONTAINER_NAME=deye-inverter-mqtt-deye-mqtt 
-- No entries --
```

2. mosquitto installed as addon in HA
3. installed as addon with automatically start on boot

When I restart HA however through the Developer Tools Page I can't see any restarting of mosquitto and therefore also no re-login on the mosquitto protocol log which is accessible through the addon page.
But i guess the addon docker container for the mosquitto addon isn't restarted then.
So it might be related to a full system restart.

When I restart the mosquitto addon I see the reconnect of the client but the `deye/status` stays at `online`.

from mosquitto log:
```
[18:58:33] INFO: Starting mosquitto MQTT broker...
2025-04-19 18:58:41: New client connected from 192.168.5.150:54105 as deye-inverter-3848502278 (p2, c1, k60, u'mqtt-user').
```

What I find somehow strange I couldn't see the log message of the new connection from the docker container of the deye-inverter-mqtt instance.
Only log messages from earlier restarts of the complete container.
I would have guessed that when i restart mosquitto completly that this message should have veen logged.
```
root@docker:~# sudo journalctl -g "Successfully connected to MQTT Broker" CONTAINER_NAME=deye-inverter-mqtt-deye-mqtt
Apr 18 19:28:30 docker 73983827c7b8[94420]: 2025-04-18 19:28:30,271 - DeyeMqttClient - INFO - Successfully connected to MQTT Broker located at 192.168.5.140:1883
Apr 19 13:15:23 docker 5c4226f289f1[94420]: 2025-04-19 13:15:23,025 - DeyeMqttClient - INFO - Successfully connected to MQTT Broker located at 192.168.5.140:1883
Apr 19 13:16:31 docker 4d9dade46021[94420]: 2025-04-19 13:16:31,360 - DeyeMqttClient - INFO - Successfully connected to MQTT Broker located at 192.168.5.140:1883
```

Instead I only found this on reconnect:
```
Apr 19 18:58:14 docker 4d9dade46021[94420]: 2025-04-19 18:58:14,212 - paho.mqtt.client - DEBUG - Sending PINGREQ
Apr 19 18:58:14 docker 4d9dade46021[94420]: 2025-04-19 18:58:14,213 - paho.mqtt.client - DEBUG - Received PINGRESP
Apr 19 18:58:27 docker 4d9dade46021[94420]: 2025-04-19 18:58:27,217 - paho.mqtt.client - DEBUG - Connection failed, retrying
Apr 19 18:58:29 docker 4d9dade46021[94420]: 2025-04-19 18:58:29,219 - paho.mqtt.client - DEBUG - Connection failed, retrying
Apr 19 18:58:33 docker 4d9dade46021[94420]: 2025-04-19 18:58:33,221 - paho.mqtt.client - DEBUG - Connection failed, retrying
Apr 19 18:58:41 docker 4d9dade46021[94420]: 2025-04-19 18:58:41,223 - paho.mqtt.client - DEBUG - Sending CONNECT (u1, p1, wr1, wq1, wf1, c1, k60) client_id=b'deye-inverter-3848502278'
Apr 19 18:58:41 docker 4d9dade46021[94420]: 2025-04-19 18:58:41,224 - paho.mqtt.client - DEBUG - Received CONNACK (0, 0)
```

Which means if the connection times out and the status = offline will be set and the reconnect is done differently as using the "connect" method (otherwise i would see a "Successfully connected to MQTT Broker located" log entry). Then the status might stay on "offline".
But thats only a guess...
I am also not sure how i could reproduce the issue.

- It doesn't happen when i restart mosquitto
- it can't happen when I stop mosquitto i guess because how should the offline state be set when the mqtt broker is not online.
but maybe you have some idea what could cause this.

kbialek, 2025-04-19:
To be sure, as per [the docs](https://github.com/kbialek/deye-inverter-mqtt?tab=readme-ov-file#additional-mqtt-topics) there are two status topics. To which one are you referring to?

Logger status is updated on each data read interval. Here is a snippet from my logs:
```
2025-04-19 17:35:31,157 - DeyeMqttClient - INFO - Logger is online
2025-04-19 17:36:25,145 - DeyeMqttClient - INFO - Logger is online
2025-04-19 17:37:31,952 - DeyeMqttClient - INFO - Logger is online
2025-04-19 17:38:25,147 - DeyeMqttClient - INFO - Logger is online
```

It's logged just right after the status is sent over MQTT https://github.com/kbialek/deye-inverter-mqtt/blob/43b980e0fa8c9dbf562fe2057a8f07b486affe4e/src/deye_mqtt.py#L131-L135

This means, that also after MQTT broker restart, the logger status should be updated.

Bhoft, 2025-04-19:
> To be sure, as per [the docs](https://github.com/kbialek/deye-inverter-mqtt?tab=readme-ov-file#additional-mqtt-topics) there are two status topics. To which one are you referring to?

the `deye/status` topic which is set on `connect` to the mqtt server.

```
 self.__status_topic = f"{config.mqtt.topic_prefix}/{config.mqtt.availability_topic}"
 self.__mqtt_client.will_set(self.__status_topic, "offline", retain=True, qos=1)
```

or in other words this part from the readme

#### **Availability topic**
Reports deye-inverter-mqtt service status (not the inverter/logger status):
* `online` - when the service is connected to the MQTT broker
* `offline` - when the service is disconnected from the MQTT broker

closed 2025-04-22 by kbialek