kbialek/deye-inverter-mqtt #239: No response to topic settings/active_power_regulation/command after reconnection to the MQTT server

chk81, 2025-04-29:
First of all, thank you for the great project :)

If the client loses the connection to the MQTT server, for example if the server is restarted, it no longer responds to the topic settings/active_power_regulation/command
When the container is restarted, everything works fine again.

I have the DEYE_METRIC_GROUPS: micro,settings_micro

kbialek, 2025-04-29:
Yeah, that's a consequence of connecting with clean_session set to true https://github.com/kbialek/deye-inverter-mqtt/blob/9e55c7a76d004d82b8b922f877a3da81c5805a07/src/deye_mqtt.py#L40

Can be also the case when the broker runs in a stateless mode.

Should be easy to fix

kbialek, 2025-04-29:
@chk81 Try [this](https://github.com/kbialek/deye-inverter-mqtt/pkgs/container/deye-inverter-mqtt/405280705?tag=2025.04.2-beta-3) beta image and let me know if the problem is solved

chk81, 2025-04-30:
thanks for your quick help. the new image solves my problem :)

closed 2025-05-03 by kbialek