kbialek/deye-inverter-mqtt #245: Please add PV3 values to "deye_hybrid" group  to suit SUN-10/12K-SG02LP1-AU-AM3

mobbster, 2025-07-19:
**Is your feature request related to a problem? Please describe.**
My new SUN-10K-SG02LP1-AU-AM3 is generating registers for PV3 power, voltage and current (at locations 188, 113 and 114 respectively), but the "deye_sensors_deye_hybrid.py" does not have those register definitions (I have added them in manually, and they work fine).

Other than those missing values, the SUN-10K-SG02LP1-AU-AM3 seems to work perfectly with the "deye_hybrid", "deye_hybrid_battery" and "deye_hybrid_bms" groups (for my use case, anyway)

**Describe the solution you'd like**
Please add those register definitions - I don't think that adding those would have any adverse side-effects with other inverters

Many thanks for the great project!

mobbster, 2025-07-19:
For my home automation system, I was also wanting to observe the "load" value (both live watts and daily kWh).

I am guessing it is simply "Solar + net grid (import-export) + net battery (discharge-charge)", which tracks pretty well with my load (although there are probably some heat losses that are hopefully not too significant).

I am happy to make those calculations myself in my system, but do you think it might be useful to add them as calculated MQTT values to help others? No problem if you disagree - just an idea.

koira, 2025-07-22:
I'd very much like to try this on my SUN-10K-SG02LP1-AU-AM3 but I struggle to patch according to @mobbster 's file and then create a docker-image from that patched version to run. when will there be a docker-image available that incorporates the above patch? Happy to contribute by verifying and reporting compatibility to my own deye system. thanks in advance

kbialek, 2025-07-23:
I've added new set of metric groups for sg02lp1 family:
* `deye_sg02lp1` - sg02lp1 inverter
* `deye_sg02lp1_battery` - sg02lp1 inverter battery
* `deye_sg02lp1_timeofuse` - sg02lp1 inverter time-of-use settings

To test it use image `ghcr.io/kbialek/deye-inverter-mqtt:2025.07.1-beta-1` and update the metric groups in your configuration.

Tell me which communication protocol are you using to communicate with the logger?

koira, 2025-07-24:
> I've added a new set of metric groups for sg02lp1 family:
> 
> * `deye_sg02lp1` - sg02lp1 inverter
> * `deye_sg02lp1_battery` - sg02lp1 inverter battery
> * `deye_sg02lp1_timeofuse` - sg02lp1 inverter time-of-use settings
> 
> To test it use image `ghcr.io/kbialek/deye-inverter-mqtt:2025.07.1-beta-1` and update the metric groups in your configuration.
> 
> Tell me which communication protocol are you using to communicate with the logger?

thanks for that. tested immediately. please see https://github.com/kbialek/deye-inverter-mqtt/issues/243#issuecomment-3113670679 as it tackles both threads.

mobbster, 2025-07-24:
> I've added a new set of metric groups for sg02lp1 family:
>
>     * `deye_sg02lp1` - sg02lp1 inverter
>
>     * `deye_sg02lp1_battery` - sg02lp1 inverter battery
>
>     * `deye_sg02lp1_timeofuse` - sg02lp1 [...]

Seems to work well. (I haven't tested manipulating the time-of-use settings, as I don't use them, but they look like they are displaying the same values as the inverter screen).

The communication protocol is "tcp"

Many thanks

kbialek, 2025-07-26:
> The communication protocol is "tcp"

@mobbster Do you use the default port (8899)?

mobbster, 2025-08-28:
Thanks for your fix to this matter - it works perfectly.

closed 2025-08-28 by mobbster