kbialek/deye-inverter-mqtt #249: SUN-M200G4-EU-Q0

fjumeaux, 2025-09-17:
Hello,

Thank you for this project — the integration of my DEYE M200G4-EU-Q0 with deye-mqtt-inverter works perfectly.

However, I noticed an issue with the radiator_temp value published via MQTT. Every 60 seconds, the inverter sends two consecutive messages:
	•	the first message contains the temperature multiplied by 10 (e.g. 297),
	•	the second message contains the correct value (e.g. 29.7).

This leads to incorrect spikes in Domoticz/Home Assistant sensor history and causes false alarms (for example when I set a threshold at 47 °C -> 297>47 = alarm).

Would it be possible to filter out the first (x10) value or normalize it inside deye-mqtt-inverter, so that only the correct temperature is published to MQTT?

Thanks a lot for your work!

kbialek, 2025-09-17:
@fjumeaux Share your configuration

fjumeaux, 2025-09-18:
Hello,

I confirm that it works. Without “string”, and only with “micro”, I no longer have this double sending. And I can properly handle my radiator temperature alarm.

Many thanks.

However, I also confirm that I no longer have the “igbt_temp” information, which must have come from “string”.

closed 2025-09-25 by kbialek